wled/WLED #4252: matter support

Night-Coder2, 2024-11-06:
matter protocol has become quite popular and wled can have a massive upgrade with this. matter is supported by almost all platforms. also support for esp32 c6 will aid to this request as esp32 c6 supports zigbee

DedeHai, 2024-11-06:
C6 support is a year or two away.
Matter does not seem feasible: when looking at this https://docs.espressif.com/projects/esp-matter/en/latest/esp32/optimizations.html it seems like it uses A LOT of flash. Even when disabling every non-essential WLED function, there is  maybe 100kB available, the Matter protocol implementation appears to be using several time this amount. So for now, this is not possible.

closed 2024-11-06 by DedeHai

Night-Coder2, 2024-11-06:
why does c6 cause problems other chip supports wer not so bad

DedeHai, 2024-11-06:
https://github.com/Aircoookie/WLED/issues/3078#