wled/WLED #4257: How to achieve a chasing effect without covering？

czweb, 2024-11-08:
I want to use WLED to achieve this chasing effect. I have been thinking about it for a long time but I can't think of a way. Can you help me?
1. It can be triggered by UDP, Button, API, or any other method.
2. After being triggered, the light strip starts to display the chasing effect, and the newly triggered chasing effect will not cover the previous effect;
3. The ultimate goal is to use this chasing effect to display energy convergence on the LED screen at the end point;
Reference Video：https://www.youtube.com/shorts/fDVOS8r-pdg

DedeHai, 2024-11-08:
that is not WLED in the video right?
you need to write a custom effect, probably even a usermod to handle the button trigger.

czweb, 2024-11-08:
> 视频中不是 WLED 对吧？ 您需要编写自定义效果，甚至可能需要 usermod 来处理按钮触发器。

The video is not made by WLED.
I don't know how to use the method you mentioned. . .

softhack007, 2024-11-08:
### Overlay
https://kno.wled.ge/features/effects/#effect-overlay

"Effects without covering" can be achieved by effects that have an "Overlay" option - for example the "Two Dot" effect. 

<br/>

[image]

### Triggering presets by button 
.... or by other means is a different matter - please ask on discord
[image] to get help from the WLED user community.

softhack007, 2024-11-08:
> 3\. Reference Video：
>     https://www.youtube.com/shorts/fDVOS8r-pdg

This is currently not possible in WLED. As @DedeHai said, you would need to program your own usermod (i.e. an extension of WLED) to react on an external event (button) and run your own custom effect code.

czweb, 2024-11-11:
Finally found a way to achieve this effect using audio firmware. Just replace the microphone with a switch. I'll take screenshots and videos and send them in later this week.