wled/WLED #4259: WLED Staircase Mod: Not saving number led, led segments and presets

mani94321, 2024-11-08:
### What happened?

I used the bin files from 
https://github.com/srg74/WLED-wemos-shield/blob/master/resources/Firmware/%40Aircoookie/Latest/WLED_0.14.4_esp32_staircase_mod.bin

and 
https://github.com/srg74/WLED-wemos-shield/blob/master/resources/Firmware/%40Aircoookie/Dev/0.15.0-b7/WLED_0.15.0-b7_esp32_staircase_mod.bin

I set the number of leds in LED preference, also created present and segments. But when WLED is restarted all setting are gone. 

### To Reproduce Bug

Installed WLED from above sources for ESP-32. change number of LEDs and save, create presets or segments and save. Restart WLED and all segments, number of Leds are gone.

### Expected Behavior

I should be able to store the number of LEDs and segments.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.4 (build 2405180)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-11-08:
@mani94321 did you create a preset with all your segments etc, and set this in "Apply Preset at boot" in LED settings?

https://kno.wled.ge/basics/faq/#my-segments-are-gone-after-a-reboot
(the KB entry is a bit old - you may choose any preset for boot)

[image]

mani94321, 2024-11-08:
yes i have created segments and tried saving at different presets. I get the message encircled in red. 
[image]

yes i have tried applying preset 16 at boot but after restart , all gone

.

softhack007, 2024-11-08:
Do you use "Make a segment for each output"?
I think that there were some bugfixes in 0.15.0 related to this feature.

mani94321, 2024-11-09:
> Do you use "Make a segment for each output"? I think that there were some bugfixes in 0.15.0 related to this feature.

no, i did not use it. Its not only the segments i cant save, the number of LEDs are also lost once WLED is restarted. 

[image]

closed 2025-03-17 by github-actions[bot]