wled/WLED #4260: In 0.15.0-b7 and -b6, the "release" is always returned as `1`

Moustachauve, 2024-11-08:
### What happened?

Release is always `1` instead of `ESP32` or whatever should be needed.

### To Reproduce Bug

1) Download the release version from Github for 0.15.0-b7-ESP32.bin
2) flash it to an esp32
3) go to [ipAddress]/json/info
4) notice the `release` field is always `1`

### Expected Behavior

the `release` field should be `ESP32`.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0-b7 1 (ESP32-D0WDQ5 build 2410270)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Moustachauve, 2024-11-08:
I didn't try just manually building the release on my side, maybe they just need to be rebuilt with the correct flags in platformIO? Maybe the person that built them had an override or something.

softhack007, 2024-11-08:
Hi @Moustachauve,

I've just tried it, and indeed the firmware `WLED_0.15.0-b7_ESP32.bin` shows a wrong release string:

[image]
```JSON
{
    "ver": "0.15.0-b7",
    "vid": 2410270,
    "cn": "Kōsen",
    "release": "1",
```

<br/>

While `WLED_0.15.0-b7_ESP32_WROVER.bin` has the correct info

[image]

```JSON
{
    "ver": "0.15.0-b7",
    "vid": 2410270,
    "cn": "Kōsen",
    "release": "ESP32_WROVER",
```

netmindz, 2024-11-23:
It is based on the WLED_RELEASE_NAME define, which is also used to define the name of the output bin by reading this define with _get_cpp_define_value in output_bins.py

Therefore I think the value is being set correctly as the define, but is failing to then be properly read into releaseString in wled.h in some cases. 

I don't think it's anything to do with xml.cpp as all the version and release is single replacement

     `d.getElementsByClassName("sip")[0].innerHTML = "WLED 0.15.0-b7<br>1<br>(ESP32-D0WDQ6 build 2410270)";
`

netmindz, 2024-11-23:
I think I've worked out what is wrong, just confirming fix

netmindz, 2024-11-23:
So the issue is that anytime  `WLED_RELEASE_NAME`'s value matches the name of another constant, then we don't see the value of `WLED_RELEASE_NAME` but the expanded value of the macro of the same name, so `ESP32` has the value of 1 so hence why we see 1

I am unsure if this is something about how this line works

`WLED_GLOBAL char releaseString[] _INIT(TOSTRING(WLED_RELEASE_NAME)); // somehow this will not work if using "const char releaseString[]`

Or if it's part of the actual build process that is doing the expanding

I have tried using `WLED_RELEASE_NAME="ESP32"` and `WLED_RELEASE_NAME='ESP32'` but both suffer the same fate, using a value that can't expand, e.g ESP32zz  works as expected

willmmiles, 2024-11-23:
Try replacing `TOSTRING` with `STRINGIFY`.  IIRC the difference between the two is that `TOSTRING` performs recursive macro expansion, but `STRINGIFY` doesn't.

That said, I think best practices would be to ensure the define is always a literal right from the getgo - eg. `-D WLED_RELEASE_NAME=\"ESP32\"` - instead of using `STRINGIFY` at all.

netmindz, 2024-11-23:
> Try replacing `TOSTRING` with `STRINGIFY`. IIRC the difference between the two is that `TOSTRING` performs recursive macro expansion, but `STRINGIFY` doesn't.
>
> That said, I think best practices [...]

I tried your suggestion of STRINGIFY, but then we just get no expansion at all, so we see "WLED_RELEASE_NAME" as release

softhack007, 2024-11-23:
Interesting find, and totally makes sense 👍 
I think the it's best to add this to our toDo list for 0.15.0-RC1 #4287 

* either finding a solution that avoids inner macro expansion
* or by putting release name into \" quotes in platformIO.ini, and checking if the python scripts for the pio build need to be adjusted to work with quoted release names.

netmindz, 2024-11-23:
Swapping the release name to already have the quotes is tricky as we are defining it as a command line arg with -D, so it does not behave in the same way as say `DEFAULT_OTA_PASS`

i.e

```
#define WLED_FOO "ESP32"
WLED_GLOBAL char releaseString[] _INIT(WLED_FOO); // somehow this will not work if using "const char releaseString[]

```
works, however

`-D WLED_RELEASE_NAME="ESP32"` with `WLED_GLOBAL char releaseString[] _INIT(WLED_RELEASE_NAME);` does not, but `-D WLED_RELEASE_NAME=\"ESP32\"` does

netmindz, 2024-11-23:
Fixed, see https://github.com/netmindz/WLED/releases/tag/v0.15.0-rc1c

closed 2024-11-25 by netmindz