wled/WLED #4261: Preset upload / Wifi settings via serial console / Documentation... :(

Sebbb, 2024-11-09:
### What happened?

I'm frustrated :( I really like WLED. Great software, open source, what could I want more.. But sometimes I really hate it. Like now.
4 hours ago, I only wanted to update a preset file from another WLED system. 
After I did it, it rebooted, but it seems the wifi settings resetted. The AP is again called "WLED-AP", but "wled1234" doesn't work as password. I tried a lot, but I wasn't able to connect via wifi.
So I opened the glued chassis of my LED (...), and went to https://install.wled.me/ in the hope to be able to do anything via the serial console in that nice install wizard.
That gives me 2 options, "Install Wled" and "Console". I would have expected the "change wifi settings" dialog that I got when I originally installed WLED, but that's not shown there.
WLED kind of reacts and is active, when sending a "v", it responds with "WLED 2407070". So it must be somehow responsive...
But I don't find any documentation on how to set the wifi password or how to connect to an AP via that serial console :(

So, my expectations would have been:
- uploading presets doesn't somehow kill the wifi configuration
- connecting via serial console to my ESP32-S2 shows all options in the installer, like "connect to wifi", which is there on my esp8266.
- there would be a way to reset the wifi config (I found an old bug report stating that power cycling several times wold reset it, that the default password would work, but that wasn't the case for me)
- there was a way to reset the wifi password or to connect to an AP via serial console, without completey reinstalling WLED and losing all config. At least, for my esp32-s2 I dind't find a way
- having a documentation for all serial commands
- Not being up at 7:30am trying to fix that stuff :(

Sorry again, I know, WLED is open source, you don't get paid and it's a lot of effort you put into that.. But right now, I'm just frustrated because there's so little missing to make it a even better product... :(

Am I missing something? Is that stuff documented? Is there a way to easily set wifi credentials without losing the LED config via serial console? Have I missed that documtation?

### To Reproduce Bug

See above

### Expected Behavior

See above

### Install Method

Binary from WLED.me

### What version of WLED?

2407070

### Which microcontroller/board are you seeing the problem on?

ESP32-S2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2024-11-09:
I am sorry for your troubles. you are using:

- a beta version (not even the latest)
- an S2: "Some of the newer types like ESP32-C3, ESP32-S2 and ESP32-S3 are experimentally supported, but productive use is not recommended"

you may have stumbled upon a bug, it may be fixed in the latest beta (not sure). To recover your ESP, try uploading the latest beta, if that does not work, try erasing the flash and do a fresh install.

Sebbb, 2024-11-10:
@DedeHai You're absolutely right... :) And I didn't even have backups...
Uploading the latest beta helped, afterwards I was also able to configure the wifi via the installer, only my confiuration was lost.
Part of my frustration was that WLED reacted (showed the version number), but I wasn't able to configure the wifi via serial console because the protocol/serial commands were nowhere documented.

If to take anything from this, what would have helped was a documentation about the serial console protocol, or maybe additional commands to dump the configuration, to create a backup in case all other access fails :)

DedeHai, 2024-11-10:
its up to anyone to contribute to the project, if you find missing documentation, feel free to add it: https://github.com/Aircoookie/WLED-Docs