wled/WLED #4265: BLE Mesh support

dburner, 2024-11-09:
**Is your feature request related to a problem? Please describe.**
I have some existing LEDs on my stairs, but they are wired using only 2 wires (wires goes in some conduit inside the wall, cant add a third one). 
I want to enhance my LEDs with ESP32-C3 Mini and a 7 Ring LED. In each LED channel there seems to be enough room to squeeze an ESP32-C3 Super mini.
The problem is I don't think adding 16 clients to my Wifi would be too great since they are not on wifi6 and I think the connection in there is pretty pour.

**Describe the solution you'd like**
I would like 1 WLED to be a primary, and the rest to be secondary connected to the primary only using Bluetooth Mesh. 
When the primary changes the effect or color I want to signal the secondaries (via BLE mesh) the new configuration.

[image]

**Describe alternatives you've considered**
Connecting all controllers via WiFi.

**Additional context**
Does anyone who knows the codebase think it's possible to implement this as a user mod? I could give it a try

DedeHai, 2024-11-10:
Bluetooth is not possible, it uses too much flash.
Can you use ESPnow sync?

dburner, 2024-11-10:
I think I could, but how? From what i see online its only for the wizz remote?

DedeHai, 2024-11-10:
you can add ESPnow sync in settings, this can be used to sync ESPs without the use of wifi connection.

DedeHai, 2024-11-11:
its still a beta feature, see the changelog:
https://github.com/Aircoookie/WLED/releases/tag/v0.15.0-b1