wled/WLED #4268: DDP with mdns names - feature request

DarthWeber, 2024-11-10:
**Is your feature request related to a problem? Please describe.**
Sometimes i need to change the IP of my WLED Controller. Since i am using DDP i need to change the according ip in the DDP settings

**Describe the solution you'd like**
A test field which could be filled with the IP or the mDNS names like wled-1234.local

**Describe alternatives you've considered**
fixed IP - but in rare cases like changes of the router fixed IPs are lost.

Thank you for your ideas for making WLED better!

netmindz, 2025-07-03:
Was there a PR or just commit straight to main @blazoncek ?

Can we update with reference then close if completed?

blazoncek, 2025-07-03:
My fork only, for now, as there are some issues to be resolved and thoroughly tested.

> Can we update with reference then close if completed?

Once PR is made and merged.

blazoncek, 2025-07-14:
@DarthWeber please test #4768 and report back (in PR).

PR #4768 (2025-08-02): Add mDNS resolution for network bus
Fixes #4268

Only for ESP32

## Summary by CodeRabbit

* **New Features**
  * Added support for specifying a hostname or custom text for each network-based LED output (excluding ESP8266 devices).
  * Hostnames are now resolved to IP addresses automatically on ESP32 devices for network LED outputs.
  * The web UI for LED outputs now includes a "Host" input field for compatible devices.

* **Improvements**
  * Custom text/hostname is now saved and restored with LED bus configurations.
  * Invalid input fields are visually highlighted in red for easier identifi [... cut]
commit a24420ae (2025-08-02): Fix #4268
closed 2025-08-02 by blazoncek