wled/WLED #4269: Crashes when using HTTP API within MQTT

dosipod, 2024-11-10:
### What happened?

Crashes when using HTTP API within MQTT  . I am raising this on behave of @mcsSolutions  as it is also confirmed from our side .Original post on discord 
https://discord.com/channels/473448917040758787/779395228624617512/1304888652161220729

( I never used nor will use HTTP API within MQTT but only helping log this )

### To Reproduce Bug

User PL=1 with mqtt 
[image]

### Expected Behavior

No crash

### Install Method

Self-Compiled

### What version of WLED?

0.15.0-b7 (build 2410270)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
[21:28:53]Connected! IP address: 10.0.6.18
[21:28:53]Init STA interfaces
[21:28:53]mDNS started
[21:28:53]Reconnecting MQTT
[21:28:53]Reconnecting MQTT
[21:28:53]Loop took 38ms.
[21:28:53]Usermods took 0ms.
[21:28:53]Strip took 0ms.
[21:28:53]MQTT ready
[21:28:53]Publish MQTT
[21:28:53]MQTT ready
[21:28:53]Publish MQTT
[21:28:56]MQTT msg: wled/WLED18/api
[21:28:56]PL=1
[21:28:56]JSON buffer locked. (15)
[21:28:56]API req: win&PL=1
[21:28:56]Playlist unloaded.
[21:28:56]Request to apply preset: 1
[21:28:56]Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
[21:28:56]Core 1 register dump:
[21:28:56]PC      : 0x401329d6  PS      : 0x00060230  A0      : 0x8010b1b8  A1      : 0x3ffdcf00  
[21:28:56]A2      : 0x00000000  A3      : 0x3ffe37c8  A4      : 0x3ffc3954  A5      : 0x0000003c  
[21:28:56]A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80123518  A9      : 0x3ffdce10  
[21:28:56]A10     : 0x000000ad  A11     : 0x94cb258c  A12     : 0x94cb258c  A13     : 0x3ffc3954  
[21:28:56]A14     : 0x3ffdce4c  A15     : 0x3ffe39bc  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
[21:28:56]EXCVADDR: 0x0000000c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
[21:28:56]
[21:28:56]ELF file SHA256: 0000000000000000
[21:28:56]
[21:28:56]Backtrace: 0x401329d3:0x3ffdcf00 0x4010b1b5:0x3ffdcf40 0x40106646:0x3ffdcfe0 0x401c56c6:0x3ffdd080 0x40112a75:0x3ffdd0a0 0x401c589d:0x3ffdd110 0x40113cfe:0x3ffdd140 0x40112641:0x3ffdd190 0x401126d2:0x3ffdd200 0x40129d15:0x3ffdd220 0x40129d95:0x3ffdd260 0x4012a3c6:0x3ffdd280 0x4008cd1e:0x3ffdd2b0
[21:28:56]
[21:28:56]Rebooting...
[21:28:56]ets Jun  8 2016 00:22:57
[21:28:56]
[21:28:56]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[21:28:56]configsip: 0, SPIWP:0xee
[21:28:56]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[21:28:56]mode:DOUT, clock div:2
[21:28:56]load:0x3fff0030,len:1184
[21:28:56]load:0x40078000,len:12784
[21:28:56]load:0x40080400,len:3032
[21:28:56]entry 0x400805e4
[21:28:57]
[21:28:57]---WLED 0.15.0-b7 2410270 INIT---
[21:28:57]
[21:28:57]esp32 v3.3.6-16-gcc5440f6a2
[21:28:57]arduino-esp32 v1.0.x
[21:28:57]
[21:28:57]CPU:   ESP32-D0WDQ5 rev.1, 2 core(s), 240 MHz.
[21:28:57]FLASH: 4 MB, Mode 3 (DOUT), speed 40 MHz.
[21:28:57]heap 308336
[21:28:57]JSON buffer allocated: 32767
[21:28:57]TX power: 78/78
[21:28:57]PIN ALLOC: Pin 1 successfully allocated by 0x89.
[21:28:57]PIN ALLOC: 0x00000000000002.
[21:28:57]Registering usermods ...
```

### Anything else?

[MQTT_PL_CRASH.txt](https://github.com/user-attachments/files/17692601/MQTT_PL_CRASH.txt)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mcsSolutions, 2024-11-10:
DEBUG from WLED 0.15.0-b7
Stimuli MQTT message to select preset 2 and then preset 3.

[MQTT LOG]
12:04:46 PM Topic wled/11020c/api Payload PL=2
12:04:50 PM Topic wled/11020c/status Payload offline
12:04:50 PM Topic wled/11020c/status Payload offline
12:04:50 PM Topic wled/11020c/g Payload 128
12:04:50 PM Topic wled/11020c/c Payload #0D39FF
12:04:50 PM Topic wled/11020c/status Payload online
12:04:50 PM Topic wled/11020c/v Payload <?xml version="1.0" ?><vs><ac>128</ac><cl>13</cl><cl>57</cl><cl>255</cl><cs>0</cs><cs>0</cs><cs>0</cs><ns>0</ns><nr>1</nr><nl>0</nl><nf>1</nf><nd>60</nd><nt>0</nt><fx>0</fx><sx>128</sx><ix>128</ix><fp>0</fp><wv>-1</wv><ws>0</ws><ps>1</ps><cy>0</cy><ds>WLED</ds><ss>0</ss></vs>
12:04:50 PM Topic wled/11020c/g Payload 128
12:04:50 PM Topic wled/11020c/c Payload #0D39FF
12:04:50 PM Topic wled/11020c/status Payload online
12:04:50 PM Topic wled/11020c/v Payload <?xml version="1.0" ?><vs><ac>128</ac><cl>13</cl><cl>57</cl><cl>255</cl><cs>0</cs><cs>0</cs><cs>0</cs><ns>0</ns><nr>1</nr><nl>0</nl><nf>1</nf><nd>60</nd><nt>0</nt><fx>0</fx><sx>128</sx><ix>128</ix><fp>0</fp><wv>-1</wv><ws>0</ws><ps>1</ps><cy>0</cy><ds>WLED</ds><ss>0</ss></vs>
12:06:59 PM Topic wled/11020c/status Payload offline
12:06:59 PM Topic wled/11020c/status Payload offline
12:06:59 PM Topic wled/11020c/g Payload 128
12:06:59 PM Topic wled/11020c/c Payload #0D39FF
12:06:59 PM Topic wled/11020c/status Payload online
12:06:59 PM Topic wled/11020c/v Payload <?xml version="1.0" ?><vs><ac>128</ac><cl>13</cl><cl>57</cl><cl>255</cl><cs>0</cs><cs>0</cs><cs>0</cs><ns>0</ns><nr>1</nr><nl>0</nl><nf>1</nf><nd>60</nd><nt>0</nt><fx>0</fx><sx>128</sx><ix>128</ix><fp>0</fp><wv>-1</wv><ws>0</ws><ps>1</ps><cy>0</cy><ds>WLED</ds><ss>0</ss></vs>
12:08:29 PM Topic wled/11020c/status Payload offline
12:08:59 PM Topic wled/11020c/g Payload 128
12:08:59 PM Topic wled/11020c/c Payload #0D39FF
12:08:59 PM Topic wled/11020c/status Payload online
12:08:59 PM Topic wled/11020c/v Payload <?xml version="1.0" ?><vs><ac>128</ac><cl>13</cl><cl>57</cl><cl>255</cl><cs>0</cs><cs>0</cs><cs>0</cs><ns>0</ns><nr>1</nr><nl>0</nl><nf>1</nf><nd>60</nd><nt>0</nt><fx>0</fx><sx>128</sx><ix>128</ix><fp>0</fp><wv>-1</wv><ws>0</ws><ps>1</ps><cy>0</cy><ds>WLED</ds><ss>0</ss></vs>
12:09:58 PM Topic wled/11020c/api Payload PL=3
12:09:58 PM Topic wled/11020c/g Payload 14
12:09:58 PM Topic wled/11020c/c Payload #0D39FF
12:09:58 PM Topic wled/11020c/status Payload online
12:09:58 PM Topic wled/11020c/v Payload <?xml version="1.0" ?><vs><ac>14</ac><cl>13</cl><cl>57</cl><cl>255</cl><cs>0</cs><cs>0</cs><cs>0</cs><ns>0</ns><nr>1</nr><nl>0</nl><nf>1</nf><nd>60</nd><nt>0</nt><fx>91</fx><sx>128</sx><ix>128</ix><fp>0</fp><wv>-1</wv><ws>0</ws><ps>3</ps><cy>0</cy><ds>WLED</ds><ss>0</ss></vs>
12:11:29 PM Topic wled/11020c/status Payload offline

[WLED SERIAL]
N9[1D][01]:5[05]"[05]Q[05][01]"=multiple clients.
JSON buffer released. (12)
---DEBUG INFO---
Runtime: 30000
Unix time: 30,000
Free heap: 181072
TX power: 75/78
Wifi state: 3
ESP-NOW state: 0
State time: 0
NTP last sync: 999000000
Client IP: 192.168.0.204
Loops/sec: 2034
Loop time[ms]: 0/717
UM time[ms]: 0/1
Strip time[ms]:0/26
Segments: 1 -> 68/0B
  Seg: 30,1 [A=1, 2D=0, RGB=1, W=0, CCT=0]
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B
Reconnecting MQTT
MQTT msg: wled/11020c/api
PL=2
JSON buffer locked. (15)
API req: win&PL=2
Playlist unloaded.
Request to apply preset: 2
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x401329d6  PS      : 0x00060830  A0      : 0x8010b1b8  A1      : 0x3ffdd360  
A2      : 0x00000000  A3      : 0x3ffdfa54  A4      : 0x3ffc3954  A5      : 0x0000003c  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80123518  A9      : 0x3ffdd270  
A10     : 0x000000ae  A11     : 0x36943335  A12     : 0x36943335  A13     : 0x3ffc3954  
A14     : 0x3ffdd2ac  A15     : 0x3ffe1544  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x0000000c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
[... 249 lines cut ...]
Initial connect or forced reconnect.
Defaulf WiFi used.
initConnection() called.
Access point disabled (init).
Connecting to Anthem...
JSON buffer locked. (9)
Applying preset: 1
JSON buffer released. (9)
Loop took 53ms.
Usermods took 0ms.
Strip took 26ms.
Loop took 717ms.
Usermods took 0ms.
Strip took 0ms.

Connected! IP address: 192.168.0.204
Init STA interfaces
mDNS started
Reconnecting MQTT
Reconnecting MQTT
Loop took 35ms.
Usermods took 0ms.
Strip took 0ms.
MQTT ready
Publish MQTT

willmmiles, 2024-11-10:
A repeatable crash, my favorite kind!

willmmiles, 2024-11-11:
Thanks for the report!  This was indeed my fault. ;)  Fix is in PR #4271.

blazoncek, 2024-11-11:
Confirming crash in AWS.

```
Playlist unloaded.
Request to apply preset: 1
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40135f32  PS      : 0x00060130  A0      : 0x80107ddc  A1      : 0x3ffe79c0  
A2      : 0x00000000  A3      : 0x3ffebbac  A4      : 0x3ffc8d54  A5      : 0x0000000f  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x80126bc8  A9      : 0x3ffe78d0  
A10     : 0x000000b2  A11     : 0x0d0db0f9  A12     : 0x0d0db0f9  A13     : 0x3ffc8d54  
A14     : 0x3ffe790c  A15     : 0x3ffecb20  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x0000000c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x40135f32:0x3ffe79c0 0x40107dd9:0x3ffe7a00 0x401033ea:0x3ffe7ab0 0x401cca06:0x3ffe7b50 0x4010c6d5:0x3ffe7b70 0x401ccc55:0x3ffe7be0 0x4010d972:0x3ffe7c10 0x4010c291:0x3ffe7c60 0x4010c322:0x3ffe7cd0 0x4012d49d:0x3ffe7cf0 0x4012d51d:0x3ffe7d30 0x4012db4e:0x3ffe7d50 0x4008d55e:0x3ffe7d80
  #0  0x40135f32:0x3ffe79c0 in AsyncWebServerRequest::send(AsyncWebServerResponse*) at .pio/libdeps/wemos_d1_mini32_debug/ESPAsyncWebServerWLED/src/WebRequest.cpp:1015
  #1  0x40107dd9:0x3ffe7a00 in handleSet(AsyncWebServerRequest*, String const&, bool) at wled00/set.cpp:1196
  #2  0x401033ea:0x3ffe7ab0 in onMqttMessage(char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int) at wled00/mqtt.cpp:112
  #3  0x401cca06:0x3ffe7b50 in std::_Function_handler<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int), void (*)(char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>::_M_invoke(std::_Any_data const&, char*&&, char*&&, AsyncMqttClientMessageProperties&&, unsigned int&&, unsigned int&&, unsigned int&&) at /Users/blaz/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1871
  #4  0x4010c6d5:0x3ffe7b70 in std::function<void (char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int)>::operator()(char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int) const at wled00/src/dependencies/async-mqtt-client/AsyncMqttClient.cpp:50
      (inlined by) AsyncMqttClient::_onMessage(char*, char*, unsigned char, bool, bool, unsigned int, unsigned int, unsigned int, unsigned short) at wled00/src/dependencies/async-mqtt-client/AsyncMqttClient.cpp:523
  #5  0x401ccc55:0x3ffe7be0 in void std::_Mem_fn_base<void (AsyncMqttClient::*)(char*, char*, unsigned char, bool, bool, unsigned int, unsigned int, unsigned int, unsigned short), true>::operator()<char*, char*, unsigned char, bool, bool, unsigned int, unsigned int, unsigned int, unsigned short, void>(AsyncMqttClient*, char*&&, char*&&, unsigned char&&, bool&&, bool&&, unsigned int&&, unsigned int&&, unsigned int&&, unsigned short&&) const at /Users/blaz/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:600
      (inlined by) void std::_Bind<std::_Mem_fn<void (AsyncMqttClient::*)(char*, char*, unsigned char, bool, bool, unsigned int, unsigned int, unsigned int, unsigned short)> (AsyncMqttClient*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>, std::_Placeholder<5>, std::_Placeholder<6>, std::_Placeholder<7>, std::_Placeholder<8>, std::_Placeholder<9>)>::__call<void, char*&&, char*&&, unsigned char&&, bool&&, bool&&, unsigned int&&, unsigned int&&, unsigned int&&, unsigned short&&, 0u, 1u, 2u, 3u, 4u, 5u, 6u, 7u, 8u, 9u>(std::tuple<char*&&, char*&&, unsigned char&&, bool&&, bool&&, unsigned int&&, unsigned int&&, unsigned int&&, unsigned short&&>&&, std::_Index_tuple<0u, 1u, 2u, 3u, 4u, 5u, 6u, 7u, 8u, 9u>) at /Users/blaz/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1074
      (inlined by) void std::_Bind<std::_Mem_fn<void (AsyncMqttClient::*)(char*, char*, unsigned char, bool, bool, unsigned int, unsigned int, unsigned int, unsigned short)> (AsyncMqttClient*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>, std::_Placeholder<5>, std::_Placeholder<6>, std::_Placeholder<7>, std::_Placeholder<8>, std::_Placeholder<9>)>::operator()<char*, char*, unsigned char, bool, bool, unsigned int, unsigned int, unsigned int, unsigned short, void>(char*&&, char*&&, unsigned char&&, bool&&, bool&&, unsigned int&&, unsigned int&&, unsigned int&&, unsigned short&&) at /Users/blaz/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1133
      (inlined by) std::_Function_handler<void (char*, char*, unsigned char, bool, bool, unsigned int, unsigned int, unsigned int, unsigned short), std::_Bind<std::_Mem_fn<void (AsyncMqttClient::*)(char*, char*, unsigned char, bool, bool, unsigned int, unsigned int, unsigned int, unsigned short)> (AsyncMqttClient*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>, std::_Placeholder<4>, std::_Placeholder<5>, std::_Placeholder<6>, std::_Placeholder<7>, std::_Placeholder<8>, std::_Placeholder<9>)> >::_M_invoke(std::_Any_data const&, char*&&, char*&&, unsigned char&&, bool&&, bool&&, unsigned int&&, unsigned int&&, unsigned int&&, unsigned short&&) at /Users/blaz/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1871
[... cut ...]
  #12 0x4008d55e:0x3ffe7d80 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6360
entry 0x400806b4
```

commit 223b97b8 (2024-11-12): handleSet: Fix incorrect response generation

Don't generate a response if there's no HTTP request.

Fixes #4269
commit d56fd0e1 (2024-11-25): handleSet: Fix incorrect response generation

Don't generate a response if there's no HTTP request.

Fixes #4269
commit a1b332fc (2024-11-25): handleSet: Fix incorrect response generation

Don't generate a response if there's no HTTP request.

Fixes #4269
closed 2024-11-14 by willmmiles