wled/WLED #4273: Skip Last LEDs

Fonzi03, 2024-11-11:
Having skip first LEDs is helpful but so it having skip last LEDs. 

I use this frequently between props to not have to cut the wire if its only a handful of pixels.

willmmiles, 2024-11-11:
This is already something you can do: try setting the strip or segment length to the number of LEDs you want WLED to control.  If there are more physical LEDs than you've specified, the extra LEDs will be left in their power on state, which is typically off.   (Note that if you're lowering the physical strip length setting, you may need to power cycle the LEDs to turn them off, as WLED will stop controlling them.)

dosipod, 2024-11-11:
Also using ledmaps

closed 2024-11-11 by softhack007

Fonzi03, 2024-11-11:
> Technicially speaking, " skip last LEDs" is the same as changing number of LEDs. ESP32 does not know how many leds you have on hardware side. we're just driving as many leds as the users configures. So if there are extra LEDs not configured, WLED will skip driving them.
> 
> If you want to be 100% sure that unused LEDs are "out", you can create segments and set unused segments to "solid" effect with color BLACK.

I would need to do your second suggestion. As I have multiple strings connected and I need to skip LEDs between strings which are connected to a single port. 

For now I plan to create a dummy string in XLights with the amount of pixels I need to skip and all of my sequences won't include that but it'll be assigned to the port. Therefore the e131/DDP data will just skip over those pixels.