wled/WLED #4275: ESP32-C SuperMini gets very hot

SunboX, 2024-11-11:
### What happened?

ESP32-C3 get's very hot, you nearly can't touch it anymore after only 10 seconds.

### To Reproduce Bug

I flashed the latest WLED [env:esp32c3dev] from today onto a ESP32-C SuperMini from here: https://de.aliexpress.com/item/1005006960183178.html

No LEDs were connected at this time.

After doing doing a search, I came across other users with the same Issue. Flashing ESPHome does not heat it up so much. So it is not related to Wifi usage or the board in general. It must be somehow WLED related - but why?

### Expected Behavior

ESP32-C3 gets only warm not burning hot.

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.4 (build 2405180)

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Anything else?

No, nothing was connected to the board, just USB power. The ESP32 chip itself got hot and also the voltage regulator.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2024-11-11:
not a bug, that is just how it is.

closed 2024-11-11 by DedeHai

SunboX, 2024-11-11:
@DedeHai And why doesn't it happen if I install ESPHome?

DedeHai, 2024-11-11:
Enable wifi sleep.

SunboX, 2024-11-11:
It seems to help a tiny little bit, but still getting very hot.

SunboX, 2024-11-11:
Even reducing f_cpu to 80MHz does no change.

softhack007, 2024-11-11:
@SunboX there are small -C3 boards out there which have thermal problems. Maybe you were unlucky to get one.

Some known ways to reduce heat production:
* important: do not run audioreactive - it creates high CPU load on -C3
* uncheck "disable wifi sleep" (=allow wifi sleep) in WLED led settings
* reduce target fps in WLED LED settings
* custom build with reduced CPU frequency (80mhz instead of 160mhz)

You can't compare esphome with WLED. 
Esphome basicially reads out a handful of sensors and sends them over to MQTT or HA, maybe drives a handful of outputs at low rates. So in esphome the CPU is idle most of the time, while WLED is actively calculating effects and doing other things that keep the CPU busy = produce heat.

SunboX, 2024-11-11:
@softhack007 Thanks for the explanation! That's really sad, the WEMOS D1 Mini (ESP8266) is also low CPU power but does not seem to have those issues. Do you know why?

softhack007, 2024-11-11:
No idea, sorry. 8266 is a completely different CPU.

softhack007, 2024-11-11:
Another idea that might help: 
Make a custom build with ` -DARDUINO_USB_CDC_ON_BOOT=0` instead of "=1" which is the default.

We had problems in the past with CDC_ON_BOOT enabled - the driver from espressif sometimes gets into an infinite loop at startup, which heats up the board. Unfortunately without CDC you can't set the wifi credentials in the online installer...

softhack007, 2024-11-11:
Did I mention that -C3 is cheap useless stuff?  Well that's my personal opinion as a developer. If you can, buy -S3 (but not the superMini) or classic esp32.

SunboX, 2024-11-12:
Yes, I know it's cheap, that's the reason why I want to use them. It's for small decorative stuff. I don't need an high end MyC for that. If it's cheap, it's fine. USB Type-C would be nice, because most chargers and cables nowadays have USB Type-C.

DedeHai, 2024-11-13:
Reason your WEMOS D1 Mini does not suffer from this: it is roughtly 3x the size. I have C3 supermini's mounted on a just slightly larger carrier PCB and it does not get hot. The main issue is: it was designed to be mounted on a PCB, if not mounted, it will run hot. The temperature is not an issue for the hardware, it can easily take it.