wled/WLED #4277: Disabling brightness limiter requires additional changes before saving

Fonzi03, 2024-11-13:
### What happened?

Cannot save after disabling automatic brightness

### To Reproduce Bug

If you go to LED Preferences and de-select "Enable automatic brightness limiter" and try to save without making any other changes you cannot save. Change any value anywhere else and you can save. 

### Expected Behavior

you should be able to save after changing a setting. 

### Install Method

Binary from WLED.me

### What version of WLED?

2410270

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-11-22:
As this affects the LEDs settings page, it would be good to understand what's going on before 0.15.0-RC1 https://github.com/Aircoookie/WLED/pull/4287

Fonzi03, 2024-11-22:
> As this affects the LEDs settings page, it would be good to understand what's going on before 0.15.0-RC1 https://github.com/Aircoookie/WLED/pull/4287

If you need me to test anything or provide more details I'm more than happy to do so.

DedeHai, 2024-11-24:
I just tried this and and saving works. 
@Fonzi03  did you try this on a fresh install or what is the history? Did you try clearing local browser storage?

Fonzi03, 2024-11-24:
> I just tried this and and saving works. 
> @Fonzi03  did you try this on a fresh install or what is the history? Did you try clearing local browser storage?

I don't remember if when this occured it was on a fresh install or not. I was having multiple different issues at the time and flashed a handful of times. If it can't be reproduced it's likely it was a secondary order issue. 

I can test again but probably not for another week or so.

dosipod, 2024-11-25:
If by "requires additional changes before saving" meaning it will not save unless you enter the "Maximum PSU Current" and next time also enter the "mA/LED" then yes we have reported that among others multiple times . Mostly in discord but I think it was seen as not important 

Uploading pref_no_save.mov…

Fonzi03, 2024-11-25:
> If by "requires additional changes before saving" meaning it will not save unless you enter the "Maximum PSU Current" and next time also enter the "mA/LED" then yes we have reported that among [...]

Precisely the issue, except it's not limited to mA/LED you just have to change literally any other value on the page for it to them allow saving.

dosipod, 2024-11-25:
Just to clarify as I had issue loading the video , in my test the issue is once you check the limiter so not the other way around , can you take a video of the issue so it is more clear