wled/WLED #4281: Boards freezing with V15 B6 and B7

DarthWeber, 2024-11-14:
### What happened?

Complete freezing, effect stops and board no longer connectable via wifi

### To Reproduce Bug

Would say this happens after V15 B4. Have observed this on several boards, but only ESP32. Self compiled with sound, dallas, multirelay.  Used in sync receiver and sender.

Difficult to reproduce since i do not find any real cause. Would like to know if anybody also observed this with the newer releases.

### Expected Behavior

No freezing?

### Install Method

Self-Compiled

### What version of WLED?

15 B7

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

KrX3D, 2024-11-14:
hi, i also have this bihavor but mostly with ESP32 S2 (lolin_s2_mini) but some different usermods as you

DedeHai, 2024-11-14:
can you use a debug build to see what the issue may be?

WouterGritter, 2024-11-19:
I had something similar on an ESP8266 board when upgrading from a v14 pre-release to the newest v15. My fix was to use the `compat` binary.