wled/WLED #4283: After update from 14.4 to 15.0 xx beta no wlan esp32-c3 supermini

oltas123, 2024-11-15:
### What happened?

no wlan connection after update from 14.4 to 15.0 beta  over OTA. Downgrade to 14.4 with the installer also no wlan connection anymore. Only erase over esp32 downlod tool and once again flash the 14.4 solves the problem. Serial Update to 15.0 xx also no wlan AP. 

### To Reproduce Bug

upgrade esp32-3 supermini to 15.0 xx beta

### Expected Behavior

wlan connection possible after update to 15.0 xx

### Install Method

Binary from WLED.me

### What version of WLED?

15.0 b7 

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2024-11-15:
some superminis have really bad wifi. did you try them very close to the router?
if they work then: check if the wifi power on the C3 is set to maximum under wifi settings.

oltas123, 2024-11-18:
Supermini is 30cm away from the router. WLED 14.4 info shows  signal strength 100%(- 26 dBm). After upgrade to 15.0 b7 again no connection to ESP-C3 and not seen in  the router interface.  No power settings  in webinterface of 14.4. No connection to the web interface of 15.0.

softhack007, 2024-11-18:
0.14.4 had the LOLIN_WIFI_FIX for all c3 boards, which was limiting power output to `WIFI_POWER_8_5dBm`. The fix is required for boards with a tiny "crystal" radio, and for boards with an under-sized LDO. 

If tx power is too high in some lolin s2 and c3 boards, as well as for "super mini" boards, wifi completely stops working.

This commit in 0_15 changed the logic: https://github.com/Aircoookie/WLED/commit/0af3063127079428f6c5cd9017977f560a98e47a allows users to increase the tx output in wifi settings. 

Unfortunately it means that - if you have a board that needs the fix - changing tx power to max permanently renders the WLED setup unusable. To recover, it's necessary to first erase flash (to remove cfg.json with the bad TX default), then re-install the firmware and start fresh. Uploading a previous cfg.json will trigger the TX power problem again, if this was changed in wifi settings.

softhack007, 2024-11-18:
@DedeHai, @blazoncek I'm not sure how to solve this problem - once the new tx power setting is put to max, the cfg.json is "poisoned" and users would need to erase flash. I think that's not a good situation.

Maybe we can:
a)  put a tx power limit when `LOLIN_WIFI_FIX` is defined, so users are not accidentally overloading their hardware, or
b) still use limited TX power in AP mode (so user may be able to reconnect to ap)

Plus the code in improv.cpp for finding wifi networks should also use the limited power output settings.

What do you guys think?

softhack007, 2024-11-18:
Update: found this commit in arduino-esp32 https://github.com/espressif/arduino-esp32/commit/c43187a3864bf81be632afb2c6eea42e6df63a01

To me this suggests that setTxPower is only working once that wifi is fully started -  up to ~1 second after the call to wifi.begin(). Maybe the basic wifi class that we used in 0.14 automatically waits, while multiWifi does not wait?

blazoncek, 2024-11-18:
I checked the build environments (for officially released binaries) and C3 does have LOLIN_WIFI_FIX included so default TX power is correct (or good).

There is also a warning in settings that says:
[image]
Perhaps it may need to be rephrased into "may become unuseable".

My own experiments showed that some boards are capable of handling stronger TX power so that was the reason to allow overrides.

If you take a look at use of `WiFi` object you can see that `setTxPower()` is called *after* `begin()` or `softAP()` in all cases (if I read the commit correctly the call fails if WiFi is not initialised). I do not know if adding a wait is possible in WLED code but never notices any issues with current implementation (except perhaps on S3 on which I have given up on as it was very unreliable for me).
The difference of *multiwifi* compared to the old "wifi" is the inclusion of multiple configurations but only one is active at any time. The switch to a different configuration takes place if connection cannot be established (retried every 18 s). This was rigorously tested.

The only way to know for sure what's happening is the inclusion of debug printouts which are not provided by @oltas123

oltas123, 2024-11-18:
Sorry, i don't know how to get debug printouts.

softhack007, 2024-11-18:
@oltas123 it's never too late to ask ;-)

Debug builds are providing additional diagnostic outputs when connected to a serial monitor by USB.
You can find a few generic debug builds here: https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/%40Aircoookie/Latest/Debug_builds

* upload a debug build for your board (OTA or via usb)
* connect the board to a computer by USB cable
* start a serial monitor tool (arduino IDE, VSCode, or even putty will do). 115200 baud, 8N1
* press rest on the board to restart your esp
* you should see lots of debug messages. Post them here so we can have a look.

blazoncek, 2024-11-19:
I would also like to point out to [this](https://wled.discourse.group/t/some-c3-super-minis-cant-connect-to-wi-fi/12153).

WouterGritter, 2024-11-19:
I had the same issue. WiFi would seem to work, I got around 20% of pings to go through but the latency was way high. HTTP did not work at all. Fix was to use the `compat` binary.

blazoncek, 2024-11-19:
@WouterGritter please open a new issue (or update existing one for ESP8266), this one is related to newer chips like C3 and S-series that are able to set TX power and use newer libraries.

WouterGritter, 2024-11-19:
> @WouterGritter please open a new issue (or update existing one for ESP8266), this one is related to newer chips like C3 and S-series that are able to set TX power and use newer libraries.

It isn't necessarily a problem because the fix (for me) is already known; to use the `compat` version of the binary. Not sure if this is something that needs to be fixed in the "main" binary then maybe... If so I can open an issue for that.

DedeHai, 2024-11-20:
> once the new tx power setting is put above 8.5db, the cfg.json is "poisoned" and users would need to erase flash. I think that's not a good situation.

Does a factory reset not set the config to defaults (or delete the file)?

> What do you guys think?

Its the age old question on what to do with all the bad hardware designs out there. Making WLED compatible to all of them limits the use of features for properly designed hardware. I think adjusting the warning should be good enough if the default setting is safe.

oltas123, 2024-11-20:
Hello, I was able to get the diagnostic output with an ESP32 and WLED-0.14.4_esp32_debug.bin. Unfortunately I couldn't find a suitable bin file for the ESP32-C3 and WLED_0.15.xx. I can live with version 14.4 on the ESP32-C3. It just won't be updated. Maybe the developers can create all the beta images with the debug option. I don't know how much work that takes and how much performance it costs. Thank you for the great project. Regards Olaf

DedeHai, 2024-11-20:
@oltas123 wladi just added a new feature to the compile helper page: you should be able to compile and download any configuration: [https://wled-compile.github.io/](https://wled-compile.github.io/)
select the version, the board and under options, just check "Include Debug flags", then compile and download

oltas123, 2024-11-20:
I hope it's the right one.

[debug_14_4_and_15_0_b7.txt](https://github.com/user-attachments/files/17835174/debug_14_4_and_15_0_b7.txt)

oltas123, 2024-11-20:
generierte Konfig

[env:custom_build]
extends = env:esp32c3dev
build_flags = ${env:esp32c3dev.build_flags} ${common.debug_flags} -D LOLIN_WIFI_FIX -D WLED_DEBUG

blazoncek, 2024-11-20:
Please erase flash before uploading b7 (or press and hole button for more than 10s if you have button attached and correctly configured). Then upload new dump.
It looks like C3 cannot decipher existing configuration properly.

oltas123, 2024-11-21:
I erased the flash. 
[esp-web-tools-logs (1).txt](https://github.com/user-attachments/files/17840662/esp-web-tools-logs.1.txt)

oltas123, 2024-11-21:
But no AP appears that you can connect to. Downgrade to 14.4 and all is fine.

blazoncek, 2024-11-21:
@DedeHai you also have C3, can you check on your devices? Mine work as expected (including C3 supermini).
If they work I would suspect HW issue combined with the newer ESP32 platform (6.3.2) used in 0.15.

blazoncek, 2024-11-21:
@oltas123 did you check forum link?

oltas123, 2024-11-21:
I checked the link. No success to manual connect over smartphone. But maybe you should close the issue. I stay at 14.4with this board. Thank you for your support.

DedeHai, 2024-11-21:
I use the latest beta all the time, but flashed from VScode if that makes a difference. Never noticed any issues with non-working wifi, even when set to max TX power.

blazoncek, 2024-11-21:
@oltas123 you can try and compile 0.15 using older platform for ESP32-C3. Use version 5.3.0 which was used in 0.14.

closed 2024-11-23 by oltas123

blazoncek, 2024-11-23:
You could explain what you did to alleviate your problem if you managed to resolve it.
Someone else might benefit form it.

oltas123, 2024-11-24:
Not solved so far. The esp32-c3 super mini passed away during the tests.

oltas123, 2024-11-25:
Ordered a new ESP32-C3 and all is fine. Update to 15xxbeta works.  The board is has a different design.
[image]