wled/WLED #4288: Firmware presets and playlist

viktor6, 2024-11-16:
How can I build a firmware with presets and a playlist?
Why can't I put the wled_presets.json file next to platformio.ini?

netmindz, 2024-12-05:
I know that some users have been experimenting with building a firmware file with default presets added, hopefully we can get that added as an official option once they get that working reliably

david-hogberg, 2024-12-05:
If you're using PlatformIO, you can add to your board configuration:

`board_build.filesystem = littlefs`
`board_build.filesystem_size = xxxx`

where xxxx is the size of the filesystem partition. Check the size of the partition in the csv file listed in `board_build.partitions` (probably named 'spiffs' rather than 'littlefs' though). 

Put your presets.json in a data directory and change the 'data_dir' config parameter to point to this directory (at the top of the platformio.ini file). You can of course use the default ./wled00/data folder for this as well but then it'll also encode all the htm-files etc. 

Finally use 'Build Filesystem Image' in PlatformIO to build a littelfs.bin file with your preset file (found in .pio/build/your_env). Flash this to your board at the appropriate offset. 

I've done this for an s3 mini board with 8MB flash (custom pcb design). Seems to be working fine so far, though I've also heard some accounts of discrepancies between pio and wled regarding littlefs handling that could cause issues. If anyone knows more about this let me know.

viktor6, 2024-12-05:
> If you're using PlatformIO, you can add to your board configuration:
>
> `board_build.filesystem = littlefs` `board_build.filesystem_size = xxxx`
>
> where xxxx is the size of the filesystem partition. [...]

Is this a theory or do you do it this way? Can you make a video on how to do it?

netmindz, 2024-12-06:
Thanks for sharing @david-hogberg, can you explain how you flash the littlefs.bin with the correct offset and where that value comes from?

david-hogberg, 2024-12-06:
@viktor6 Yes I do it this way. Any specific part you want more details on?

@netmindz You can also press 'Upload Filesystem Image' in PlatformIO after you've built the image :) 

Otherwise offsets are defined in a separate column in the partitions table. It'll look something like this:

| Name | Type | SubType | Offset | Size | Flags |
|------|------|---------|---------|------|-------|
| nvs | data | nvs | 0x9000 | 0x5000 | |
| otadata | data | ota | 0xe000 | 0x2000 | |
| app0 | app | ota_0 | 0x10000 | 0x200000 | |
| app1 | app | ota_1 | 0x210000 | 0x200000 | |
| spiffs | data | spiffs | 0x410000 | 0x3E0000 | |
| coredump | data | coredump | | 64K | |

If you're using esptool, you can then do `esptool.py --port [PORT] write_flash [OFFSET] [IMAGE_FILE]`

There's also a `merge_bin` command that lets you merge all the bin files to a single file.