wled/WLED #4292: wrong devicename (hostname) in the network after power cut

Rodney2K, 2024-11-18:
### What happened?

Hello,
I set the device name in User Interface Settings for instance "WLED". In the Network you can find the device under "wled-WLED".  After power cut  the network name of the device is always something like "esp32c3-00d6d0". Now you can click on "Save & Connect" in the WIFI Settings and the device is "wled-WLED" again. 
I need to find the device by network name. 
Greetings ..

### To Reproduce Bug

1) set the device name in user interface settings an save it
2) save the wifi settings - the device name in user interface setting will be the network name of the device by nslookup over the ip
3) switch off the current - than switch on the current after seconds 
4) now the device has a name like "esp32c3-00D6D0" 

### Expected Behavior

If I set the devicename, I want to reach the device by its name in the network.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_0.14.4_ESP32-C3.bin (2024-05-18)

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

BBaoVanC, 2025-03-19:
This issue is not stale. Please don't close it. I will test the commit that was mentioned early in a little bit, just waiting to have some time.

Rodney2K, 2025-03-20:
The discussion on this issue you find here -> https://github.com/wled/WLED/issues/4586!

Closing this issue here!

closed 2025-04-28 by blazoncek