wled/WLED #4293: "magic numbers" for MQTT topic buffer length in mqtt.cpp and button.cpp (use already defined MQTT_MAX_TOPIC_LEN)

WouterGritter, 2024-11-19:
### What happened?

I was modifying the source code on a fork for my custom MQTT topic structure and noticed `MQTT_MAX_TOPIC_LEN` is not used in all places the associated variables `mqttDeviceTopic` and `mqttGroupTopic` are used.

When increasing this defined variable, buffer overflows will occur when trying to generate topics to publish on in `mqtt.cpp` and `button.cpp`. I personally organize all MQTT devices in a `<building>/<room>/<device>` topic structure, and this could potentially be more than the current 32 max topic length.

Currently, when trying to increase this variable, the following things need to be changed:
- MQTT_MAX_TOPIC_LEN
- lots of buffer allocation code in `mqtt.cpp` and `button.cpp`
- `maxlength` attribute on <input> elements in `settings_sync.html`

It's bug-prone to have to manually adjust the buffer sizes in both cpp files. However it might be nice to have to also automatically populate the `maxlength` attribute on the <input> element. Not sure if this easy though.

### To Reproduce Bug

n/a

### Expected Behavior

n/a

### Install Method

Self-Compiled

### What version of WLED?

0.15.0-b7

### Which microcontroller/board are you seeing the problem on?

Other

### Relevant log/trace output

```shell
n/a
```

### Anything else?

If no one else is one this after a while I'll try to submit a PR myself. Bare with me if I do that, because I've never contributed to open source before!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WouterGritter, 2024-11-20:
Opened a PR for this.

WouterGritter, 2024-11-21:
!4295 uses `MQTT_MAX_TOPIC_LEN` to allocate topic string buffers.

> Currently, when trying to increase this variable, the following things need to be changed:
> - **`maxlength` attribute on `<input>` elements in settings_sync.html**

This was already implemented but the feature was broken. !4296 fixes this.

blazoncek, 2024-11-21:
You can close this now.

WouterGritter, 2024-11-22:
[image] (the one that fixes magic numbers) is still open. Two PRs were spawned from this issue.

closed 2026-09-21 by softhack007