wled/WLED #4300: Color Order Override can not be removed once activated

dosipod, 2024-11-21:
### What happened?

If you enabled Color Order Override then it would work as expected but if you try to remove that from 
/settings/leds page then you no longer can but even worst the page will be populated with so many entries
[image]
In this example i only added two but things went haywire when I tried to delete it  . I am not sure how to solve this without factory reset or editing cfg.json but will do that if nothing to be done from ui 
 

### To Reproduce Bug

 enabled Color Order Override ,save and exist and then try to remove the Color Order Override

### Expected Behavior

 enabled /remove  works as expected 

### Install Method

Self-Compiled

### What version of WLED?

0.15.0-b7 (build 2410270)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
I see nothing on serial
```

### Anything else?

Take it easy as we tested that on multiple units 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2024-11-21:
Quick work around is set "com":[] in cfg.json so thankfully no reset is needed

softhack007, 2024-11-22:
It would be good to understand the root cause of this problem before releasing 0.15.0-RC1
* https://github.com/Aircoookie/WLED/pull/4287

blazoncek, 2024-11-25:
@dosipod can you provide browser's console log if there are any errors in JS?

dosipod, 2024-11-25:
@blazoncek  yes there is an error 
[image]

This happened once I added only one and pressed saved ( the number of overrides will go higher if you press save again even if no changes are done to the page ) 
[image]

dosipod, 2024-11-26:
I just checked other units that do not have color order override and they have the same error so this could be unrelated error

dosipod, 2024-11-26:
Just in case something could be spotted ( I could not see any error ) this is the debug log after fresh serial install with only the override  set and nothing else changed 
[s_debug.txt](https://github.com/user-attachments/files/17913067/s_debug.txt)

This is also the cfg.json which shows the "com" entries after the issue 
  
[cfg.json](https://github.com/user-attachments/files/17913073/cfg.json)

blazoncek, 2024-11-26:
I have the fix ready.

closed 2024-11-28 by netmindz