wled/WLED #4301: Continue reboots using GPIO14 and GPIO 15 in led out after change from AudioReactive digitalmic

chybeat, 2024-11-22:
### What happened?

After add an SK6812 and other WS2812B in WLED settings led preferences ESP reboots.

### To Reproduce Bug

### Installing WLED
1. Install WLED 0.15.0-b7 from https://install.wled.me (using option Erase all user data)
2. Set Wi-Fi (Joined to hidden SSID, so name and password needs to be writen)
3. Visit device.
4. Go to config> Wifi Setup > Set static IP (10.10.1.135) and gateway to router's IP. Click onsave and connect
5. open http://10.10.1.135
6. Go to config > Usermods > and change the AudioReactive's Digital mic pins to my config: SD-22, WS-21, SCK-23
7. Save and type ir browser's address http://10.10.1.135/reset
8. Go to config > Led preferences, then start to configure LED & Hardware setup

### Hardware & setup
1. Disable automatic brightness limiter ( my PSU is 300w, and I setup 234 leds)
2. Change the 1st led ouput and adde5 WS2812B led strip with same lenght: 18 LEDS using GPIOs: 26, 5, 14, 16,17)
 ```
to be more clear: Pin 14 needs to be changed from audioreactive digital mic.
First time I try, fails when save my led settings (GPIO already used),
after save led ouputs (1 to 5), I change GPIO from AudioReactive preferences,
then return to add more led ouputs (8 in total for my config) and can't set any other
led output either rebooting or disconnecting the ESP32.
I need reinstall WLED and change the digital mic pins before set my led ouput preferences.
````
3. added 1 WS2812B led strip with 11 leds length and GPIO 13
4. Added 1 SK6812 les strip with length 112 leds and GPIO 15

After step 5 save settings. and the ligtstrips starts to flash
Watching the console from https://install.wled.me get this: 
```
ets ets Jul 29 2019 12:21:46
rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Ada
abort() was called at PC 0x4008da26 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008ad54:0x3ffbff60 0x4008b0b1:0x3ffbff80 0x4008da26:0x3ffbffa0 0x4008cad1:0x3ffbffd0 0x4008409a:0x3ffbfff0 0x40086291:0x3ffc0020 0x4000d085:0x3ffb1f00 0x40082c77:0x3ffb1f20 0x4011bcef:0x3ffb1f40 0x4011c45a:0x3ffb1f90 0x4013ce31:0x3ffb1fb0 0x4008cb3a:0x3ffb1fd0
Rebooting...
```

### Expected Behavior

In version 0.15.0-b1, I can setup this configuration without problem using the ESP32 and WLED as I describe before. but starting from 0.15.0-b2 the bug is exactly the same.

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0-b7

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
ets ets Jul 29 2019 12:21:46
rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Ada
abort() was called at PC 0x4008da26 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008ad54:0x3ffbff60 0x4008b0b1:0x3ffbff80 0x4008da26:0x3ffbffa0 0x4008cad1:0x3ffbffd0 0x4008409a:0x3ffbfff0 0x40086291:0x3ffc0020 0x4000d085:0x3ffb1f00 0x40082c77:0x3ffb1f20 0x4011bcef:0x3ffb1f40 0x4011c45a:0x3ffb1f90 0x4013ce31:0x3ffb1fb0 0x4008cb3a:0x3ffb1fd0
Rebooting...
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2024-11-22:
I can see the same crash 

```
[12:09:41]Rebooting...
[12:09:41]ets Jun  8 2016 00:22:57
[12:09:41]
[12:09:41]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[12:09:41]configsip: 0, SPIWP:0xee
[12:09:41]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[12:09:41]mode:DOUT, clock div:2
[12:09:41]load:0x3fff0030,len:1184
[12:09:41]load:0x40078000,len:12784
[12:09:41]load:0x40080400,len:3032
[12:09:41]entry 0x400805e4
[12:09:41]
[12:09:41]---WLED 0.15.0-b7 2410270 INIT---
[12:09:41]
[12:09:41]esp32 v3.3.6-16-gcc5440f6a2
[12:09:41]arduino-esp32 v1.0.x
[12:09:41]
[12:09:41]CPU:   ESP32-D0WDQ5 rev.1, 2 core(s), 240 MHz.
[12:09:41]FLASH: 4 MB, Mode 3 (DOUT), speed 40 MHz.
[12:09:41]heap 308332
[12:09:41]JSON buffer allocated: 32767
[12:09:41]TX power: 78/78
[12:09:41]PIN ALLOC: Pin 1 successfully allocated by 0x89.
[12:09:41]PIN ALLOC: 0x00000000000002.
[12:09:41]Registering usermods ...
[12:09:41]heap 275320
[12:09:41]Reading config
[12:09:41]Reading settings from /wsec.json...
[12:09:41]JSON buffer locked. (3)
[12:09:41]JSON buffer released. (3)
[12:09:41]JSON buffer locked. (1)
[12:09:41]Reading settings from /cfg.json...
[12:09:42]Heap before buses: 273004
[12:09:42]Removing all.
[12:09:42]Maximum LEDs on a bus: 112
[12:09:42]Digital buses: 7
[12:09:42]Switching to parallel I2S.
[12:09:42]PIN ALLOC: Pin 26 successfully allocated by 0x82.
[12:09:42]PIN ALLOC: 0x00000004000002.
[12:09:42]Successfully inited strip 0 (len 18) with type 22 and pins 26,255 (itype 23). mA=55/0
[12:09:42]PIN ALLOC: Pin 5 successfully allocated by 0x82.
[12:09:42]PIN ALLOC: 0x00000004000022.
[12:09:42]Successfully inited strip 1 (len 18) with type 22 and pins 5,255 (itype 23). mA=0/0
[12:09:42]PIN ALLOC: Pin 14 successfully allocated by 0x82.
[12:09:42]PIN ALLOC: 0x00000004004022.
[12:09:42]Successfully inited strip 2 (len 18) with type 22 and pins 14,255 (itype 23). mA=0/0
[12:09:42]PIN ALLOC: Pin 16 successfully allocated by 0x82.
[12:09:42]PIN ALLOC: 0x00000004014022.
[12:09:42]Successfully inited strip 3 (len 18) with type 22 and pins 16,255 (itype 23). mA=0/0
[12:09:42]PIN ALLOC: Pin 17 successfully allocated by 0x82.
[12:09:42]PIN ALLOC: 0x00000004034022.
[12:09:42]Successfully inited strip 4 (len 18) with type 22 and pins 17,255 (itype 23). mA=0/0
[12:09:42]PIN ALLOC: Pin 13 successfully allocated by 0x82.
[12:09:42]PIN ALLOC: 0x00000004036022.
[12:09:42]Successfully inited strip 5 (len 11) with type 22 and pins 13,255 (itype 23). mA=0/0
[12:09:42]PIN ALLOC: Pin 15 successfully allocated by 0x82.
[12:09:42]PIN ALLOC: 0x0000000403E022.
[12:09:42]Successfully inited strip 6 (len 112) with type 30 and pins 15,255 (itype 27). mA=0/0
[12:09:42]LED buffer size: 11200B
[12:09:42]Heap after buses: 267760
[12:09:42]Starting usermod config.
[12:09:42]JSON buffer released. (1)
[12:09:42]heap 267716
[12:09:42]Initializing strip
[12:09:42]Loading custom palettes
[12:09:42]Loading custom ledmaps
[12:09:42]heap 267512
[12:09:42]Usermods setup
[12:09:42]AR: Failed to initialize sound input driver. Please check input PIN settings.
[12:09:42]heap 267292
[12:09:42]Defaulf WiFi used.
[12:09:42]Ada
[12:09:42]initServer
[12:09:42]heap 212644
[12:09:42]initIR
[12:09:42]heap 212644
[12:09:42]Initial connect or forced reconnect.
[12:09:42]Defaulf WiFi used.
[12:09:42]initConnection() called.
[12:09:42]Access point disabled (init).
[12:09:42]Connecting to toi...
[12:09:42]abort() was called at PC 0x4008dc0a on core 0
[12:09:42]
[12:09:43]ELF file SHA256: 0000000000000000
[12:09:43]
[12:09:43]Backtrace: 0x4008af38:0x3ffb5e80 0x4008b295:0x3ffb5ea0 0x4008dc0a:0x3ffb5ec0 0x4008ca58:0x3ffb5ef0 0x4017291e:0x3ffb5f30 0x40164142:0x3ffb5f60 0x4015da88:0x3ffb5f80 0x4008cd1e:0x3ffb5fb0
[12:09:43]
[12:09:43]Rebooting...
```

The error related to "AR: Failed to initialize sound input driver. Please check input PIN settings. " is there all the time  even if the usermod is not active and pins are unused

chybeat, 2024-11-22:
Thanks for your complete log @dosipod. I think was a problem with my board, and amazon works all night... so, I bought 2 more ESP32 boards thinking mine is broken 😅.... Your log is much better and complete than mine. So Thanks for take time to reproduce the error and share the information.

softhack007, 2024-11-22:
I could reproduce the crash, but its still very mysterious to me what's going on.
In fact it has something to do with adding the SK6812 led, because it was crashing right afterwards.

to reproduce, its enough to add LEDs in preferences
1. AR mic pins : SD-22, WS-21, SCK-23 (could be unrelated)
2. create 5 WS2812B led strips with same lenght: 18 LEDS, using GPIOs: 26, 5, 14, 16,17. Save LEDs preferences.
3. re-enter LEDs page, add S2812B led strip with 11 leds length and GPIO 13. Save.
4. re-enter LEDs page, add SK6812 led strip with length 112 leds and GPIO 15, Save.
--> bootloop.

```gdb
initServer
heap 208700
initIR
heap 208700
Initial connect or forced reconnect.
Defaulf WiFi used.
initConnection() called.
Access point disabled (init).
Connecting to WLAN2...
abort() was called at PC 0x4008da92 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008adc0:0x3ffbff70 0x4008b11d:0x3ffbff90 0x4008da92:0x3ffbffb0 0x4008cb3d:0x3ffbffe0 0x40084106:0x3ffc0000 0x400862fd:0x3ffc0030 0x4012ce6f:0x3ffb1df0 0x40114799:0x3ffb1e20 0x4011d2ed:0x3ffb1e50 0x4011d57c:0x3ffb1f10 0x4011e05a:0x3ffb1f90 0x4013ed05:0x3ffb1fb0 0x4008cba6:0x3ffb1fd0
  #0  0x4008adc0:0x3ffbff70 in invoke_abort at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:648
  #1  0x4008b11d:0x3ffbff90 in abort at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:648
  #2  0x4008da92:0x3ffbffb0 in vPortCPUAcquireMutexIntsDisabledInternal at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
      (inlined by) vPortCPUAcquireMutexIntsDisabled at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.h:98
      (inlined by) vTaskEnterCritical at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4201
  #3  0x4008cb3d:0x3ffbffe0 in xQueueIsQueueFullFromISR at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:2041
  #4  0x40084106:0x3ffc0000 in i2s_intr_handler_default at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/driver/i2s.c:1330
  #5  0x400862fd:0x3ffc0030 in _xt_lowint1 at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #6  0x4012ce6f:0x3ffb1df0 in WiFiSTAClass::localIP() at C:/Users/frank/.platformio/packages/framework-arduinoespressif32@src-e9b1fbd6563a55e19ddae15e1fc09589/libraries/WiFi/src/WiFiSTA.cpp:721
  #7  0x40114799:0x3ffb1e20 in NetworkClass::isConnected() at wled00/src/dependencies/network/Network.cpp:89
  #8  0x4011d2ed:0x3ffb1e50 in WLED::handleConnection() at wled00/bus_manager.h:122
  #9  0x4011d57c:0x3ffb1f10 in WLED::loop() at wled00/bus_manager.h:122
  #10 0x4011e05a:0x3ffb1f90 in loop() at wled00/wled_main.cpp:23
  #11 0x4013ed05:0x3ffb1fb0 in loopTask(void*) at C:/Users/frank/.platformio/packages/framework-arduinoespressif32@src-e9b1fbd6563a55e19ddae15e1fc09589/cores/esp32/main.cpp:23
  #12 0x4008cba6:0x3ffb1fd0 in vPortTaskWrapper at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jul 29 2019 12:21:46
```

after that we have a bootloop, with several different crash locations
```gdb
Defaulf WiFi used.
initConnection() called.
Access point disabled (init).
Connecting to WLAN2...
abort() was called at PC 0x4008da92 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008adc0:0x3ffbff70 0x4008b11d:0x3ffbff90 0x4008da92:0x3ffbffb0 0x4008cb3d:0x3ffbffe0 0x40084106:0x3ffc0000 0x400862fd:0x3ffc0030 0x401055af:0x3ffb1ef0 0x4011d588:0x3ffb1f10 0x4011e05a:0x3ffb1f90 0x4013ed05:0x3ffb1fb0 0x4008cba6:0x3ffb1fd0
  #0  0x4008adc0:0x3ffbff70 in invoke_abort at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:648
  #1  0x4008b11d:0x3ffbff90 in abort at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:648
  #2  0x4008da92:0x3ffbffb0 in vPortCPUAcquireMutexIntsDisabledInternal at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
      (inlined by) vPortCPUAcquireMutexIntsDisabled at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.h:98
      (inlined by) vTaskEnterCritical at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4201
  #3  0x4008cb3d:0x3ffbffe0 in xQueueIsQueueFullFromISR at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:2041
  #4  0x40084106:0x3ffc0000 in i2s_intr_handler_default at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/driver/i2s.c:1330
  #5  0x400862fd:0x3ffc0030 in _xt_lowint1 at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #6  0x401055af:0x3ffb1ef0 in handleTransitions() at wled00/led.cpp:181
  #7  0x4011d588:0x3ffb1f10 in WLED::loop() at wled00/bus_manager.h:122
  #8  0x4011e05a:0x3ffb1f90 in loop() at wled00/wled_main.cpp:23
  #9  0x4013ed05:0x3ffb1fb0 in loopTask(void*) at C:/Users/frank/.platformio/packages/framework-arduinoespressif32@src-e9b1fbd6563a55e19ddae15e1fc09589/cores/esp32/main.cpp:23
  #10 0x4008cba6:0x3ffb1fd0 in vPortTaskWrapper at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```
[... 2 lines cut ...]
Access point disabled (init).
Connecting to WLAN2...
abort() was called at PC 0x4008da92 on core 1

ELF file SHA256: 0000000000000000

[... cut ...]
  #8  0x40082cd7:0x3ffb1ef0 in millis at C:/Users/frank/.platformio/packages/framework-arduinoespressif32@src-e9b1fbd6563a55e19ddae15e1fc09589/cores/esp32/esp32-hal-misc.c:172
  #9  0x4011d6e3:0x3ffb1f10 in WLED::loop() at wled00/bus_manager.h:122
  #10 0x4011e05a:0x3ffb1f90 in loop() at wled00/wled_main.cpp:23
  #11 0x4013ed05:0x3ffb1fb0 in loopTask(void*) at C:/Users/frank/.platformio/packages/framework-arduinoespressif32@src-e9b1fbd6563a55e19ddae15e1fc09589/cores/esp32/main.cpp:23
  #12 0x4008cba6:0x3ffb1fd0 in vPortTaskWrapper at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

```

softhack007, 2024-11-22:
It would be good to understand the root cause of this problem before releasing 0.15.0-RC1
* https://github.com/Aircoookie/WLED/pull/4287

dosipod, 2024-11-22:
@softhack007 We could try more scenarios if that would help , just one thing that might help in this and in other cases how did you get the Backtrace details in your post , could you make an environment available as an example

softhack007, 2024-11-22:
> just one thing that might help in this and in other cases how did you get the Backtrace details in your post , could you make an environment available as an example

Hi, the backtrace comes from "esp exception decoder". It only works when you build _and_ upload in VSCode, as exception decoder needs information from the compilation process.

You only need this in your buildenv
```ini
monitor_filters = esp32_exception_decoder
```

In VSCode

0) pick the buildenv you want, like `esp32dev` 
1) click on "build and upload" (right arrow icon)
2) open serial monitor (plug icon)

If WLED crashes, you should see the function backtrace
The backtrace is not always correct, but usually a good start for analysis.

dosipod, 2024-11-22:
I do not have access to  VSC  because of big brother   but only compile on github  with both debug flag and exception_decoder but can not get the detailed Backtrace but only the same as in my post . Was hoping that is possible online

dosipod, 2024-11-23:
> @dosipod @chybeat please check if the crash also happens if you configure all outputs as ws2812b
> 
> It might be that the bug is related to having one "sk6812" (RGBW) output among all other being ws2812b.

@softhack007 I was able to reproduce the crash without adding SK6812 so it also happens with all the output set to ws2812b

```
[08:33:13]Rebooting...
[08:33:13]ets Jun  8 2016 00:22:57
[08:33:13]
[08:33:13]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[08:33:13]configsip: 0, SPIWP:0xee
[08:33:13]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[08:33:13]mode:DOUT, clock div:2
[08:33:13]load:0x3fff0030,len:1184
[08:33:13]load:0x40078000,len:12784
[08:33:13]load:0x40080400,len:3032
[08:33:13]entry 0x400805e4
[08:33:14]
[08:33:14]---WLED 0.15.0-b7 2410270 INIT---
[08:33:14]
[08:33:14]esp32 v3.3.6-16-gcc5440f6a2
[08:33:14]arduino-esp32 v1.0.x
[08:33:14]
[08:33:14]CPU:   ESP32-D0WDQ5 rev.1, 2 core(s), 240 MHz.
[08:33:14]FLASH: 4 MB, Mode 3 (DOUT), speed 40 MHz.
[08:33:14]heap 308980
[08:33:14]JSON buffer allocated: 32767
[08:33:14]TX power: 78/78
[08:33:14]PIN ALLOC: Pin 1 successfully allocated by 0x89.
[08:33:14]PIN ALLOC: 0x00000000000002.
[08:33:14]Registering usermods ...
[08:33:14]heap 275968
[08:33:14]Reading config
[08:33:14]Reading settings from /wsec.json...
[08:33:14]JSON buffer locked. (3)
[08:33:14]JSON buffer released. (3)
[08:33:14]JSON buffer locked. (1)
[08:33:14]Reading settings from /cfg.json...
[08:33:14]Heap before buses: 273628
[08:33:14]Removing all.
[08:33:14]Maximum LEDs on a bus: 112
[08:33:14]Digital buses: 7
[08:33:14]Switching to parallel I2S.
[08:33:14]PIN ALLOC: Pin 26 successfully allocated by 0x82.
[08:33:14]PIN ALLOC: 0x00000004000002.
[08:33:14]Successfully inited strip 0 (len 18) with type 22 and pins 26,255 (itype 23). mA=55/0
[08:33:14]PIN ALLOC: Pin 5 successfully allocated by 0x82.
[08:33:14]PIN ALLOC: 0x00000004000022.
[08:33:14]Successfully inited strip 1 (len 18) with type 22 and pins 5,255 (itype 23). mA=0/0
[08:33:14]PIN ALLOC: Pin 14 successfully allocated by 0x82.
[08:33:14]PIN ALLOC: 0x00000004004022.
[08:33:14]Successfully inited strip 2 (len 18) with type 22 and pins 14,255 (itype 23). mA=0/0
[08:33:14]PIN ALLOC: Pin 16 successfully allocated by 0x82.
[08:33:14]PIN ALLOC: 0x00000004014022.
[08:33:14]Successfully inited strip 3 (len 18) with type 22 and pins 16,255 (itype 23). mA=0/0
[08:33:14]PIN ALLOC: Pin 17 successfully allocated by 0x82.
[08:33:14]PIN ALLOC: 0x00000004034022.
[08:33:14]Successfully inited strip 4 (len 18) with type 22 and pins 17,255 (itype 23). mA=0/0
[08:33:14]PIN ALLOC: Pin 13 successfully allocated by 0x82.
[08:33:14]PIN ALLOC: 0x00000004036022.
[08:33:14]Successfully inited strip 5 (len 18) with type 22 and pins 13,255 (itype 23). mA=0/0
[08:33:14]PIN ALLOC: Pin 15 successfully allocated by 0x82.
[08:33:14]PIN ALLOC: 0x0000000403E022.
[08:33:14]Successfully inited strip 6 (len 112) with type 22 and pins 15,255 (itype 23). mA=0/0
[08:33:14]LED buffer size: 8400B
[08:33:14]Heap after buses: 268568
[08:33:14]Starting usermod config.
[08:33:14]JSON buffer released. (1)
[08:33:14]heap 268524
[08:33:14]Initializing strip
[08:33:14]JSON buffer locked. (21)
[08:33:14]JSON buffer released. (21)
[08:33:14]Loading custom palettes
[08:33:14]Loading custom ledmaps
[08:33:14]Matrix ledmap:
[08:33:14]   0,  31,  32,  63,  64,  95,  96, 127, 128, 159, 160, 191, 192, 223, 224, 255,
[08:33:14]   1,  30,  33,  62,  65,  94,  97, 126, 129, 158, 161, 190, 193, 222, 225, 254,
[08:33:14]   2,  29,  34,  61,  66,  93,  98, 125, 130, 157, 162, 189, 194, 221, 226, 253,
[08:33:14]   3,  28,  35,  60,  67,  92,  99, 124, 131, 156, 163, 188, 195, 220, 227, 252,
[08:33:14]   4,  27,  36,  59,  68,  91, 100, 123, 132, 155, 164, 187, 196, 219, 228, 251,
[08:33:14]   5,  26,  37,  58,  69,  90, 101, 122, 133, 154, 165, 186, 197, 218, 229, 250,
[08:33:14]   6,  25,  38,  57,  70,  89, 102, 121, 134, 153, 166, 185, 198, 217, 230, 249,
[08:33:14]   7,  24,  39,  56,  71,  88, 103, 120, 135, 152, 167, 184, 199, 216, 231, 248,
[08:33:14]   8,  23,  40,  55,  72,  87, 104, 119, 136, 151, 168, 183, 200, 215, 232, 247,
[08:33:14]   9,  22,  41,  54,  73,  86, 105, 118, 137, 150, 169, 182, 201, 214, 233, 246,
[08:33:14]  10,  21,  42,  53,  74,  85, 106, 117, 138, 149, 170, 181, 202, 213, 234, 245,
[08:33:14]  11,  20,  43,  52,  75,  84, 107, 116, 139, 148, 171, 180, 203, 212, 235, 244,
[08:33:14]  12,  19,  44,  51,  76,  83, 108, 115, 140, 147, 172, 179, 204, 211, 236, 243,
[08:33:14]  13,  18,  45,  50,  77,  82, 109, 114, 141, 146, 173, 178, 205, 210, 237, 242,
[08:33:14]  14,  17,  46,  49,  78,  81, 110, 113, 142, 145, 174, 177, 206, 209, 238, 241,
[08:33:14]  15,  16,  47,  48,  79,  80, 111, 112, 143, 144, 175, 176, 207, 208, 239, 240,
[08:33:14]heap 267760
[08:33:14]Usermods setup
[08:33:15]AR: Failed to initialize sound input driver. Please check input PIN settings.
[08:33:15]heap 267540
[08:33:15]Defaulf WiFi used.
[08:33:15]Ada
[08:33:15]initServer
[08:33:15]heap 212984
[08:33:15]initIR
[08:33:15]heap 212984
[08:33:16]Initial connect or forced reconnect.
[08:33:16]Defaulf WiFi used.
[08:33:16]initConnection() called.
[08:33:16]abort() was called at PC 0x4008da92 on core 0
[08:33:16]
[08:33:16]ELF file SHA256: 0000000000000000
[08:33:16]
[08:33:16]Backtrace: 0x4008adc0:0x3ffafb60 0x4008b11d:0x3ffafb80 0x4008da92:0x3ffafba0 0x4008c571:0x3ffafbd0 0x40159111:0x3ffafc10 0x40181f3c:0x3ffafcb0 0x4018a327:0x3ffafcd0 0x4018b099:0x3ffafd20 0x40188a5e:0x3ffafd40 0x401ac936:0x3ffafd60 0x4008cba6:0x3ffafd90
[08:33:16]
[08:33:16]Rebooting...
```

I did not test on older builds but since the OP mentioned it is working  on 0.15.0-b1  but not on 0.15.0-b2 , might be that could 
isolated the changes to either led preference page or good forbid the audio usermod

chybeat, 2024-11-23:
Just and idea. @softhack007. Try to watch the saved cfg.json after crash (if possible), I remember time ago when implemented the brightness limiter, when saving does nothing, the google inspector says something like 'not possible to point over input area' , with unmarked brightness limiter checkbox and points to a 'required' element in ouputs. This maybe saves wrong configuration, may be happens in global saves as while saving (for fisrt time) on any settings pages, and maybe is not saving o reading GPIO config for Audioreactive, when use GPIO. Again, is just an idea :)

willmmiles, 2024-11-23:
I've replicated this on my test setup; it looks like a buffer overflow of some kind when using the parallel I2S output mode.

dosipod, 2024-11-23:
Not sure it is this but I see this note in the release page  but not link to PR and could not fine the commit for it 
[image]
When adding 6 digital you will start to see in the debug log " Switching to parallel I2S. " and once you add the 7th it would start to loop .

willmmiles, 2024-11-23:
OK, found it.  For parallel I2S, NeoPixelBus requires that all buses be constructed before *any* bus calls Begin().   Working a fix now.

netmindz, 2024-11-25:
Can we close this as fixed the @willmmiles ?

dosipod, 2024-11-26:
The bootloop is fixed so for sure this is better then current situation and you already merged that so this is after the fact question 

But the 10 buses limit is not clear ( I know that was increased recently as I see in const.h ) but if you try to add more then 10 and you reboot then  it will go back to 10 buses only .

The behavior before was that you can not add more then 10 digital buses as after 10 the led output will be grayed same as below
[image]

I see also he opened another issue here  https://github.com/Aircoookie/WLED/issues/4315 so maybe settings/leds page is not
100 % ready and it is very critical page

netmindz, 2024-11-26:
Please log a separate issue for the limit of 10 @dosipod

dosipod, 2024-11-26:
@netmindz  Sure but the question  was is the PR that fixed the crash responsible for breaking the 10 digital buses limit  or was there another change that broke it as I have hard time tracking old PRs due to strange naming and combing multiple fixes in the same PR  . It may also be better to engorge  testing , feedback and  issue reporting  is not to close PRs that someone toke the time to test and open unless they confirmed . Keep it open  for a while marked as fixed in source  as a courtesy to the OP @chybeat

willmmiles, 2024-11-26:
#4312 made no changes to the bus settings logic -- it only changed the call order for bus hardware initialization.  I think we can close this issue; problems with the settings page are unrelated.

closed 2024-11-26 by willmmiles

dosipod, 2024-11-27:
This is not the proper way to close raised cases but with that I assume you have tested with real leds on 10 outputs as I do not have the setup to test that at the moment

willmmiles, 2024-11-27:
This issue entry covers what the title says and first post state and no more: a bootloop if more than 6 entries are added.  This problem has been fixed; the updated software no longer crashes and enters a boot loop.  Hence we have closed this issue as there is no more work required on it.

If there's some other problem with 10 outputs, that is a different and unrelated issue, and should be raised as a new entry as @netmindz said earlier.  As I see things, issue entries aren't support cases - each entry should cover one specific software problem; they aren't meant to be an ongoing conversation with a specific user until all their problems are solved.

PR #4312 (2024-11-24): Defer calling begin() on buses
NeoPixelBus requires that all parallel I2S bus members be constructed before any of them call Begin().  Implement this by deferring the Begin() call to the end of the bus initialization process.  This replaces the "reinit" logic previously used to work around a similar initialization ordering issue for ESP8266 pins.

Fixes #4301.

This PR introduces a new virtual begin() call on Bus types, although it's only currently used on digital buses.  This seemed like a cleaner abstraction than requiring a bunch of type-checking logic at the point of use, and maps more clo [... cut]
commit d53d7aa2 (2024-11-24): Defer calling begin() on digital buses

NeoPixelBus requires that all parallel I2S bus members be constructed
before any of them call Begin().  Implement this by deferring the
call to the end of bus construction.

Fixes #4301.