wled/WLED #4306: WT32-ETH01 v1.4 No Output Signal

lplassman, 2024-11-22:
### What happened?

Any connected addressable LED strip does nothing when using WLED on this board. When reading any set LED output pin with a multimeter, there is no voltage. The board boots up successfully and creates a WiFi access point, as well as successfully connects to Ethernet.

### To Reproduce Bug

Flash current firmware (v14.4) to a WT32-ETH01 board. Change output pin to GPIO2, 4, 12, or 14. Observe no signal output on any of the pins set. If the LED output mode is set to "On/Off" instead of WS281X, the pin responds as it should and outputs 3.3v when WLED is on, and 0v when WLED is off. The problem seems to be software related with this particular board revision. I have tested multiple v1.4 boards with no success with any of them. I have also tested with a v1.2 board and everything works properly.

### Expected Behavior

The LED output pin should output a signal

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.4

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netmindz, 2024-11-25:
You can't test a data pin with a multimeter you need a oscilloscope or logic analyser, that's why you only see the output when you are setting as just on/off

lplassman, 2024-11-25:
You are right. I tested with GPIO4 with a new LED strip and everything works as it should. Thanks!

closed 2024-11-25 by lplassman