wled/WLED #4310: GPIO 32 can not be used

ProjectProcrastinator, 2024-11-23:
### What happened?

I just updated to 0.15.0-b7 and now if I want to save my LED settings it says: "Sorry, pins [6,7,8,9,10,11,16,17,24,28,29,30,31,15,14,32,15,14] can't be used."

I use GPIO 32 to control an analog LED channel on my custom board using ESP32-Mini-1. It worked flawless before the update and even after the update GPIO 32 is still set for my CCT-Strip and it works. But in LED settings the 32 is red and i can not save. I don't see a reason to ban GPIO 32 from usage.

### To Reproduce Bug

Update from 0.14.2 to 0.15.0-b7

### Expected Behavior

Just let me save.

### Install Method

Self-Compiled

### What version of WLED?

0.15.0-b7

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
Sorry, pins [6,7,8,9,10,11,16,17,24,28,29,30,31,15,14,32,15,14] can't be used.
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ProjectProcrastinator, 2024-11-23:
Ok, I just found out it was because GPIO 32 is automatically set as I2S-Pin for Digitalmic. Clears my issue but nonetheless it's irritating that these pins are set and therefore reserved if I never used AudioReactive. 

Would be better if these pins would just be -1 until one activates the feature.

softhack007, 2024-11-24:
Hi, 

Yes your explanation is correct - all pins are allocated at startup. It was a design decision to also apply this rule to usermods (like audioreactive), even when the usermod is not active.