wled/WLED #4313: When using analog quantity to control wled speed, the speed cannot change smoothly. Why?

czweb, 2024-11-24:
### What happened?

When using analog quantity to control wled speed, the speed cannot change smoothly. Why?
[image]

Video：
https://www.youtube.com/shorts/9PZYwQT0Qoc
[image]

### To Reproduce Bug

using analog

### Expected Behavior

Just like stepping on the accelerator, it feels like stepless speed regulation and should not jump

### Install Method

Binary from WLED.me

### What version of WLED?

>=WLED 0.13.4 

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

czweb, 2024-11-24:
The speed will also jump when you use wled web page to adjust the speed, or control the speed through node-red or HomeAssistant. The jump looks like starting the playback from the beginning, and it cannot be infinitely changed like driving a car.

DedeHai, 2024-11-24:
Smooth speed transitions are not implemened. Also it depends on how the FX handles speed changes, different effects can react differently to a speed change

softhack007, 2024-11-24:
We have almost 200 effects and all of them handle their progress calculations differently. Some use `time = strip.now / SEGENV.speed`, which makes them jump when you change the speed slider. It has always been like this. 

We would need to comb through more than 200 effects, and each effect would needs it's own special solution. Some effect even use "speed" with a completely different meaning - like Lissajous (phase angle), GEQ (amount of fade), etc.

I don't see when/how we could find the time and people to address this topic. If you have a special effect in mind and you know how to change the effect source code, please feel free to contribute via pull request (PR).

czweb, 2024-11-24:
> Smooth speed transitions are not implemened. Also it depends on how the FX handles speed changes, different effects can react differently to a speed change

Thank you. It turns out that different effects will react differently to speed changes.
Not smooth: Chase, Lighthouse, Loading, TWODots
Smooth: Finally chose "meteor smooth"