wled/WLED #4315: Fix mixed timings in parallel I2S

willmmiles, 2024-11-24:
When:
- On ESP32
- Using more than 5 digital output buses, so the system switches to parallel I2S
- Buses of different timings are selected

Only the timing of the first digital bus will be used, resulting in incorrect timings for later buses.

The automatic switchover to using parallel I2S should be aware of the individual bus type timings, and fall back to using RMT channels when buses of different timings are selected.  Otherwise incorrect timings may be used.

Reference:

              > @Makuna (hope your reading this) we are still not 100% sure if mixing different led strip types on the same "parallel I2S bus" is supported (esp32), or it may lead to errors and spurious crashes? Our most important use case would be to have both ws2812b (RGB) and sk6812 (RGBW) on the same parallel I2S driver.
> 
> It this "safe", or would you say better to use the parallel bus for only one type, and have single RMT busses for other types?

The "Feature" can be unique per parallel channel without any concern.  So, mixing RGB with RGBW or even 8 bit per element and 16 bit per element isn't an issue.
The "Method" uses the i2s bus speed of the first NPB instanced (constructed) on that i2s bus as the speed for all channels that are latter constructed on that i2s bus.  The reset time can be unique per i2s parallel channels without issues.  The ramification of this is that not all methods use the same speed.  In the NPB file "NeoBits.h" is the list and the key class member is "BitSendTimeNs".  You can see that WS2805 has a different value than the WS2812x, but SK6812 has the same as WS2812x.  So, if the speed is the same, there is no issue at all.  If the speed is different, the only effect is that it will run at the speed of the first one instanced.  Should not crash or cause other issues than timing.
NOW, with this all stated, the big caveat is that all parallel channels must be instanced (constructed) before the first Begin() is called and no other are instanced after the first Begin() is called.  That could cause issues.

_Originally posted by @Makuna in https://github.com/Aircoookie/WLED/issues/4312#issuecomment-2496030173_

softhack007, 2024-11-25:
🤔 Thinking about short-term fixing, it would be "save" to only allow ws2812b and sk6812 in the parallel bus driver, and fall back to RMT for any other type. 

This would prevent timing issues with the parallel bus, and it would give us some more time to implement a better solution   after 0.15 is released.

willmmiles, 2024-11-25:
Hm, that would at least be no worse than 0.14.  That said, I don't consider this issue a blocker for 0.15; I think it's ok to file under "known issues" for the moment.  I'd just wanted to make a note of it so we can track it.