wled/WLED #4318: Presets are not working

Djmg2k, 2024-11-25:
### What happened?

My presets disappeared, I tried to save new ones and I get this message when trying to save new presets: "sorry, there was an issue loading your presets!". 
[image]

### To Reproduce Bug

I think I explained it in the previous box. 

### Expected Behavior

I expected all of my presets to be safe and not disappear and also be able to keep saving new presets. 

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.4 (build 2405180)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netmindz, 2024-11-25:
Just copy the details into a text file and restore

netmindz, 2024-11-25:
Possibly related to https://github.com/Aircoookie/WLED/issues/3762

dosipod, 2024-11-25:
In all the cases we have seen on latest and we are able to replicate multiple times  the presets issues are related to external control ( HA , Node-red, Alexa and  wizmote )  as accessing unit presets  from multiple sources would cause the issue  . 
Even if you fix the  issue it may come back if you try to edit it again when those sources are still using it .

DedeHai, 2024-11-25:
Is this a problem that was introduced in 0.15 and needs to be fixed before the release?
sounds like a nasty bug: the code really should not corrupt its own files. If it can't be fixed due to some timing restrictions, at least it should create a backup that can be restored if its corrupted (as long as there is space in the FS).

netmindz, 2024-11-25:
#3762 was back on 0.14, so that one isn't a regression, not sure if this is the same or not

Djmg2k, 2024-11-25:
@netmindz  I will try to do that when I get home, though I use the mobile version of wled so not sure if I'm able to do that on there?  I'm new to wled but it was working fine and was able to save new presets and set them to load at boot without a problem.  I had 10 presets, they disappeared and that's when I wasn't able to save new ones.  
I copied and saved the file text from the last working state to my notes but couldn't figure out how to restore using that file.

blazoncek, 2024-11-27:
Classic ESP32 is prone to flash corruption due to old and flawed bootloader and framework used (as old as 2021) as pointed out by @Jason2866 on some other issues. Unfortunately newer framework/platform will require updating bootloader (mandatory reflash). The other issue is huge increase in binary size with newer framework.

There were several issues opened on this topic: #3780 #3387 #3382 #3347 #3181 #3139 #2793 #2755 #2434 #1981 and possibly others.

netmindz, 2024-11-27:
Thanks for context @blazoncek

Djmg2k, 2024-11-28:
Thank you guys for the help, presets are working again.

dosipod, 2024-11-30:
Here you go , issue replication on demand using HA
[image]

softhack007, 2024-12-05:
If presets were touched by an "external entity" like HA, then this problem might be related to
* #4348

closed 2025-04-12 by github-actions[bot]

blazoncek, 2025-12-29:
Anyone stumbling on this: Presets on ESP32 get corrupted (99.999%) on save by incompatible flash mode used when flashing ESP for the first time.
To resolve the issue *you will need to flash new bootloader using DIO or DOUT mode*.