wled/WLED #4320: Unresponsible API/web UI during receiving DDP

DeKaN, 2024-11-25:
### What happened?

Sometimes WLED becomes unresponsive (API and web UI don't work) and plays effects with lags after enabling streaming effects from LedFX via DDP. If I stop LedFx streaming, the issue disappears and I can use API (for switching between built-in effects for example).
The issue isn't reproducible after rebooting ESP - everything works for at least 2 hours and appears only after stopping and starting again streaming from LedFx.

I had version 0.15.0-b7 and started downgrading WLED step by step. I haven't seen the issue after installing 0.15.0-b4 yesterday

"Melt" effect from LedFx during the issue: https://github.com/user-attachments/assets/42c14a6c-d08a-4a31-8416-1460f1e8933a

### To Reproduce Bug

Install 0.15.0-b5 and newer on ESP32, run LedFX 2.0.105 (as Home Assistant addon or as a separate app) start streaming, stop and start again after some time

### Expected Behavior

WLED API and web UI should be accessible, and effects via DDP should be played without lags

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0-b5 and newer

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

My [ESP32](https://quinled.info/pre-assembled-quinled-dig-uno/)
WLED presets
```
{"0":{},"1":{"win":"T=0","n":"off"},"2":{"on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":267,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"col":[[0,255,0],[0,0,0],[0,0,0]],"fx":41,"sx":128,"ix":128,"pal":1,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"id":1,"start":267,"stop":400,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"set":0,"col":[[0,255,0],[0,0,0],[0,0,0]],"fx":41,"sx":128,"ix":128,"pal":1,"c1":128,"c2":128,"c3":16,"sel":true,"rev":true,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}],"n":"default"}}
```

I use Home Assistant to start/stop streaming from LedFx and switch built-in effects:
- WLED preset != "off" and music is paused -> HA switches between 135 built-in effects, LedFx streaming paused
- WLED preset != "off" and music is playing -> HA doesn't try to switch effects, LedFx streaming started
- WLED preset == "off" -> HA doesn't change effects and  LedFx streaming paused

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-11-25:
Hi, 

How many LEDs do you drive from LedFX?

External control via DDP needs a lot of network bandwidth and processing power on esp32 side. So knowing your LEDs setup is important to understand the problem - please also post your cfg.json, and LedFX effect config.

DeKaN, 2024-11-26:
Hi @softhack007, I have 400 LEDs connected to the board. I've added WLED config and LedFx config.

bigredfrog, 2024-11-26:
It's worth noting this is reported as an apparent regression between versions of WLED. Starting with 0.15.0-b5

Personally I benchmark on 0.14.x for ledfx development and I have seen no implication of cumulative issues like this one ( appearing after significant runtime ). I have graphs of the sustainable FPS, but in that testing I go as high as 1600 pixels ( spread across 4 outputs to avoid 800khz issues ).

DeKaN, 2024-12-29:
Still reproducible with 0.15.0

lplassman, 2025-01-02:
I am also experiencing the same issue with all my WLED devices (I have around 30). It doesn't seem to matter the chip version (ESP8266, ESP32), or if it is connected using Ethernet or not.

bigredfrog, 2025-01-06:
Continue to see slightly slower FPS performance for 15.x from boot vs 14.x

Also Ledfx users reporting instability in 15.x with disconnects, or slow downs within 24 hrs.

Pretty much A/B testable 14.x vs 15.x

14.x remains close to bullet proof under heavy testing, 15.x is just unstable, and we will need to advise any users hitting performance issues to downgrade WLED

ironcrafter54, 2025-01-24:
I also experience this exact same problem solution above works for me.

netmindz, 2025-01-25:
When you say the solution, do you just mean the work around to downgrade @ironcrafter54 ?

ironcrafter54, 2025-01-28:
Yes, @netmindz I just downgraded, 15.x still causes issues but I found 14.x to work normally

JohnnyLeone, 2025-03-21:
With new beta version 0.15.1.beta2 the error no longer appears for me. Can someone else verify it too?

closed 2025-07-27 by github-actions[bot]