wled/WLED #4325: Issue when trying to add more then 10 digital buses

dosipod, 2024-11-26:
### What happened?

The 10 buses limit is not clear , It was increased recently as I see in const.h  but if you try to add more then 10 and you reboot then  it will go back to 10 buses only .

The behavior before was that you can not add more then 10 digital buses as after 10 the led output will be grayed same as below
[image]

### To Reproduce Bug

Add more then 10 digital buses

### Expected Behavior

Exceeding the limit for digital buses should not be allowed in UI

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.15.0-b7 (build 2410270)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2024-11-26:
You could change the label as "needs investigation"  and remove bug as I could not do that .

dosipod, 2024-11-26:
One thing more noticed is that this limit issue is also there when I tried to test with DDP output , in this case if you save the page with more then 10 then settings/leds will bug and all output will be gone ( strange thing is the master still control the slave even if the setting page is empty ) , rebooting does not help and factory reset might the only option as editing cfg.json might be too much effort 

[image]

dosipod, 2024-11-27:
Just FYI , I do not need 10 outputs with leds nor with DDP . I am happy with 2 and the max I have is 4 output for a unit I made only for testing . But we have received  some questions from users  with regard to using more then 10 outputs with DDP as they wanted that for a long time  so this is not unrealistic setups if you care to check discord  .

tech1977, 2024-11-27:
Hi, I would like to try DDP more than 10 Athom GU10 WLED bulbs, currently I have 18 bulbs split into 2 ESP32 ethernet controllers and another master controller, if I can try everything with just one controller it would be great, I would like to be able to eliminate the delay to have smooth effects like wipe and another effects;
will it be possible to add more than 10 DDP from the next version of WLED? I am very interested in adding many GU10 bulbs and I am available for all tests;
thank you!!!

dosipod, 2024-12-14:
@tech1977 ,As the output of more then 10 could be edge case so might be not solved ,for DDP if you need more then 10 output then we have found a simple workaround , in one of the slaves just also use DDP to other slaves , you can see details in discord here https://discord.com/channels/473448917040758787/473448917543944193/1317398474626433024 , if you do that then you can still from one master control the end slaves ( nested DDP ) with segments on the master so that is 32 segments on esp32 .

PaoloTK, 2025-01-13:
It would be nice to have this bug fixed for an upcoming QuinLED board that has 15 analog outputs. Is anyone working on this and/or has pointers on how to go about fixing this? It sounds like the issue is with the settings page. I have yet to test actual output, but configuring 11 outputs at compile seems to at least work correctly through debug.

dosipod, 2025-01-13:
@PaoloTK  Sure it is a work in progress , look at those two 
https://github.com/Aircoookie/WLED/issues/4389
https://github.com/Aircoookie/WLED/issues/4380

Or better get in touch with the guys on QuinLED  discord for updates and workarounds .
We may close this issue or keep it for DDP related as we do not wish to have so may issues related to the same thing .

PaoloTK, 2025-01-13:
> @PaoloTK Sure it is a work in progress , look at those two #4389 #4380
I don't think those are affected by the problem reported in this issue? Or is it a more general bug that breaks both certain outputs when used with specific GPIOs due to hardware and all outputs if more than 10 are configured?

> Or better get in touch with the guys on QuinLED discord for updates and workarounds . We may close this issue or keep it for DDP related as we do not wish to have so many issues related to the same thing .
Sorry, my message wasn't clear. I wasn't speaking as a customer, but as a dev. I would like to get the bug fixed myself and have started looking into it, but was wondering if anyone had already made some progress to save time.

dosipod, 2025-01-13:
At the moment I do not have the HW to test such number of outputs so we just kept this issue for DDP workarounds , the original case was here https://github.com/Aircoookie/WLED/issues/4301  . I think when the guys are now reporting issues with less then 10 outputs it may make sense to focus on that before saying why more then 10 do not work  .

You could also open a PR  if you think you have something to add that may fix it but I think you would need to dig a bit deeper for that as what you just listed about the settings page is just a ui thing .

DedeHai, 2025-05-19:
does https://github.com/wled/WLED/pull/4657 fix this?

blazoncek, 2025-05-19:
> does [#4657](https://github.com/wled/WLED/pull/4657) fix this?

I think it does.