wled/WLED #4329: 9个esp32控制9条独立的灯带，同时间只能有1个起作用，不知道是为什么？

czweb, 2024-11-28:
WLEDSR_0.13.4 soundReactive_esp32dev.bin uses this firmware and uses voice control to trigger switches. 9 esp32s are used to control 9 independent light strips, but only 1 can work at the same time. Why?

Attachment 1: Wiring diagram
[2024.11.25Wiring diagram.pdf](https://github.com/user-attachments/files/17945148/2024.11.25Wiring.diagram.pdf)
Attachment 2: Physical picture
[image]
[image]
[image]

softhack007, 2024-11-28:
1. Please translate the title of your feature proposal to englisch - most people here cannot read Korean.
2. What is the feature you want to suggest?
3. Where is your microphone? What model do you use?

czweb, 2024-11-28:
Sorry, the title has been modified.
Instead of using a microphone, a switch is used. I understand that directly giving an analog voltage of 0v or 3.3v can trigger voice control to achieve a chasing effect.
tihs：https://www.youtube.com/shorts/fDVOS8r-pdg

softhack007, 2024-11-28:
please ask on discord [image] 
to get help from the WLED user community.

softhack007, 2024-11-28:
> I understand that directly giving an analog voltage of 0v or 3.3v can trigger voice control to achieve a chasing effect.

Unfortunately, this is not supported. It may or may not work. Audioreactive is meant for audio input not for switches.

czweb, 2024-11-28:
> > I understand that directly giving an analog voltage of 0v or 3.3v can trigger voice control to achieve a chasing effect.
> 
> Unfortunately, this is not supported. It may or may not work. Audioreactive is meant for audio input not for switches.

You are absolutely right. Sometimes it works, sometimes it doesn't.
Why use audio to mess with the chasing effect? ​​Using macro + switch to trigger preset can only play one chasing beam at a time, but what I want is the chasing produced by the characteristics of audio, which can be repeatedly superimposed and has a good forward energy convergence effect;

The following video is the effect produced by ESP32 + WLED audio + switch when it can be used normally: https://www.youtube.com/shorts/8812W4DsiBo

closed 2025-04-06 by github-actions[bot]