wled/WLED #4330: Lights turn off automatically after power off on

chrisp1001, 2024-11-28:
### What happened?

Every time I power the controller off/on and switch on the lights they just switch off by themselves after about 1 minute. I need to restart the router to stop that behaviour. The controller is connected to the router by ethernet. It doesn't matter if I switch on the lights by the WLED web interface or UDP command. The router is a Fritzbox 7590. The controller always gets the same IP from the DHCP.

### To Reproduce Bug

Switch controller power off and on again

### Expected Behavior

The lights would not switch off automatically without any external command

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.4

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

I tried to upgrade to 0.15.0 however my hardware doesn't work with it.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chrisp1001, 2024-11-28:
It didn't happen when I was still using a Fritzbox 7580 so must be a problem with the router...

softhack007, 2024-11-28:
@chrisp1001 indeed it might be a problem with the FritzBox router. You might find some help here:
* #2932

closed 2025-04-06 by github-actions[bot]