wled/WLED #4331: Feature Request: Smart Dimming with Dynamic LED Distribution

unterdrueckt, 2024-11-28:
## Feature Request: Smart Dimming with Dynamic LED Distribution

### Description
Currently, WLED does not support dynamic dimming where only a subset of LEDs are lit at low brightness levels. At 1% brightness, all LEDs remain on, which can result in excessive brightness. A more efficient approach would be to selectively light a fraction of the LEDs at low brightness settings, improving both aesthetics and energy efficiency.

### Proposed Solution

1. **Low Brightness Control**:
   - At low brightness levels (e.g., 1%-10%), only a subset of LEDs should be lit (e.g., every 5th LED).

2. **Dynamic LED Distribution**:
   - **At higher brightness levels (e.g., 60%-100%)**, all LEDs should be fully lit, as they currently are.
   - **At mid-range brightness levels (e.g., 30%-50%)**, a mix of fully lit and selectively dimmed LEDs should be used to balance brightness and ensure smooth transitions.

3. **Improved Low-Light Dynamic Range**:
   - This would allow for better light and color control at low brightness levels.

### Additional Considerations

- This feature should be optional and configurable via the WLED settings, allowing users to control how many LEDs are grouped for dimming.
- The transition between LED states should be smooth to avoid flickering or abrupt brightness changes.

Thank you for considering this feature!

willmmiles, 2024-11-28:
I, for one, can relate to dynamic range issues at low brightness levels - for several of my projects I've been wishing for dimmer LEDs!

Spatial brightness distribution could potentially be implemented as an option at the segment level when grouping > 1.  Probably we'd want some kind of LED performance model to decide how to map energy to pixels.

DedeHai, 2024-11-29:
This would be quite easy to do as an FX. There then are even sliders to control the min/max point to enable this "dithering".

DedeHai, 2024-11-29:
@unterdrueckt if I were to make this into an FX: since I have no setup to test this I need some hints. Here is what I am thinking:
- create an FX that has a slider setting, if brightness goes below the threshold, leds will be turned to black -> how? which ones? in what pattern(s)?  (could make different patterns and have them selectable with another slider)
- default would be just like the "solid" FX but with added feature
- if "overlay" checkmark is set, it will not draw anything but just set certain LEDs to black, so it can be used as overlay segment to any other FX