wled/WLED #4332: Turn off leds on power up/reset failure

heyhewmike, 2024-11-29:
### What happened?

I have "Turn LEDs on after power up/reset" unchecked.
Upon power up my LED strip turns on.

Upon restoration of power after a power cut the LEDs turned on for 6+ hours with nobody home and for no reason.
[image]

### To Reproduce Bug

I am using an Athom controller, SKU: LS4P-WLED
WLED High Power Addressable LED Strip Controller. I am using SK6812 LEDs. From BTF-Lighting.
I have Preset 1 on boot selected which is the preferred preset anytime the device is turned on

### Expected Behavior

With the option unchecked the LEDs are to remain off until the button on the controller is pressed, power button is selected in the UI or the App/web page or Home Assistant.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.4 (build 2405180)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2024-11-29:
this is a feature not a bug: if you want them not to turn on, you can not load a preset that turns the leds on.

heyhewmike, 2024-11-29:
@DedeHai 
Can you please explain to me how I can have a default preset for when I turn on the LEDs & have them not turn on after the power is restored from a power cut?

It appears these 2 features conflict.

DedeHai, 2024-11-29:
this is not a support channel, please use discourse or discord for that. this is a development issue tracker. look at the JSON API used by presets as well: https://kno.wled.ge/interfaces/json-api/

heyhewmike, 2024-11-29:
I have gotten support from other channels. The feedback to use a preset that disables the LEDs on boot is not an option in my application nor does it make sense.

When the "Turn on LEDs after power up/reboot" option is unchecked that would indicate the controller would not "boot" upon receiving power.

The preset selection of 1 on boot would be when the power button is pressed on the controller, in the web UI or via alternate methods such as Home Assistant.

This is why I opened the issue as the current implementation of these 2 features are in conflict.

In terms of a computer, tower or laptop, you need to go into the Bios/UEFI to tell it to Turn On when power is received (Turn on LEDs after power up/reboot) and then in the Bootloader/Grub you instruct the system which Operating system to select (Preset on Boot), this part is often skipped when only running 1 Operating system.

Selecting Preset on Boot should not override the unchecked "Turn on LEDs after power up/reboot" option. 

Otherwise why have that Check Box option at all?

DedeHai, 2024-11-29:
The checkbox does exactly what it sais: it will not turn on the lights. If afterwards you overrule that with a preset you load at bootup, it also does what is expected. There is no conflict.

willmmiles, 2024-11-29:
I agree that the interaction between the checkbox and boot presets is confusing.  Given that the boot preset can now be easily identified and managed from the main UI, I think the checkbox is somewhat redundant now; the boot preset can fully define the desired target state and is easy to set.

Probably we should remove the checkbox from the LED settings panel, always initialize the internal strip state to off/solid black, and automatically create a boot preset (respecting the old checkbox setting if loading an old config) if none is selected.

willmmiles, 2024-11-29:
> The checkbox isn't redundant for my use case. I need the LEDs to be black/off when it receives initial power and remain that way and only follow the preset when the power button/GPIO trigger or the Power Button in the UI is used

If I'm understanding correctly, you want the final bootup state to be "preset 1, but off".  Is this correct?

This is not the intended purpose or behaviour of the existing checkbox.  I can see why the labeling might give the impression that that feature is related to what you're trying to achieve.  I agree with @DedeHai that repurposing the existing checkbox to a different, only tangentially related behaviour will cause more problems and confusion than it will solve.  Further, on an implementation level, I would object to conditional special-case overriding of preset settings on principle - IMO that's a recipe for exactly this kind of unclear behaviour expectations in the future!

From an immediately practical sense, the recommended approach for the current software would be:
- Load preset 1
- Push the global power button
- Save preset 100 (which will save all the preset state, plus the new power state)
- Set preset 100 as your boot preset

I recognize that this approach has the limitation that any future changes to preset 1 need to be saved twice.  I think that's something that could be improved on in a general-purpose fashion: either with an explicit preset include feature, allowing the boot preset to be a direct expression of "preset 1 but off", or maybe a clever use of playlists to load partial states back-to-back with no delay.

DedeHai, 2024-11-29:
@heyhewmike then you are one of the many that will need to update the config ;)
you have full control with the presets, as Will stated. You disable the bootup preset and assign that to whatever you use to switch the LEDs on.

willmmiles, 2024-11-29:
> @heyhewmike then you are one of the many that will need to update the config ;) you have full control with the presets, as Will stated. You disable the bootup preset and assign that to whatever you use to switch the LEDs on.

If I'm understanding correctly, the integration challenge here is a mixed bag of different interfaces -- some send "power on", others send "load preset 1", and the goal is to make both of these calls land in the same end state.

heyhewmike, 2024-11-29:
> @heyhewmike then you are one of the many that will need to update the config ;)
> you have full control with the presets, as Will stated. You disable the bootup preset and assign that to whatever you use to switch the LEDs on.

There is no clear place for me to set the button to use the Preset 1.

I did not roll my own hardware and I am using this one.
 https://www.athom.tech/blank-1/wled-high-power-led-strip-controller

DedeHai, 2024-11-30:
Any change to current behaviour would be a breaking change. IMHO the current way the current way is convoluted but I see no easy way out. Let me break down what current behaviour at bootup is:
- the `turnOnAtBoot` variable is checked
  - true -> set brightness to configured value
  - false -> keep brightness at 0
- load bootup-preset if set to not 0 (which can set brightness > 0 as well as change the on state)
- if brightness is now > 0 or preset uses "on" command: turn LEDs on
- update LEDs according to current settings

Removing the turn-on option makes it more complicated for users, as each and every one using that feature needs to understand (and edit) the JSON commands used in presets. 
Behaviour could be changed such that if `turnOnAtBoot` is false, the preset brightness and "on" state are ignored in this very first call to update the preset (not sure how, may require a lot of code changes as loading a preset is quite an elaborate process). This would break the behaviour for all users, that use a preset to turn on their LEDs after bootup and have `turnOnAtBoot` disabled. On top: once you do turn on the LEDs, the brightness would be the default one and not the one saved in the preset, that may be solvable though by still loading that brightness but not switching the state to "on" in a bootup call. Solving it this way would be the cleanest IMHO with the least breaks.

dosipod, 2024-11-30:
> how I can have a default preset for when I turn on the LEDs & have them not turn on after the power is restored from a power cut?
> It appears these 2 features conflict.
@heyhewmike Have you tried the option under user interface 
[image]

blazoncek, 2024-11-30:
FYI current behaviour is available since 0.11.0 and any change will be a breaking change.

Any user has the ability to unselect "Include brightness" which will not store brightness (AKA "on" mode) into a preset. If all presets do not have brightness saved (as some users have) than it is mandatory to respect "turn on at boot" setting.

I would discourage any change and rather add better explanation to KB.

closed 2025-04-08 by github-actions[bot]