wled/WLED #4333: Color Order Override - Deleting adds more on save

beasthouse-au, 2024-11-30:
### What happened?

When I try to delete Color Order Override's, after saving the config, it actually adds the same Color Order Override again. Then the lights start having problems, flickering etc.

### To Reproduce Bug

When I try to delete Color Order Override's, after saving the config, it actually adds the same Color Order Override again. Then the lights start having problems, flickering etc. E.g. below I have added two, then for some reason it was duplicated. I hit the minus button and they disappear, hit Save config and then when the page is reloaded the original duplicate is there with another set of duplicates. This will happen over and over.

[image]

### Expected Behavior

Additional overrides are deleted.

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0-b7

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

beasthouse-au, 2024-11-30:
Dropping back to 0.15.0-b5 resolves the issue.

dosipod, 2024-11-30:
@beasthouse-au This is fixed already in this PR https://github.com/Aircoookie/WLED/pull/4326
compile from source and it will be fine

closed 2024-11-30 by DedeHai