wled/WLED #4334: Drip doesn't use palette

djdevin, 2024-12-01:
### What happened?

Drip is marked as using a palette but only solid colors can be used

### To Reproduce Bug

Set drip effect, select palette

### Expected Behavior

Effect uses selected palette

### Install Method

Binary from WLED.me

### What version of WLED?

0.14

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2024-12-01:
easy fix: remove palette setting

djdevin, 2024-12-02:
It would be nice if it could use palette! Could make a winter/ice setting with blues and whites.

Should be marked as a feature then.

softhack007, 2024-12-02:
> I have a new, much enhanced version of the effect that supports palette. It may come into WLED soon.

@DedeHai maybe I can merge my MoonModules changes with your version? I've also added a few extras - mainly to make the effect look better on large panels (slowed it down, added palette support, made drops slightly more random).

https://github.com/MoonModules/WLED/blob/796eb6e23e493cfce9b942fc950afdfebd83deac/wled00/FX.cpp#L3747

DedeHai, 2024-12-02:
@softhack007 my version uses the particle system... so not really compatible

softhack007, 2024-12-02:
> the question is, should the current 1D FX be replaced with the PS versions I made or should both coexist.

Can your effect also support virtual strips ("Bars" mapping where each bar is treated like an effect instance) ? 
[image]

If yes, then maybe we just need one effect based on the particle system.

DedeHai, 2024-12-02:
currently, bar mapping/virtual strips is not possible, that is something I need to look into how to solve once the rest of the code is done