wled/WLED #4335: FX115 ( Blends) Missing pixels on peek and fixtures

dosipod, 2024-12-01:
### What happened?

FX115 ( Blends) missing pixels 
[image]

### To Reproduce Bug

Use FX115 ( Blends)  and you will notice missing pixels on peek and fixtures 

### Expected Behavior

No missing pixels  

### Install Method

Self-Compiled

### What version of WLED?

0.15.0-rc1 (build 2411250)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-12-01:
FYI Blends is a 1D effect with limitations to accommodate ESP8266.

softhack007, 2024-12-01:
Looks like the effect should be "modernised" ... it creates a fastLed - style pixel buffer the effect cannot handle more than 255 pixels in total.

I think the complete "pixels" buffer is not needed and it could be replaced by `getPixelColor`.

This loop could use getPixelColor instead of the pixels buffer
https://github.com/Aircoookie/WLED/blob/a121f5b61b2dbb2934630120f22527a5cdab91d4/wled00/FX.cpp#L4492-L4495

The second loop just writes the pixels back
https://github.com/Aircoookie/WLED/blob/a121f5b61b2dbb2934630120f22527a5cdab91d4/wled00/FX.cpp#L4498-L4501

dosipod, 2024-12-02:
It is fixed now on MM

dosipod, 2024-12-02:
The two 16x16 and 32x16 fixtures I tested with in the beginning  of the post are okay now  without any missing pixels on effect default .

Below I would not call it a bug because ARC might be meant to behave this way so just to log test results with few scenarios ( I even like the last test with 32x16 having a hole in the middle ) 
[image]

You may close at your discretion

blazoncek, 2024-12-03:
@dosipod that's a problem Arc expansion has on some dimensions of segments. But may be related to how that particular effect calculates its maximum dimension.

dosipod, 2024-12-03:
@blazoncek Yeah I see the same with multiple effects when using Arc so not an issue with Blends , as long as there is symmetry it is fine

DedeHai, 2024-12-03:
> i would you like to keep this report open, as a reminder for modernizing the logic?

does it need modernizing? it will change the looks if you remove the buffer (on all setups that have global buffer disabled), not sure its worth it.