wled/WLED #4338: Present state should be saved as preset 0

djdevin, 2024-12-02:
**Is your feature request related to a problem? Please describe.**
I was surprised that my settings weren't saved upon rebooting the controller, which is how all of my non-WLED controllers function. I had to manually save the settings to a preset and configure that preset to load by default in LED settings.

I don't see a benefit to discarding live changes, especially if there is power loss or a crash. Saving in near-realtime would make general adoption easier as well, as you would not have to manage presets.

**Describe the solution you'd like**
Live changes should persist to state 0.

**Describe alternatives you've considered**
Live changes could persist to a different state, so if people are saving to state 0 that behavior could be preserved and you could load the "last state".

**Additional context**
Any non-WLED controller I have found preserves the current state unless explicitly set.

dosipod, 2024-12-02:
https://github.com/Aircoookie/WLED/tree/0_15/usermods/usermod_v2_auto_save

blazoncek, 2024-12-03:
Be aware that writing to flash will wear it down.
I sacrificed an ESP to determine number of writes: it was about 43000.

closed 2024-12-03 by netmindz