wled/WLED #4339: 0.15.0-rc.1 Sync behaving intermittently.

GVDub, 2024-12-02:
### What happened?

Two systems set up to sync: first is three strips, master set to send sync on channel 1, two strips set to receive on channel 1. All running Gledopto non-audio reactive controllers (8266-based GL-C-008WL) and updated using WLED_0.15.0-rc1_ESP02.bin.gz. Second is two 32x32 (four 16x16 panel) matrices, one with the GL-C-310WL mini controller, the other with the GL-C- 010WL-D. On 0.14.4, sync was reliable. Now, I have to toggle power on and off several times and toggle sync on the masters on and off several times for it to take effect. Even once it's working, over the course of the day, it can drop. This is with both 0.15.0b7 and 0.15.1rc1. 

### To Reproduce Bug

Just turning on sync (or setup with "Enable Sync on Boot" option doesn't seem to communicate sync to the slaved controllers.  

### Expected Behavior

Expected sync to behave reliably. 

### Install Method

Binary from WLED.me

### What version of WLED?

Server message: WLED 0.15.0-rc1 (build 2411250)

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-12-03:
You need to enable "events" that trigger a broadcast: direct change, Alexa, IR, etc.

WizBangCrash, 2024-12-05:
I am seeing a similar issue. I have put Wireshark on my network and it looks like WLED is not always transmitting the UDP packets.
I changed the `UDP packet retransmissions` count to 5 and Wireshark sees anything from 0 to 5 packets transmitted each time I change an effect in the UI.
I have `Send notifications on direct change` enabled. See screenshot below.
Here is the capture from 4 separate & different changes to effects with the retransmissions set to 5. It is only showing a total of 5 packets, with one of the effect changes missing altogether.

[image]

I am running WLED 0.15.0-rc1 (build 2411250) too, on multiple Athom LS8P/IR ESP32 lightstrip controllers.
All devices are on the same Ubiquiti U6-Pro Access Point

[image]

blazoncek, 2024-12-05:
FYI UDP can be very unreliable on WiFi networks as it has no confirmation of delivery. Make sure your WiFi signal strength is as high as possible.

WizBangCrash, 2024-12-05:
Thanks @blazoncek . I have been investigating this further and it seems that just one of the 4 Athom controllers I have is missing 99% of the sync requests even though it seems to have a better signal strength than 2 of the others.
I have just changed it's power supply as I wondered if the original PSU is noisy, but no difference.

I have have experimented and if I increase the UDP resends to 30 it has become reliable. Although the effect change can lag by a few seconds 😁

I have definitely come to the conclusion it is not a WLED issue though. Thanks for the quick response.

closed 2025-04-12 by github-actions[bot]