wled/WLED #4340: WLED Release 0.15.0-rc.1 presets not the same as before

nevergitithub, 2024-12-02:
### What happened?

After update from WLED 0.14.4 (build 2405180) on a ESP8266, some presets have change the appearance depite correct code.

### To Reproduce Bug

loading the preset code 

### Expected Behavior

same appearance as before

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0-rc.1

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
Example of preset not working correct:

{"on":true,"bri":37,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":300,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"col":[[255,106,25,255],[255,160,0,0],[198,101,71,0]],"fx":36,"sx":35,"ix":251,"pal":4,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}
```

### Anything else?

Other presets work as before

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2024-12-02:
🤔 fx=36 is "wipe random". I guess that this commit causes different behaviour of the effect: https://github.com/Aircoookie/WLED/commit/68a7282b2765f4c43c838e7554a93a95f4385406

several `uint16_t` were replaced by `unsigned` and might need an additional `& 0xFFFF` now to preserve the old behaviour

https://github.com/Aircoookie/WLED/blob/99427c2ef734eb0bd04ec2c5f9d21655d0349e25/wled00/FX.cpp#L161

softhack007, 2024-12-02:
> some presets have change the appearance

@nevergitithub can you be more specific about the effects used in your presets?

blazoncek, 2024-12-03:
> 🤔 fx=36 is "wipe random". I guess that this commit causes different behaviour of the effect: [68a7282](https://github.com/Aircoookie/WLED/commit/68a7282b2765f4c43c838e7554a93a95f4385406)
> 
> several `uint16_t` were replaced by `unsigned` and might need an additional `& 0xFFFF` now to preserve the old behaviour
> 
> https://github.com/Aircoookie/WLED/blob/99427c2ef734eb0bd04ec2c5f9d21655d0349e25/wled00/FX.cpp#L161

Indeed there was no response on these changes (from beta testers) when requested on Discord. I tested most but not all.