wled/WLED #4343: WLED doesn't compile with BME680 usermod for lolin s2 mini

thepuncake, 2024-12-03:
### What happened?

Tried to compile WLED through https://wled-compile.github.io/ with just the BME680 usermod enabled for lolin s2 mini board but it fails with this error: `/home/runner/.platformio/packages/toolchain-xtensa-esp32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: cannot find -lalgobsec
collect2: error: ld returned 1 exit status`

### To Reproduce Bug

Compile WLED for lolin s2 mini with the bme680 usermod

### Expected Behavior

Compile succesfully

### Install Method

Self-Compiled

### What version of WLED?

wled v0.15.0-rc.1

### Which microcontroller/board are you seeing the problem on?

ESP32-S2

### Relevant log/trace output

```shell
/home/runner/.platformio/packages/toolchain-xtensa-esp32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: cannot find -lalgobsec
collect2: error: ld returned 1 exit status
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netmindz, 2024-12-03:
It failed for me with the regular esp32 too

netmindz, 2024-12-03:
https://community.platformio.org/t/error-on-link-process-cannot-find-lalgobsec/18729

softhack007, 2025-01-03:
Might be related to
* #4362 
* #4363

closed 2025-05-12 by github-actions[bot]