wled/WLED #4344: Add "degree" sign a.k.a. °C into scrolling text effect for temperature display

robertvb83, 2024-12-04:
**Is your feature request related to a problem? Please describe.**

i collect weather data from openweathermap and forward the temperature value to a scrolling text segment via json request. Unfortunately the "degree" sign as in °C is not accepted by that effect.

**Describe the solution you'd like**

The degree sign °C should be visualized as a single Pixel at the top of the text level. I cannot achieve that with any other option like "', `´, ^"

DedeHai, 2024-12-04:
no new chars can be added without replacing an existing one. but you can do that yourself:
edit the font file of the font size you are using and replace one of the existing characters with what you need.
https://github.com/Aircoookie/WLED/blob/0_15/wled00/src/font/console_font_5x8.h
then compile from source.

robertvb83, 2024-12-04:
Thanks for the hint. I can't get this to work.

I still get "2\u00b0C" instead of "2°C"

In the font file (in all of them) i uncommented the following

    /*
     * code=176, hex=0xB0, ascii="!0"
     */
    0x40,  /* 0100 */
    0x10,  /* 0001 */
    0x40,  /* 0100 */
    0x10,  /* 0001 */
    0x40,  /* 0100 */
    0x10,  /* 0001 */

In exchange i commented the following:

    // /*
    //  * code=126, hex=0x7E, ascii="~"
    //  */
    // 0x50,  /* 0101 */
    // 0xA0,  /* 1010 */
    // 0x00,  /* 0000 */
    // 0x00,  /* 0000 */
    // 0x00,  /* 0000 */
    // 0x00,  /* 0000 */

Did i do that correctly?

DedeHai, 2024-12-04:
almost, you exchanged the ~ char with some  arrowy thing. the binary code kind of reveals what it will look like (1 is colored, 0 is black), that binary is encoded in hex, so if you want a single dot that would be something like
0x01,  /* 0001 */
0x00,  /* 0000*/
0x00,  /* 0000*/
0x00,  /* 0000*/
0x00,  /* 0000*/
0x00,  /* 0000*/

and you need to use the ~ char to display that

robertvb83, 2024-12-04:
Thanks for explaining, finally i got it... The visualization of the ~ character (or whatever you chose) is changed to ° symbol.

Actually it is given in code=248 for all the font sizes

    /*
     * code=126, hex=0x7E, ascii="~"
     */
    // 0x50,  /* 0101 */
    // 0xA0,  /* 1010 */
    // 0x00,  /* 0000 */
    // 0x00,  /* 0000 */
    // 0x00,  /* 0000 */
    // 0x00,  /* 0000 */
    0x20,  /* 0010 */
    0x50,  /* 0101 */
    0x20,  /* 0010 */
    0x00,  /* 0000 */
    0x00,  /* 0000 */
    0x00,  /* 0000 */

closed 2025-03-10 by blazoncek