wled/WLED #4347: ESP32-C3 becomes unresponsible after a couple of days

AdmiralRaccoon, 2024-12-05:
### What happened?

I´ve already read that the ESP32-C3 is not the best solution, but maybe it´s a bug anyway causing the issue:

After a couple of days, the device because unresponsible within Home Assistant as well as pinging it / access to the WebUI. Even pressing the button for toggeling won´t work. I have to remove power to get it working again, and then it will work again for the next 3-5 days.

I will notice when it stops working in the morning, as it supposed to wake me up in the morning.

### To Reproduce Bug

Waiting several days

### Expected Behavior

Working 24/7 

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0-b7

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Anything else?

It´s a WLED Slim LED Strip Controller from Athom:
https://de.athom.tech/blank-1/wled-slim-led-strip-controller

It´s doesn´t get hot
WIFI sleep is disabled
WIFI TX Power is at 8.5 dBm (that was the default)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netmindz, 2024-12-05:
Is this an issue only after you upgraded to 0.15 ?

What version was it previously stable on?

AdmiralRaccoon, 2024-12-05:
I used the 0.14.4 for about a week after i bought the controller, but had an issue with presets, which were actually fixed with the 0.15, so i directly started using 0.15.0-b6 (which already had these unresponsible issues)

DedeHai, 2024-12-05:
This could be a hardware related issue as well: please try this: add a large 1000uF capacitor to both 5V input and 3.3V rail on the ESP and report if the issue persists.

AdmiralRaccoon, 2024-12-05:
Now that exceeds my knowledge, as this is a ready-to-user controller with an usb-c power input:

[image]

I thought about that it also could be hardware related issue, but I rather replace the C3 with an ordinary ESP32, as these work rock solid with wled.

KrX3D, 2024-12-07:
> F capacitor to both 5V input and

hi, thanks for that tip, i will try it on my devides since i got the same problem but not just with the C3 but also with s2 and some esp32 dev boards

daggerrz, 2024-12-11:
I think this issue more likely is related to an underlying problem with the ESPAsyncWebServer dependency. @AdmiralRaccoon, are you using Home Assistant by any chance? HA can be pretty aggressive in its polling and open many concurrent conenctions to the WLED device, which might lock it up. 

See https://github.com/me-no-dev/ESPAsyncWebServer/issues/157 and https://github.com/me-no-dev/ESPAsyncWebServer/pull/1116

This is still easy to reproduce with a synthetic load:

```
wrk -c 1 -d 20 -t 1 --timeout 5 http://IP/json
```

Increasing concurrency, `-c` to e.g `8` will consistently lock up an ESP32-C3 on 0.15. I haven't tested other ESP32 variants recently, but they will likely fail around the same level of concurrency as well.

AdmiralRaccoon, 2024-12-12:
So might this be something for the HA WLED Integration Developer to look at and do something about polling interval?

willmmiles, 2024-12-12:
I've been working on some upgrades to the web server components to better handle concurrency and excessive load in #4119 -- the underlying AsyncTCP library melts down rather dramatically if its event queue overflows.  While the code there will probably improve load related problems for C3s, I've still got some unresolved issues in low memory environments such as S2s.

mattfox27, 2024-12-17:
I have been having the same issue, and yes im using home assistant

u20p17, 2024-12-29:
i have the same issue with a quinled dig-uno… will there be a solution available in the next releases of wled? :)

dtswk, 2025-01-08:
Likewise HA used same issue with esp32 c3 hanging.  Happy to test fixes etc if I can help.  Not sure if there is a way to throttle HA polling I'll do some researching.

dosipod, 2025-01-08:
@dtswk 
[image]

dtswk, 2025-01-08:
@dosipod Thank you so much, I searched overnight and didn't find this, was heading to making my own MQTT integration.  You saved me a bunch of time.  :-)

dtswk, 2025-01-09:
ok I have something kinda interesting to report.  My lamps have a button so you can turn on and off, long press etc.   I turned off polling and put in an automation to update every minute.  But I've found that the entity updates in realtime to a button press... thinking that it was timing I tested multiple times and I believe the WLED device pushes realtime updates to HA even with polling disabled.  If so AWESOME...!   you can turn off polling and still have 'realtime' updates going to HA.

u20p17, 2025-01-20:
> [@dtswk](https://github.com/dtswk) [image]

i tried this and since over a week I never had the problem again - thanks!

wernerhp, 2025-02-08:
Same issue where WLED becomes unresponsive.

When I disable the HA Integration, WLED seems stable. When the HA Integration is enabled, after a couple of days WLED hangs and becomes unresponsive. Lights remain in their last state and can see on my network controller that the devices are still connected to WiFi, but cannot be accessed or controlled via HA or the WLED or IP.  The WLED controller needs to be power cycled and then it works for a few days again.

Serko's ESP32 mini board.
https://www.tindie.com/products/28858/

I have two in my garage and both do the same since I had them. Think v0.13.0 originally. Now on 0.15.0 still.

I don't have this issue with the SP501e in my Living Room or any other SP501e I used in the past.

wernerhp, 2025-06-16:
Woah! Can we please take a look at this!

willmmiles, 2025-06-16:
I've ordered a C3 board for local testing.  I'll see what I can find out when it arrives.

wernerhp, 2025-08-09:
Why is it closed as not planned?
@willmmiles is looking into it?

Something I noticed is that, while unreachable, they are still connected to the WiFi.

[image]

[image]

wernerhp, 2025-08-09:
I rebooted the devices and turned on Disable WiFi sleep [x]

It didn't take long for them to go Offline again.

[image]

[image]

netmindz, 2025-08-09:
What version is that?

blazoncek, 2025-08-10:
You won't get far without cfg.json and debug log.

blazoncek, 2025-08-10:
By using a debug build and monitoring serial output.

Just FYI all of my C3 units work as expected. So it must be something in the config or network environment.

wernerhp, 2025-08-10:
Could you hook me up with a and WLED_0.15.1_ESP32 debug build?

blazoncek, 2025-08-10:
No and HA is known to cause issues with WLED by polling for state too often and depleting ESP resources.
Try disabling it or reducing the frequency of updates.

wernerhp, 2025-08-10:
> No and HA is known to cause issues with WLED by polling for state too often and depleting ESP resources. Try disabling it or reducing the frequency of updates.

I've do have HA, and I saw those comments and have tested without HA polling.  It still stops responding, it just takes longer, but it still stops responding after some time.

I have another WLED controller ([SP501e - ESP8266](https://github.com/wled/WLED/issues/4347#issuecomment-2645899526)) which has zero drops or freezes even with HA polling enabled.

wernerhp, 2025-08-12:
[image]

Ok, experiment failed.  Still freezes after using a known stable AC power source for 2 days. The other one is still going. They don't freeze at the exact same time.

wernerhp, 2025-08-12:
Garage Door Light (BMW) stopped responding
[image]
[image]

Garage Door Light (VW) still going
[image]

blazoncek, 2025-08-12:
Your file system is of wrong size or you are having a very unusual 12MB unit.

willmmiles, 2025-08-12:
16MB C3 units are technically possible...

[WLED_0.15.2-beta1_webserver_ESP32-C3.bin.zip](https://github.com/user-attachments/files/21742809/WLED_0.15.2-beta1_webserver_ESP32-C3.bin.zip)

@wernerhp Would you mind giving this build a try?  It's based on 0.15.1 with an updated web server and AsyncTCP, which should fix one possible source of memory corruption.

willmmiles, 2025-08-12:
Also the relevant patch is already in the 0.16 nightlies, you could try one of those.

wernerhp, 2025-08-14:
> Test 0.15.1 based build is here: https://github.com/wled/WLED/actions/runs/16925022565/

Thanks.  I installed 0.15.1 on both and rebooted.  Will let you know how it goes.

willmmiles, 2025-08-30:
Thanks for the report - sorry to hear it didn't solve the problem.  Apologies I haven't got back to you sooner, I'm a bit oversubscribed at the moment.  It does sound like something's blocking the network task, and we've got some evidence that there's still some correctness issue with AsyncWebServer.  I don't know exactly when I'll get to looking in to it, though - I haven't been able to reproduce this case on the bench yet; it doesn't seem to be triggerable by web server stress testing tools.

wernerhp, 2026-01-16:
Send some pictures of your complete setups  

I had the [ESP32-mini-SerKo](https://github.com/srg74/ESP32-mini-SerKo) connected to the [WLED mini shield](https://www.tindie.com/products/serg74/wled-mini-shield-board-for-addressable-leds/) and 143 LED strip with the MeanWell power supply as describe [here](https://github.com/wled/WLED/issues/4347#issuecomment-3194415507).  It crashes very frequently.  Sometimes within seconds.

I've now had the ESP32-mini-SerKo alone powered via USB-C and no WLED mini shield and it's been going for over 9 days.

I want to test it on the shield next, but the fuse is in the way of the USB-C, so I'm not sure how I'm going to monitor the logs for when it crashes.  Will have to make a plan.

DedeHai, 2026-01-16:
The current state of knowledge are that there are two reasons for crashes:
1. bad hardware / power design, stick to the wiring guide to fix that and use quality supplies, add capacitors (!)
2. a bug in async webserver which can be triggered by using HA's aggressive polling system