wled/WLED #4348: Handle preset deletion asynchronously for consistency with preset saving

david-hogberg, 2024-12-05:
**Is your feature request related to a problem? Please describe.**
I'm developing a BLE mod for wled which works fine for commands except preset/playlist deletion (device crashes). I then noticed that presets are deleted using immediate file operations rather than asynchronously using flags, which is how presets are saved for example. Same for API calls. Is this there a reason for this or a bug? Issue solved if I move deletion to be handled later in the main loop (see below). 

Deletion still works fine using wifi, but thought it might be a general stability improvement anyway. 

**Describe the solution you'd like**
In `presets.cpp`: Instead of directly initiating file operations in `deletePreset(index)`, called from `deserializeState`, set a flag `presetToDelete = index` that's picked up in `handlePresets()`. Something like this:

```
void handlePresets()
{
  if (presetToDelete) {
    doDeleteState();
    return;
  }

  if (presetToSave) {
    doSaveState();
    return;
  }
  
  ...
  
}
```
Adds consistent behaviour between preset deletion and saving, and prevents potential blocking operations during preset deletion.

DedeHai, 2024-12-06:
this looks like it could solve some issues users have with corrupted files (as referenced by @softhack007)
@david-hogberg since you already looked into this: are you able to implement the solution you suggested and do a PR?

david-hogberg, 2024-12-06:
Sure, I'll create a PR

david-hogberg, 2024-12-11:
@blazoncek Thanks for your input. My point here is more about consistency, that saving and deleting are approached in different ways as of now. Which option to choose of course depends on where WLED is heading in general with immediate vs. asynchronous operations. Let me know if a PR might be useful still.

blazoncek, 2024-12-11:
I would prefer synchronous preset handling as it has less drawbacks than current implementation. After we solve corruption issues, one way or the other.

willmmiles, 2024-12-11:
Probably we need a mutex over the presets database.  I think the current code can get in to a lot of trouble if there's requests via different APIs in flight.

blazoncek, 2024-12-12:
I disagree. We need to fix preset corruption and then return original immediate/synchronous preset (load/save/delete) handling.

willmmiles, 2024-12-12:
File corruption can be caused by attempting parallel operations on the same file from different tasks (eg. web server callback is writing while loop is trying to read -- or worse, *also* trying to write).  Maybe I haven't been around long enough, but if I was facing corruption issues, the first thing I'd try is making sure only one task can use the file at a time, so every operation runs to completion before the next operation can see the changes.

softhack007, 2026-03-08:
> Probably we need a mutex over the presets database. I think the current code can get in to a lot of trouble if there's requests via different APIs in flight.

Yes, I recently did exactly this in WLED-MM (experiment) - add a muted to prevent parallel preset file writes from several tasks. It's an easy solution and it works.