wled/WLED #4364: Randomly selecting an effect falls back to Solid.

thermseekr, 2024-12-07:
### What happened?

In Home Assistant, when randomly selecting an effect using a template, sometimes WLED selects Solid instead of an existing effect.

### To Reproduce Bug

Paste the template below in a Render Template Node in NodeRed.

`{{ state_attr('light.light_name', 'effect_list') | random }}`

Connect this node to a Debug and to a Call Service nodes. Inside the Call Service node, paste the expression below to the Data field.

`{ "effect": msg.payload }`	

Every time the flow is run, the name of the selected effect will be shown in the Debug window in NodeRed and will be fed to the service call. Depending on the selected effect WLED will fall back to SOLID. Looking at the effects list in WLED one can see that in these cases the randomly selected effect do not match the name of any existing effect.

### Expected Behavior

The name of the effects in `effect_list` should match the current effect names, so WLED will never fall back to SOLID.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.4 Hoshi Build 2405180

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netmindz, 2024-12-10:
If you do not have a 2D setup and you select a 2D effect then you will get solid as fallback

netmindz, 2024-12-10:
I'm inclined to agree that the list of effects in HA should be the same as shown in the UI, I'm not sure however if your technique of selecting effects at random is valid as I'm not familiar with NodeRed

blazoncek, 2024-12-10:
Effect list is universal an includes all available effects.
What is shown in UI depends on filtering selection and capabilities of current configuration. This is UI action.

thermseekr, 2024-12-11:
Sorry I'm a very basic WLED user and I was not even aware of the existance of 2D effects (oh gosh one more itch to scratch). I thought maybe the effects list in the home assistant integration was not updated to the actual effects list on WLED thus causing the problem. I'll try to check with the owner of the integration if there is something that could be done, maybe a toggle to enable/disable 2D effects reflecting what is configured in WLED, then disabling the selection of an invalid effect from the integration.

Thans for the replies, wish you all a merry Xmas and a happy 2025.
Tales

closed 2024-12-11 by thermseekr