wled/WLED #4369: [ESP32] Unable to upgrade from 14.4.4 to any 15.x releases

jsloyer, 2024-12-09:
### What happened?

I am unable to update to any 15.x releases for one of my ESP32’s. I tried rc1, b3, and b7. All fail.

[image]
[image]
[image]
[image]

### To Reproduce Bug

I’ve only had this on a single ESP32. All of my other ESP’s upgraded fine.

### Expected Behavior

To be able to upgrade to 15.c.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.4 (build 2405180)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2024-12-09:
maybe related to tasmota bootloader: https://github.com/Aircoookie/WLED/issues/4241

netmindz, 2024-12-09:
Please confirm @jsloyer exactly how and where you did your **_initial_** install of WLED onto this device from (not how you installed any previous updates)

jsloyer, 2024-12-09:
I'm actually not sure. The ESP32 came from quindoor, https://quinled.info/quinled-esp32/.

blazoncek, 2024-12-10:
This is related to inadequate application partitions. Your file system should read 983kB when correct partitions are in place.
You will need to flash using USB connection but doing so will erase all file system content (configuration and presets) so back up before flashing.

misosec, 2024-12-11:
I have the same problem with Quinled ESP32 updated via Homeassistant. My file system size is the same: 20/1707 kB.
HA doesn't update is either - updating just stops.

chrisella, 2024-12-11:
I'm experiencing the same problem +1
I've tried updating directly with the various downloaded binaries AND through HA, neither works.

DedeHai, 2024-12-11:
@misosec @chrisella  the answer was already given by blazoncek. your file system MUST BE 983kB (see info tab in WLED) or you will not be able to do over the air update.

chrisella, 2024-12-11:
Ah that's awkward then, particularly for several of my units that are "deployed" in positions that make that not easily possible.
Guess this is the end of the road for me with WLED then until I possibly get a chance to physically pull them out 😢

softhack007, 2024-12-11:
> I have the same problem with Quinled ESP32 updated via Homeassistant. My file system size is the same: 20/1707 kB.
> HA doesn't update is either - updating just stops.

@intermittech (Quindor) Can you check quinled boards still ship with the (very old) 1.5MB LittleFS partition scheme? Or is this user having a possibly very old install?

FYI, we have changed to a new partition scheme - 1MB LitleFS - already 3 years ago (WLED 0.13.0).

erikvaliquette, 2024-12-11:
Same issue, can't update to version 15 on all my quin led devices. Unable to communicate with wled api

[image]

blazoncek, 2024-12-11:
@erikvaliquette please use USB method as described above.

gzxto1, 2024-12-11:
Same issues here, all my ESP8266 worked fine, all ESP32 don't work via HA or binary upload

erikvaliquette, 2024-12-11:
> @erikvaliquette please use USB method as described above.

Which is not really possible for the moment as they are part of the holiday decorations 🎅

jsloyer, 2024-12-11:
Just for others that find this using the installer at install.wled.me fixed this for me.

closed 2024-12-11 by jsloyer

chiefymuc, 2024-12-12:
Quinled also has binaries here: https://github.com/intermittech/QuinLED-Firmware/releases

intermittech, 2024-12-12:
> > I have the same problem with Quinled ESP32 updated via Homeassistant. My file system size is the same: 20/1707 kB.
> > HA doesn't update is either - updating just stops.
> 
> @intermittech (Quindor) Can you check if quinled boards still ship with the (very old) 1.5MB LittleFS partition scheme? Or is this user having a possibly very old install?
> 
> FYI, we have changed to a new partition scheme - 1MB LitleFS - already 3 years ago (WLED 0.13.0).

We've been flashing WLED v0.14.x for quite a while now so that should be fine in regards to partition size. We also offer https://install.quinled.info which has also had that version for quite a while.

It is however possible people have had these boards since the v0.13.x days and never updated other then doing OTA. My boards all come with USB-C partly because of this reason, although I get it's sometimes inconvenient when it's part of a show right now!

I will doublecheck though to make sure boards aren't going out with a too old firmware, maybe update the pre-installed version to v0.15 now that it's been released.

softplus, 2024-12-13:
Is there a way to change the partition without a USB connection? I set up 2 ESP32 modules a few years ago, updates until 14.4 worked fine, they're installed in pretty inaccessible places. The generic "update failed" was a bit annoying to figure out, but I realize "back-updating" 14.4 to check for an incompatible setup required for the next release is ~impossible :).

blazoncek, 2024-12-13:
Short answer: No.

Long one: Technically possible but will require specially crafted software that would copy itself out of regular application partitions. Then update partition table, format partitions and start OTA download process.

chiefymuc, 2024-12-13:
Someone apparently figured it out here: 
https://www.reddit.com/r/esp32/comments/18dtwip/changing_the_partitiontable_over_the_air/

But I am not smart enough to understand without a step-by-step guide... But would be highly interested as my DigUno is also hidden.

blazoncek, 2024-12-13:
@chiefymuc I wrote it is technically possible. But it is not for the faint-hearted.

softhack007, 2024-12-15:
@chiefymuc I just found the same reddit post.

If you cannot access the esp32 to update via USB, I would not recommend trying the hack described by the linked articles. It anything goes wrong, your device will be in a bootloop (not able to start wled) and you must take it down from the roof to your desk, in order to attach it to a computer by USB.

https://gist.github.com/floitsch/ed2530caa613581057d8998dee0a911f

The author of the hack says "it's like replacing the jet engine while you are flying it". I think that's a serious warning.

softplus, 2024-12-15:
I created a firmware to fix this for me. You basically OTA to this, have the partition table updated (and hope, I guess), then you can OTA to WLED 0.15.0. https://github.com/softplus/Esp32Repartition

I've tried this with a test device, and my two WLED devices that were stuck because of the partition table sizes, but I can't promise that it'll work everywhere, obviously. Also, if something goes weird with your device at the wrong time (there's a really short window), it can leave the partition table broken.

rhinodavid, 2024-12-16:
> I created a firmware to fix this for me. You basically OTA to this, have the partition table updated (and hope, I guess), then you can OTA to WLED 0.15.0. https://github.com/softplus/Esp32Repartition

Works for me on a DigUno that initially had v13 installed. Legend.

chrisella, 2024-12-16:
> I created a firmware to fix this for me. You basically OTA to this, have the partition table updated (and hope, I guess), then you can OTA to WLED 0.15.0. [softplus/Esp32Repartition](https://github.com/softplus/Esp32Repartition)

As a warning, this unfortunately didn't work for me, luckily I tried it first on one of my "easier" to access (still PITA) units", so now that it's unresponsive I'll have to go about the harder route. This could in all likelihood be due to my board or config or something else, but just be warned that path is still full of possible nastiness.

blazoncek, 2024-12-16:
One (possibly crucial) note when updating partitions from previous version using OTA:
When an ESP device was 1st flashed (or last time it was flashed using serial connection) it was flashed with bootloader and app0 binary. Bootloaders change! What was originally flashed may no longer work (well) with recent WLED firmware as some features may require newer bootloader features.
I just want you to be aware of the possibilities.

nmbgeek, 2024-12-22:
> I created a firmware to fix this for me. You basically OTA to this, have the partition table updated (and hope, I guess), then you can OTA to WLED 0.15.0. [...]

This worked for me!  Thank you @softplus

netmindz, 2024-12-23:
Thanks for sharing that @softplus 

Both the WLED (AC) and WLED-MM teams have both been struggling with image size Vs the current partition table out in the wild that allocates quite a large amount of space for the file system that only has the config and presets files in most cases. 

We have wanted to reduce the size of the filesystem to give more space to the app, perhaps your approach of using an intermediate specialist bin might be something we can use for that

chatziko, 2025-04-15:
@softplus : just wanted to say a big thanks for your Esp32Repartition tool, it worked like a charm.

Fun story: after running the tool and flashing WLED 0.15 I was waiting for WLED-AP to appear to restore my config. But it didn't, so I got dissapointed and started preparing ladders for USB flashing. Only to realize that my WLED configuration was actually __preserved__ by the partition upgrade for some reason, so the upgraded WLED just happily connected to my wifi and I was ready to go 😄

JHarding86, 2026-01-19:
@softplus You're a hero. Worked like a charm on my Quinled board. Configuration preserved throughout the update process. Wonderful! Thank you.