wled/WLED #4375: Usermods  TTGO-T-Display does not compile

dosipod, 2024-12-11:
### What happened?

When trying to compile https://github.com/Aircoookie/WLED/tree/main/usermods/TTGO-T-Display  [On github] you would get errors 

```
 Compiling .pio/build/esp32dev/lib831/TFT_eSPI/TFT_eSPI.cpp.o
Compiling .pio/build/esp32dev/libf3e/arduinoFFT/arduinoFFT.cpp.o
In file included from .pio/libdeps/esp32dev/TFT_eSPI/TFT_eSPI.h:101:0,
                 from .pio/libdeps/esp32dev/TFT_eSPI/TFT_eSPI.cpp:16:
.pio/libdeps/esp32dev/TFT_eSPI/Processors/TFT_eSPI_ESP32.h:14:25: fatal error: hal/gpio_ll.h: No such file or directory
compilation terminated.
*** [.pio/build/esp32dev/lib831/TFT_eSPI/TFT_eSPI.cpp.o] Error 1
========================= [FAILED] Took 97.41 seconds =========================

Environment    Status    Duration
-------------  --------  ------------
esp32dev       FAILED    00:01:37.406
```
I recall facing the same error with other usermods and it was related to below  
https://github.com/Bodmer/TFT_eSPI/discussions/3048

Other usermods compile fine on older source 

### To Reproduce Bug

Uncomment the TFT_eSPI line within the [common] section as mentioned in the usermod readme 

### Expected Behavior

No compile error 

### Install Method

Self-Compiled

### What version of WLED?

0.15.0 (build 2412100)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
https://github.com/dosipod/WLED_15/actions/runs/12267662189/job/34228167624
```

### Anything else?

Take it easy 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2024-12-11:
@spiff72 FYI if you are the author of this

softhack007, 2024-12-16:
It looks like TFT_eSPI verison 2.5.33 is the last one that does not have a compile error with our WLED default framework.

```ini
lib_deps =
  ${esp32.lib_deps}
  TFT_eSPI @ 2.5.33  ;; this is the last version that compiles with the WLED default framework - newer versions require platform = espressif32 @ ^6.3.2
```

dosipod, 2024-12-16:
@softhack007 Yeah 2.5.33  seems to compile fine for the TTGO-T-Display usermod   .  I did not have a display at the moment to test though

spiff72, 2024-12-16:
I'm not sure I am much help here.  I haven't compiled this in a while as I haven't had a need, but I just tried with a fresh download and I am having trouble just getting the stock platformio.ini file to not error out on me.  "No option 'build_flags_esp32' in section "common".

I think some PlatformIO stuff has changed since the last time I tried to use it.

dosipod, 2024-12-16:
It is fine , seems to compile as listed above , I will be working on re-basing the other usermod from here
https://github.com/Victoare/WLED_TTGO-TDisplay-output

But in the current TTGO-T-Display the TFT_eSPI version  2.5.33 reference should be in the board environment selected and not 
globally   as softhack highlighted above , seems cleaner that way , once I get a board from someone I will try to clean both

spiff72, 2024-12-16:
> It is fine , seems to compile as listed above , I will be working on re-basing the other usermod from here https://github.com/Victoare/WLED_TTGO-TDisplay-output
>
> But in the current TTGO-T-Display [...]

Makes sense.  I think I resolved my platformio.ini file issue - it was due to my platformio_overrides.ini file having an error.

I'm a bit confused by the fact that it can't find the WiFi.h library (required in my usermod.cpp file).  Has the name of this changed causing this issue?

I also can't figure out why there's no section in the platformio.ini for determining a specific board to compile for.  My instructions in the readme.md for the usermod direct you to comment out the "default_envs" line (which lists every env), and to uncomment the in the single binaries section: "default_envs = esp32dev"

dosipod, 2024-12-16:
@softhack007  I mixed up old build info  but even using  TFT_eSPI version 2.5.33 the compile still fails 

https://github.com/dosipod/WLED_15/actions/runs/12358075274/job/34487771233

[image]

I will edit the post above to indicate that  , @spiff72 might be your can still look at fixing that

spiff72, 2024-12-16:
I"m stumped here - I can't even get WLED to compile "stock" without my usermod.

I'm also still not clear whether I am doing the compilation specifically for esp32dev correctly.  I tried deleting everything in the default_envs line in platformio.ini except for esp32dev, and it appeared to install a much older version of espressif32 (3.5 instead of the already installed 6.9.0).

Clicking the "build" icon at the bottom does nothing, so I tried picking a different env from the bottom status bar in VS Code and picked esp32dev.  It then attempts to compile but errors out with some wled_server errors.

`Processing esp32dev (board: esp32dev; platform: espressif32@3.5.0; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
*** use existing my_config.h ***
npm run build
'npm' is not recognized as an internal or external command,
operable program or batch file.
*** Error 1
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.5.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 0.0.0+sha.4b0c087
 - tool-esptoolpy @ 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- LittleFS_esp32 @ 1.0.6+sha.f0817ca
|-- AsyncTCP @ 1.2.0+sha.38b8475
|-- FastLED @ 3.6.0
|-- IRremoteESP8266 @ 2.8.2
|-- NeoPixelBus @ 2.8.0
|-- ESPAsyncWebServerWLED @ 2.2.1+sha.6d2540d
|-- QuickEspNow @ 0.6.2+sha.aa61f26
|-- arduinoFFT @ 2.0.1
|-- Wire @ 1.0.1
|-- SD(esp32) @ 1.0.5
|-- SD_MMC @ 1.0
|-- SPI @ 1.0
|-- EEPROM @ 1.0.3
|-- ArduinoOTA @ 1.0
|-- DNSServer @ 1.1.0
|-- ESPmDNS @ 1.0
|-- WiFi @ 1.0
|-- ESP32 Async UDP @ 1.0.0
|-- WebServer @ 1.0
Building in release mode
Compiling .pio\build\esp32dev\libcd5\FastLED\FastLED.cpp.o
Compiling .pio\build\esp32dev\libcd5\FastLED\bitswap.cpp.o
Compiling .pio\build\esp32dev\libcd5\FastLED\colorpalettes.cpp.o
Compiling .pio\build\esp32dev\libcd5\FastLED\colorutils.cpp.o
Compiling .pio\build\esp32dev\libcd5\FastLED\hsv2rgb.cpp.o
Compiling .pio\build\esp32dev\libcd5\FastLED\lib8tion.cpp.o
Compiling .pio\build\esp32dev\libcd5\FastLED\noise.cpp.o
Compiling .pio\build\esp32dev\libcd5\FastLED\platforms.cpp.o
Compiling .pio\build\esp32dev\libcd5\FastLED\platforms\esp\32\clockless_rmt_esp32.cpp.o
Compiling .pio\build\esp32dev\libcd5\FastLED\power_mgt.cpp.o
Compiling .pio\build\esp32dev\libcd5\FastLED\wiring.cpp.o
Compiling .pio\build\esp32dev\src\FX.cpp.o
Compiling .pio\build\esp32dev\src\FX_2Dfcn.cpp.o
Compiling .pio\build\esp32dev\src\FX_fcn.cpp.o
Compiling .pio\build\esp32dev\src\alexa.cpp.o
Compiling .pio\build\esp32dev\src\bus_manager.cpp.o
Compiling .pio\build\esp32dev\src\button.cpp.o
Compiling .pio\build\esp32dev\src\cfg.cpp.o
Compiling .pio\build\esp32dev\src\colors.cpp.o
Compiling .pio\build\esp32dev\src\dmx.cpp.o
Compiling .pio\build\esp32dev\src\e131.cpp.o
Compiling .pio\build\esp32dev\src\file.cpp.o
Compiling .pio\build\esp32dev\src\hue.cpp.o
[... 57 lines cut ...]
wled00/wled_server.cpp:3:21: fatal error: html_ui.h: No such file or directory

*****************************************************************
* Looking for html_ui.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:html_ui.h"
* Web  > https://registry.platformio.org/search?q=header:html_ui.h
*
*****************************************************************

compilation terminated.
*** [.pio\build\esp32dev\src\wled_server.cpp.o] Error 1
wled00/wled.cpp: In member function 'void WLED::handleConnection()':
wled00/wled.cpp:925:23: warning: unused variable 'nowS' [-Wunused-variable]
   const unsigned long nowS = now/1000;
                       ^
=================================================================== [FAILED] Took 28.50 seconds ===================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32dev       FAILED    00:00:28.495
============================================================== 1 failed, 0 succeeded in 00:00:28.495 ============================================================== 

 *  The terminal process "C:\Users\mgobl\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp32dev'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. `

spiff72, 2024-12-16:
I got mine to compile (finally).  I had some housekeeping to do - including getting npm installed.

This might be restating what was already mentioned above, but I had to do the following:
- Start with a fresh download of WLED from github
- Copy the usermod.cpp from my usermod folder to the WLED00 folder
- DON'T copy over the platformio_overrides.ini file from my usermod folder
- In the platformio.ini file, uncomment the line under lib_deps to the following:
```
  #For use of the TTGO T-Display ESP32 Module with integrated TFT display uncomment the following line
    TFT_eSPI @ 2.5.33
```
- In platformio.ini, change the default_envs line to this:
`default_envs = esp32dev`
- In the usermod.cpp file, there is a variable called lastUpdate, initialized to zero.  Rename it to lastUpdate_mod as shown below and correct the name in the following section as shown here:
`long lastUpdate_mod = 0;
long lastRedraw = 0;
bool displayTurnedOff = false;
// How often we are redrawing screen
#define USER_LOOP_REFRESH_RATE_MS 5000

void userLoop() {

  // Check if we time interval for redrawing passes.
  if (millis() - lastUpdate_mod < USER_LOOP_REFRESH_RATE_MS) {
    return;
  }
  lastUpdate_mod = millis();`
- At the end of the usermod.cpp file, comment out the tft.print statement containing the currentMilliamps function and replace with placeholder print statement:
```
  //tft.print(strip.currentMilliamps);
  tft.print("test ");
```
- Follow the directions in my original readme.md file (in my usermod folder) for the process to configure the User_Setup_Select.h file.

The above allowed me to compile successfully, but I can't confirm the resulting file works because I don't have a TTGO board handy.  It may have also been necessary to NOT have powershell terminals open in VS Code when compiling - I swear that was causing me issues too - if I had command prompt "terminals" open it was fine.  I also couldn't run the npm install command from powershell terminals - I had to use command prompt.

softhack007, 2024-12-17:
@spiff72 maybe it's a good time that I get you a bit up-to-date about how to compile WLED - a few things have changed since the last 3 years. It's quite late in Germany where I live, so just a few hints for now. I can give you more info tomorrow.

1. You need to install node.js and git commandline before compiling (looks like you have that part already. Before the very first build - after downloading the source code - you need to run `non install` once. Details are in the KB https://kno.wled.ge/advanced/compiling-wled/#preparations (ignore everything except the "preparations" section).

2. Instead of fiddling with `default_envs = `, you can select the buildenv you want from the VSCode UI. Short how-to https://github.com/Aircoookie/WLED-Docs/issues/161

3. Before anything else, try to build the standard `esp32dev`. This will confirm your tools are all set up and ready.
4. Do not change platformIO.ini.
5. Now you can create a very short platformio_override.ini for your usermod. The trick is to use `extends = esp32dev`. Look at examples in platformio_override.sample.ini to understand how to add build_flags and lib_deps to your own build env (this is the part where I have to get some sleep first ;-) it's late now).

spiff72, 2024-12-17:
Thanks!

I will do some more digging in the next few days as I get some more time.

It seems that a lot has changed, indeed - libraries have changed, procedures have changed, etc! There was a method to grab the currentMilliamps in there before, but that may have gone away, too. I used this to display the current draw on the T-Display screen as it was running…

dosipod, 2024-12-17:
@spiff72 We do not actually use this or other usermods but testing just part of 15 confidences compile verification as we have seen few usermods that was not tested on latest and would fail mainly  due to  libraries .At the moment I do not also have the HW to test functionality so the focus as the title of this post is just compile testing 

Once you are okay with the current changes please post your github repo with the changes so we or others could PR  it but  it would be best if  a full test with the HW is done  prior to PR which is not possible from our side now  but I have ordered some and hopefully would be there soon as we have another usermod here https://github.com/Victoare/WLED_TTGO-TDisplay-output  we would also like to fix PR

spiff72, 2024-12-24:
I just synced my fork with the latest WLED, and completed a pull request that I believe fixes the issues.  I also tested it with the physical board to make sure it worked (it didn't, initially). 

I had to do a few things:
- Used a more in-depth platformio_override.ini file to incorporate all of the changes needed for this board (including a new environment - which I hope I did correctly).
- Forced an older version of the TFT_eSPI library (inside my platformio_override.ini file)
- Fixed the readme.md file
- I attempted to include an override for the data_pins build flag that is supposed to control which pin serves as data pin for the LED string.  This needs to be pin 2 for this board - not the default of pin 16 (which is needed for the built-in display.  It took me a LONG time to figure out that conflict!  Unfortunately even though I am setting data_pins = 2 in the override file, it doesn't seem to work and it shows up as 16 when you connect to the board to configure WLED.  Once I correct this in the WLED configuration and restart the board, the screen comes to life.  (It will be black until the LED pin is corrected in the interface).

https://github.com/Aircoookie/WLED/pull/4422/commits/1466dde60fab5e81dc9a47abdf6d29406badf347

spiff72, 2024-12-24:
I thought I had this working but wanted to make one more tweak and now I'm finding that compilation is failing because of a missing SPIFFS.h file.  I need to do some more work yet...

spiff72, 2024-12-24:
One more try then I'm throwing in the towel for a while...
https://github.com/Aircoookie/WLED/pull/4423/commits/79f3ef97b7da75f42603389a237a2b93b2e89133

dosipod, 2024-12-24:
@spiff72 I appreciate your effort really and please take your time and you can even throw the towel as the issue with TFT_eSPI library is not limited to your usermod

PJMCL1618033, 2025-03-16:
I have been able to compile and upload this Usermod to three Lilygo T-Display (TTGO T-Display) boards. I made a .7z file that contains;  a platformio.ini file, a .pio folder that has the modified TFT-eSPI library and a wled00 folder with the usermod.cpp and my_config.h files. 

Basically  I followed the instructions from spiff72 post on Dec 16, 2024. I combined all those changes into a folder that can be drag-and-dropped into the WLED-main folder and then compiled. If anyone is interested I will post the copy2WLED-folder.7z file to my github repository.

willmmiles, 2025-03-16:
There were some PlatformIO bugs in handling headers for library compiles that were recently fixed in platformio 6.1.16 and 6.1.17 -- updating to a newer platformio fixed some issues with incorrect includes in libraries that were blocking other usermods from compiling correctly.

IIRC this one was one of the ones that suffered from that problem.  I'll review the current state of affairs in the my next usermod system PR.

spiff72, 2025-03-16:
I have made some adjustments to the usermod (I updated the readme for it, along with some other adjustments that incorporated all of the required changes into the platformio_override.ini file).  It does force an older version of the display driver library, though.

willmmiles, 2025-03-18:
@spiff72 I've merged your changes with current main, and integrated this module with the new `custom_usermods` system.  (PR #4600)   The dependent library will be automatically included, and it should set the correct board using a little script, so it doesn't have to be modified.   Unfortunately I can't test that it actually does anything beyond build as I don't have this board.  Would you mind taking a look and let me know if it's working well?

spiff72, 2025-03-20:
> [@spiff72](https://github.com/spiff72) I've merged your changes with current main, and integrated this module with the new `custom_usermods` system. (PR [...]

Hello - thank you!

I just uploaded from VS Code to a TTGO T-Display module and it works correctly.

closed 2025-08-10 by github-actions[bot]