wled/WLED #4376: Failing to read map and gaps after update from 0.14.4 to 0.15.0

blind-coder, 2024-12-11:
### What happened?

After updating from 0.14.4 to 0.15.0, WLED fails to read 2d-gaps.json and ledmap.json randomly. I need to go to Config -> LED Preferences and hit Save a random number of times before the files get properly read.

### To Reproduce Bug

Update from a working 0.14.4 ESP32 deployment with 2d-gaps.json and ledmap.json to 0.15.0
Enable any effect or check in Peek. Only a long LED strip will be visible, no matrix.

### Expected Behavior

Succeed in reading 2d-gaps.json and ledmap.json every time.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

[2d-gaps.json](https://github.com/user-attachments/files/18091774/2d-gaps.json)
[ledmap.json](https://github.com/user-attachments/files/18091775/ledmap.json)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blind-coder, 2024-12-11:
Sorry I can't provide logs or full version number, I rolled the ESP32 back to 0.14.4 already. The problem disappeared after rolling back.

softhack007, 2024-12-11:
@blind-coder can you share your 2d-gaps.json and ledmap.json files, plus cfg.json from 0.14.4?
This might help to reproduce the problem.

softhack007, 2024-12-11:
@blind-coder sorry, missed that you alread attached them 😅 
Please just add your cfg.json

blazoncek, 2024-12-11:
FYI gap and ledmap cannot coexist at the same time on a device.

blind-coder, 2024-12-11:
Here's the cfg.json
[cfg.json](https://github.com/user-attachments/files/18100513/cfg.json)

@blazoncek that new format may be the issue. I have 15 LEDs in 5 rows, it's a bit weird. The setup with both ledmap and gap has worked fine on 0.14.4.
```json
{"map":[
  -1,-1,14,-1,12,-1,10,-1,-1,
  -1,15,-1,13,-1,11,-1, 9,-1,
   0,-1,-1,-1,-1,-1,-1,-1, 8,
  -1, 1,-1, 3,-1, 5,-1, 7,-1,
  -1,-1, 2,-1, 4,-1, 6,-1,-1
]}
```
if you look at the map, each LED is a hexagon, and together they form "two overlapping ∞" if that make sense. I can provide a photo if it helps. The LEDs are connected in series.

blazoncek, 2024-12-11:
Ledmap and gap file are incompatible. Gap file is ignored if default ledmap exists. You can have non-default ledmaps and use gap file but once ledmap is selected gap file will be ignored.
If you want default 2D ledmap you'll need to update ledmap file to include width and height. Gap file and non-default ledmaps have no such restrictions.

blind-coder, 2024-12-17:
This seems to be working now with the width and height values. Thank you!
Adding this information to https://kno.wled.ge/advanced/mapping/#examples might be helpful.

closed 2024-12-17 by blind-coder

blazoncek, 2024-12-17:
@blind-coder I agree and you can add it to help the project.
https://github.com/Aircoookie/WLED-Docs