wled/WLED #4377: Pallette colors not working anymore

rVlad93, 2024-12-11:
### What happened?

After upgrading to 0.15 version the color palette doesn't work anymore, instead setting the strip to one single color. I just rolled back the to the previous stable one and it works as expected.

### To Reproduce Bug

Select any color pallete

### Expected Behavior

Light the strip from one color to the other with the respective gradiend in between 

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2024-12-11:
can you give more exact details on how to reproduce this?
- number of LEDs / general setup
- 1D or 2D
- which FX is this happening on 
th

rVlad93, 2024-12-11:
- 225 WS281x leds
- 1D
- static Palette

After updating the same preset doesn't produce the same result. Orange&Teal gives me just and orange color across all leds instead of the teal to orange gradient.

DedeHai, 2024-12-11:
the effect was updated to support 2D as well, you may need to change your slider settings.

rVlad93, 2024-12-11:
Indeed. Turning the size slider to half did the trick.

closed 2024-12-11 by rVlad93