wled/WLED #4379: Cant compile

alexbst, 2024-12-11:
### What happened?

html_ui.h not found in source package 

### To Reproduce Bug

Download source package and try to build

### Expected Behavior

Try to build with platformio

### Install Method

Self-Compiled

### What version of WLED?

0_15

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Relevant log/trace output

```shell
Compiling .pio\build\esp32_eth\lib38e\IRremoteESP8266\IRsend.cpp.o
wled00/wled_server.cpp:3:21: fatal error: html_ui.h: No such file or directory
Compiling .pio\build\esp32_eth\lib38e\IRremoteESP8266\IRtext.cpp.o
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DedeHai, 2024-12-11:
please use discord or the forum for support questions. use the npm commands as mentioned here: https://kno.wled.ge/advanced/compiling-wled/

softhack007, 2024-12-11:
> wled00/wled_server.cpp:3:21: fatal error: html_ui.h: No such file or directory

@alexbst quick answer: you need to open a terminal window in VSCode, and type `npm install`. This will install the tools needed to generate UI files. 

https://kno.wled.ge/advanced/compiling-wled/#preparations

blazoncek, 2024-12-12:
Github is not the place for any compilation related questions.
Please use WLED [forum](https://wled.discourse.group) or [Discord](https://discord.gg/FMvz8nzmat) for help and support questions.

closed 2024-12-12 by blazoncek