wled/WLED #4380: SK6812 RGBW colors broken in 0.15.0 on ESP32, not ESP8266, works in 0.14.4

trullock, 2024-12-11:
### What happened?

I have two strips of SK6812s RGBWs, one lot attached to an ESP32 and one to an ESP8266

Updating to 0.15 from 0.14.4 breaks the ESP32 colors but not the ESP8266 colors

I've reverted to 0.14.4 so cant debug further at this moment but can do later

### To Reproduce Bug

SK6812s RGBWs in GBR mode on an ESP32, the colors are the wrong. They are very dim, the hue isnt far off or might be right (havent fully debugged, had to quickly revert before the kids came home to see no xmas lights!)

### Expected Behavior

Colors to be correct ;)

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
can probive later, got to
```

### Anything else?

Can debug further later, though Id get this raised for everyones sake

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

willmmiles, 2024-12-11:
>  very dim

Try checking the brightness limiter settings.  Also, if you can, please post the config from your 0.14 system; it sounds like there's either (a) a bug in the upgrade path, or (b) a setting that was previously ignored is now being correctly applied; I'd like to poke the config and see if I can work out what it is.

listlesslife, 2024-12-11:
I was having a similar problem, but my esp8266s weren't changing colors and my esp32s were fine. Rolled back the 8266s to 0.14.4 and now color changing works. 0.15.0 also knocked a couple of my esp8266s offline and they had to be reflashed via serial, but that's a different issue.

trullock, 2024-12-12:
So, automatic brightness limiting had been turned on, turning that off solves it.

However the effects wrt segments is screwy, like its sped up

I have a vid but cant post it here. Can DM/email

nomonkeynodeal, 2024-12-12:
It looks like the blue channel just doesn't work for me with 15.0. I rolled back to 14.4 and it's fine.

trullock, 2024-12-13:
0.15 now seems completely broken for me, and I can't roll back to 0.14.4. I provide the binary and seems to apply the update, but it doesnt. 

Will have to get the ladders out and flash it directly I guess

c-b5r, 2024-12-14:
I updated (OTA) 5x ESP8266 (d1 mini) with SK6812 setup. Here is the result:

- 2x updated just fine and everything worked as before
- 2x updated but the entire color setup (RGB wheel + sliders) was missing. I then switched the LED configuration to [WS2812x > apply > SK6812 > apply] and after that everything seemed back to normal.
- 1x device didn't boot after the update. I manually flashed 0.15.0 via USB and then the device came back online again with all the previous settings

Deretour, 2024-12-27:
> It looks like the blue channel just doesn't work for me with 15.0. I rolled back to 14.4 and it's fine.

Same for me. For SK6812 strips I cannot get any blue to show up. Not even when intentionally switching color order.
After reverting to 0.14.4 everything works fine again.

Affected binary: WLED_0.15.0_ESP32_Ethernet.bin
Working binary: WLED_0.14.4_ESP32_Ethernet.bin

Upgrade performed via Home Assistant.
Downgrade performed via WLED's "Manual OTA Update"

ijspear, 2024-12-28:
@nomonkeynodeal @Deretour How many channels are you using? I think this is the same bug as reported in #4380 and #4384

trullock, 2024-12-28:
> 0.15 now seems completely broken for me

By freak coincidence, my PSU died of a faulty adjuster pot at exactly the same time. So 0.15 is not "completely broken" but the color/brightness and rollback updates are real

nomonkeynodeal, 2024-12-28:
> @nomonkeynodeal @Deretour How many channels are you using? I think this is the same bug as reported in #4384

Maybe. I am using four channels, but the white channel definitely worked for me.

Deretour, 2024-12-28:
> @nomonkeynodeal @Deretour How many channels are you using? I think this is the same bug as reported in #4384

Sounds familiar indeed. Except for blue instead of white. But besides that it seems exactly the same issue.
I use 2x ws2812b RGB (144 pixels each) and 4x sk6812 RGBW (70 pixels each) which totals to 6 channels/outputs.

additional info: I‘m running a quinled dig octa.

ijspear, 2024-12-28:
@Deretour try something. Delete a channel, any channel, so you have a max of 5 channels on your Octa. Do the other strips start behaving properly when there are only 5 channels?

Deretour, 2024-12-28:
> @Deretour try something. Delete a channel, any channel, so you have a max of 5 channels on your Octa. Do the other strips start behaving properly when there are only 5 channels?

OK, I couldn't wait. :)

Gave it a try and it's exactly as expected: deleting one channel brings back blue for the remaining sk6812 channels.
Unfortunately due to my wiring setup connecting two of the sk6812 strips to share one output/channel is not an option though.
So I'm back on 0.14.4 for now.

I hope the test rseult helps narrow down the issue and it can be resolved in an upcoming release as there are definitely features in 0.15 that I'm looking forward to.

ijspear, 2024-12-28:
@Deretour that’s what I figured. This is def a bug. I’m looking into it now too but not seeing anything glaringly obvious… I’m sure @Aircoookie will have this knocked out soon.

Deretour, 2024-12-28:
Additional note for completeness:
As OP mentioned there are actually two issues: additionally to the missing blue/white when using 6+ channels after updating to 0.15.0 strips are also very dim (almost off in my case). This is due to the fact that current limiter seems to be on by default after upgrading.
In that case best solution is to enter reasonable values for the limiting current and current per LED. ...or as I did: just disable current limitation. But only if you're fully aware of what you're doing! Otherwise you could blow up your PSU/fuse.

Deretour, 2024-12-28:
> @Deretour that’s what I figured. This is def a bug. I’m looking into it now too but not seeing anything glaringly obvious… I’m sure @Aircoookie will have this knocked out soon.

From my limited understanding I'd assume a memory/buffer/array-size limitation.
Probably "something" (array/buffer) using more more memory/space than before and exceeding 5 channels somehow breaks that limit. (I could imagine something with CW + WW support or something, but that's jsut a wild guess)

blazoncek, 2025-01-03:
For anyone having issues with mixed LED types and multiple outputs (>5) please try the following:
- reorder your outputs so that 1st uses SK6812 (or WS2805), i.e. RGBW strip
- reduce the number of outputs to 5 or less
- increase LED count on one of the outputs to more than 300 (even if actual LEDs are fewer)

Report back with detailed findings.

(Reposting from #3484 and BTW @aircoookie is taking a break from WLED)

ijspear, 2025-01-03:
@blazoncek thanks so much for looking into this. I can report that decreasing channel count to 5 or less and everything works perfectly. I haven’t tried increasing to 300+ on a channel. I’ll do that in a few hours and report back.

ijspear, 2025-01-03:
@blazoncek I set one of the existing 5 channels to 301 LEDs and then added a 6th, 7th, and 8th channel. Everything is now working properly! I get all 4 colors and where behavior I previously reported is gone. I cannot tell you how much I appreciate your input. Can I test anything else for you to help further?

blazoncek, 2025-01-03:
@ijspear please reorder outputs so that 1st has SK6812. You should also reduce LED count below 300 on each output.
This will help narrow down the root cause.

blazoncek, 2025-01-03:
> actually all of my outputs are SK6812

Now that's odd. In such case parallel I2S should work normally except if there is a bug in low level driver.

blazoncek, 2025-01-04:
If anyone can compile on their own, try replacing version number in `platformio.ini` for `makuna/NeoPixelBus @ 2.8.0` with `makuna/NeoPixelBus @ 2.8.3` if that will solve the issue.

ijspear, 2025-01-04:
@blazoncek i’ve never compiled on my own before but I’ll try to figure it out. Thanks for the suggestion!

blazoncek, 2025-01-04:
There are CI build artefacts of my fork here: https://github.com/blazoncek/WLED/actions/runs/12611034886
They include NPB 2.8.3

ijspear, 2025-01-04:
@blazoncek i know this is a super noob question - which file should I use for the digi-Octa?

blazoncek, 2025-01-06:
ESP32 as it is the only chip with issues.

ijspear, 2025-01-06:
@blazoncek so esp32_eth would be the one to try?

blazoncek, 2025-01-06:
IDK, if you need Ethernet then yes. But for WiFi plain will do,

blazoncek, 2025-01-07:
Anyone facing this issue, if you have logic analyser please capture a frame on a 10 pixel output (solid color).
It was requested by @makuna to help resolve the underlying problem (if it exists in the driver).

DedeHai, 2025-01-08:
as requested by @Makuna: I could not set distinct hex values so I set all four channels set to about 80%. I set the first output to 1 pixel, the others to 10. With I2S parallel output, the white channel outputs zero:
[image]

setting one of the outputs to 400LEDs to enforce RMT it looks correct (and white LED is on):
[image]

Makuna, 2025-01-08:
Things of note, the color channels have different values even though they were set to the same (225, 223, 220)?  Should be unrelated as the RMT shows the same values.

What versions of NeoPixelBus are being used with working and no-working versions of WLED?

 I will investigate.
https://github.com/Makuna/NeoPixelBus/issues/868

blazoncek, 2025-01-08:
@Makuna reports were made on NPB 2.8.0, I've requested tests on 2.8.3 but did not receive any feedback.
@DedeHai can you test that?

DedeHai, 2025-01-08:
can confirm the behaviour is the same with NPB 2.8.3
@blazoncek I also tested your branch on S2, 5 outputs, SK6812 RGBW, same issue: white channel is zero.

ijspear, 2025-01-08:
@DedeHai just to clarify - using 2.8.3 can you add a 6th channel and make sure each channel is set to less than 300 pixels?

DedeHai, 2025-01-09:
@ijspear 2.8.3 with the (preliminary) implementation in WLED is still 5 channels max.

blazoncek, 2025-01-09:
Just for clarification:
When you configure 6 outputs with 300 LEDs or less on (classic) ESP32, WLED switches to parallel I2S use for first 8 outputs instead of mono I2S for 1st output followed by 4/8 RMT outputs otherwise.
The 300 limit is implemented to prevent RAM starvation due to huge memory requirements for I2S outputs.

Makuna, 2025-01-11:
The big change was to move to the 3Step Cadence to save memory.  It still supports 4Step Cadence, but it requires a compile time flag (NPB_CONF_4STEP_CADENCE).  Does WLED use 4 or 3 step?

ijspear, 2025-01-11:
@Makuna, @blazoncek
Perhaps it might be helpful to summarize the exact behavior to get closer to a fix. I’m currently using the latest version of the Digi-Octa running the official v15.0 with all 8 channels driving SK6812 RGBW (BRG, xW&G) led strips. The pixel counts are as follows: ch1: 12, ch2: 74, ch3: 12, ch4: 12, ch5: 10, ch6: 10, ch7: 12, ch8: 23.

When I try to use this exact configuration here is the behavior using the color wheel:
a command of 100% Blue light makes all strips turn completely off
a command of 100% Red light makes all strips turn red
a command of 100% green light makes all strips turn green
a command of 100% white (dead center of wheel) makes all strips turn a combination of green and white

When I change the pixel count to any of the channels (I chose ch2) to 300 or more, normal behavior is restored to all the strips.

I also tried a version with NPB 2.8.3 and the behavior was exactly the same.

ijspear, 2025-01-11:
@blazoncek awesome! Sorry to be redundant. Super thankful to have folks like you as contributors to the project. Anything else I can help test to push the ball further?

blazoncek, 2025-01-11:
Anyone experiencig this issue (no white channel on SK6812 LEDs when using more than 5 outputs) please use the following workaround until this is fixed:

1. Configure one of the outputs so that it has more than 300 LEDs (it can be unconnected one with no physical LEDs attaced and it can use any (unused) GPIO) or
2. Reconfigure/rewire LEDs so that less than 6 outputs are used
3. configure 8 dummy outputs followed by correct ones

Any extraneous LEDs can be hidden by adjusting segment size.