wled/WLED #4383: Estimated Current Isn't Reported When Ooff Since 0.15.0

AlexKusnezov, 2024-12-12:
### What happened?

Estimated Current Isn't Reported When Ooff Since 0.15.0:
[image]

0.14.0:
[image]

Brighness limiter was on in both cases
After reverting back to 0.14.0 brightness limiter got disbaled so I had to re-enable it manually

Also on Athom bulbs 0.15.0 doesn't report estimated current at all!

### To Reproduce Bug

 - update to 0.15.0
 - look at "estimated current"

### Expected Behavior

Estimated Current should be prperly reported

### Install Method

Binary from WLED.me

### What version of WLED?

0.15.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-12-12:
0.15.0 uses new ABL, please update settings and verify again.

AlexKusnezov, 2024-12-12:
> 0.15.0 uses new ABL, please update settings and verify again.

Same
unchecked ABL, reboort, check ABL set ma, reboot, result:
"Not Caclulated" when off

blazoncek, 2024-12-12:
That is expected. ABL calculation has been moved into LED output driver to support per output limitations.

AlexKusnezov, 2024-12-12:
@blazoncek got it, sadly that way the consuption of the esp itself is lost.

blazoncek, 2024-12-12:
ESP current is 120mA constant and irrelevant  when LEDs are off as far as calculation goes.

Riscue, 2024-12-16:
it's not reported to Home Assistant Integration also, lost my power consumption values after 0.15.0

AlexKusnezov, 2024-12-16:
> it's not reported to Home Assistant Integration also, lost my power consumption values after 0.15.0

that's right, same here.

DedeHai, 2024-12-16:
@blazoncek  would it be possible to just report the 0.12A fixed instead of 'not calculated'? Or is there any good reason to break backward compatibility with HA? If this is something that is not possible with the new ABL, then this needs to be reported to HA to implement a fix for the new behaviour IMHO.

closed 2024-12-17 by DedeHai

AlexKusnezov, 2024-12-17:
> @AlexKusnezov can you check if its ok now? latest bin with a fix is here: https://github.com/Aircoookie/WLED/actions/runs/12378696924

thanks, looks good.

PR #4402 (2024-12-17): Idle current bugfix
Fixes #4383