wled/WLED #4384: SK6812 RGBW- White LED channel won't turn on

BrickSolid28, 2024-12-12:
### What happened?

Using a DIg Octa board.  After updating to WLED 15.0 the white LEDs in a SK6812 RGBW strip won't turn on.  All the other LED channels work correctly but the white LEDs won't turn on.  They were working correctly on 14.2.

### To Reproduce Bug

Tried updating from 14.2 as well as doing multiple clean installs.  Tried using the bin file from multiple sources.  White LED's won't turn on.

### Expected Behavior

White LED's to turn on when the white channel slider that's under the color pallet is adjusted.

### Install Method

Binary from WLED.me

### What version of WLED?

15.0

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

BrickSolid28, 2024-12-12:
I can get white using a combination of RGB but no light is coming from the actual white LED's.  I've tried all of the options under "Auto-calculate W channel from RGB:" in the settings menu but still nothing comes from the actual white LED's.

ijspear, 2024-12-28:
I believe I am experiencing this behavior as well. I found that when adding less than 5 channels on my Octa everything works as expected (BRG, Swap W & G). As soon as I add the 6th or more channel(s) on their respective GPIO (6, 12, 13) I lose the blue light all my channels and when blue is targeted as the color, all strips just turn off. As soon as I delete that channel normal behavior resumes. v15.0

mtkraai, 2024-12-28:
I am also having this issue.  My sk6812 strip is on output 7 of 8 (all 8 are used) on my Dig-Octa board.  Same as the other reporters, RGB outputs work fine, just no output on the white channel.  Color order on this strip is GRB, swap: None.

All the other strips in my setup are 3 channels each.

ijspear, 2024-12-28:
@mtkraai if you delete a few of the other channels so your total channel count is 5 or less does your white come back?

trullock, 2024-12-28:
As mentioned in the  referenced issue above, try turning the auto current limit setting on/off. Dont blow your PSU up though!

blazoncek, 2025-01-03:
For anyone having issues with mixed LED types and multiple outputs (>5) please try the following:
- reorder your outputs so that 1st uses SK6812 (or WS2805), i.e. RGBW strip
- reduce the number of outputs to 5 or less
- increase LED count on one of the outputs to more than 300 (even if actual LEDs are fewer)

Report back with detailed findings.

BrickSolid28, 2025-01-04:
Blazoncek- Thanks for the tips! I had gone back to 14.2 and had no issues. I’ve reinstalled 15.0 and ran into the same issues with the white channel. I removed LED outputs until I had less than 5 and the white channel works again! I still have the brightness limiter turned on (30,000 ma) and have multiple outputs that are under 300 lights but it still worked. Increasing the number of output channels to 6 made the white channel stop working again. There must be some kind of bug with the number of channels being used and the white channel output.

blazoncek, 2025-01-04:
@BrickSolid28 it is not a "bug" with number of channels. Please see #4380 for additional info.

BrickSolid28, 2025-01-04:
@blazoncek It’s not practical in my set up to increase all of the channel outputs to be over 300 lights but I can increase the last channel. Increasing just the last led channel to be over 300 lights made ALL of the channels work again. Even when over 5 led outputs are used.

mtkraai, 2025-01-06:
@ijspear @blazoncek Like everyone else (here and #4380), if I increase a channel to 300+ length, it works as expected.  Even a channel unrelated to the problem.  Also, if I only have 5 channels, the RGBW channel works correctly.

blazoncek, 2025-01-06:
I am closing this issue as a duplicate of #4380. Please continue discussion and other reports there.
Anyone interested in testing newer version of NPB see link I posted there.

closed 2025-01-06 by blazoncek