wled/WLED #4389: WLED 0.15.0 Introduces Flashing/Noise on GPIOs 1, 16 on QuinLED-Dig-Quad

hacustom, 2024-12-13:
### What happened?

After upgrading a QuinLED-Dig-Quad to WLED 0.15.0, LED segments on GPIOs 1 and 16 experience significant noise and flashing.  An LED segment on GPIO 3 operates fine without any issue.

Segment 1 GPIO 3 - 629 LEDs
Segment 2 GPIO 1 - 161 LEDs
Segment 3 GPIO 16 - 171 LEDs

This particular setup has been running flawlessly for just over 3 years.  The issue was immediately resolved by downgrading to WLED 0.14.3.

### To Reproduce Bug

- Upgrade a QuinLED-Dig-Quad to WLED 0.15.0
- Activate any preset or animation
- Flashing/Noise is prominent on LED segments on GPIO 1 and 16

### Expected Behavior

- No noise/flashing on any LED segment

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.15.0 2412100

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
I don't have any logs at the moment, I have since downgraded to WLED 0.14.3.  However, I would be more than happy to re-upgrade and capture any logs/traces/configs that would aid in troubleshooting.
```

### Anything else?

This appears to be similar to report #3976.  That was submitted for 0.15.0b2 and not the full release however.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2024-12-13:
GPIO 1 & 16 will use RMT. RMT is susceptible to flickering in some circumstances.
0.15 added a lot of code that may make those circumstances more frequent.

GPIO 3 uses I2S.

hacustom, 2024-12-13:
Interesting.  

I had a spare Dig-Quad on hand, and I upgraded it to WLED 0.15.0 and configured the LED outputs to be identical as the main one for my house lights.  I wired up a spare string of LEDs I had leftover and I'm seeing the same behavior as with my other - tons of flashing/noise.  If I stick to a solid color, it's OK.  There's a bunch of noise every time I change color but once it reaches the target color the noise stops.  Any animation is unusable.

hacustom, 2024-12-13:
Just for kicks, I setup my spare Dig-Quad from scratch, installing it from https://install.quinled.info/dig-quad/ just to see if it made any difference and it didn't.

blazoncek, 2024-12-14:
If you want to test (and help with resolving) set up more than 4 outputs with less than 300 pixels each and report.

eMeF1, 2024-12-14:
I encountered the same issue after upgrading to 0.15.0. There are 182 LEDs on GPIO 16, which are set to a solid orange-ish RGB[115,74,0] color. It's not apparent on solid white. I have a couple of dig2analog on GPIO 3, but I did not test whether the issue is also there. After downgrading to 0.14.3, the random flickering (every 30-60 seconds) issue is gone.

blazoncek, 2024-12-14:
Periodic issue leads me to believe this may be network related.

hacustom, 2024-12-14:
> If you want to test (and help with resolving) set up more than 4 outputs with less than 300 pixels each and report.

I will see if I can do that on the dig-quad as a second test.

kylewilliams321, 2024-12-14:
Same issue from me and similar setup as original reporter

intermittech, 2024-12-14:
If needed I can build a quick test setup and can troubleshoot on Discord too, apologies this made it through the beta somehow, I had not noticed in on any testing setups I had before. Let me know if you need anything!

blazoncek, 2024-12-14:
Please test a set-up with at least 6 outputs (WS281x). You do not need to have actual LEDs connected or GPIOs attached. It is important to have GPIO 1, 3 & 16 as first 3 outputs and each output must not exceed 300 LEDs

dokalma, 2024-12-14:
I had exactly the same problem on my digquad pushing 824 seedpixels per channel. 

Went back to 14.3 and all was good

hacustom, 2024-12-14:
> If needed I can build a quick test setup and can troubleshoot on Discord too, apologies this made it through the beta somehow, I had not noticed in on any testing setups I had before. Let me know if you need anything!

Oh my goodness, PLEASE don't apologize!  I have worked in IT & Software Dev for...ouch...nearly 30 years now?!  I've been on both sides of bug reports more times than I can count, I actually have grown to love really gnarly bugs and digging into them.

Especially with you all that contribute to open source projects, the work you do is amazing and usually pretty damn thankless.  The only intent behind my bug report is to share information and see if we can figure out the cause :) 

I'm working on re-configuring my spare Dig-Quad right now to test out @blazoncek suggestion.

hacustom, 2024-12-14:
OK, got my spare Dig-Quad setup like this:

6 LED Outputs, 190 LEDs on each output, WS281x, 30mA/12v.

Output 1 - GPIO 3
Output 2 - GPIO 16
Output 3 - GPIO 1
Output 4 - GPIO 4
Output 5 - GPIO 5
Output 6 - GPIO 13

I've been running a few different twinkle effects for ~10 minutes now without any apparent noise, using a single color and different color palettes just for testing.

hacustom, 2024-12-14:
Just for kicks I deleted Output 6 and the noise was immediately re-introduced.

Jean-Luc-Hauser, 2024-12-14:
Same issues with the Quad and 12V WS2815 on version 0.15 : heavy noises and flickering with effects such as Dancing Shadows.

GPIO 3 - 152 LEDs (60/m)
GPIO 16 - 356 LEDs (144/m)
GPIO 15 - Relay

I have a 5V PS for the relay and the Quad board and 12V for the LEDS through the board.

And I was not able to use the relay (GPIO 15) since it was taken by the I2S audio interface.

I have 2 GLEDOPTO GL-C-015WL-M controllers that have no issues with the 0.15 release. I went back to 0.14.3 for now, was scared to damage the LEDs

blazoncek, 2024-12-14:
with 6 outputs or more, 1st 8 are using I2S (parallel)
with 5 or less, 1st is using I2S (single) and 4 are using RMT.
may notify @makuna if he has any clue what's different in NPB 2.8.0 compared to 2.7.4 in 0.14.4 regarding RMT

travi, 2024-12-14:
I ran into what this sounds like with one of my digunos. I'm currently running 3 digunos and one digquad, but only saw this on one diguno and was able to fix it by swapping the esp from a spare diguno. 

A detail that I noticed was unique on that instance was that I had the output order switched around so that the first output was mapped in wled as gpio 3 and the second to gpio 16 instead of the other way around, for whatever reason. 

I imagine swapping the esp worked in my case only because that one had the outputs orders as gpio 16 then 3.

promess, 2024-12-14:
I'm dealing with this same issue, except it's on a single ESP32. There are 260 LEDs in this configuration, all of which are in GPIO 16.

it's only the last section where I'm adding power back in after a split and it's after that point that the LEDs are misfiring. I've posted video of it flashing blue and red after a few minutes in the discord channel. It was working fine in 14.4. I'm not sure if I'm sending too much power and the additional power is fucking with it or something. Let me know if there's anything you need me to do.

blazoncek, 2024-12-15:
FWIIW I'm using Dig-Quad with 16 (465), 1 (270), 3 (15) - in that order - with no flickering whatsoever.
Using generic ESP32 in Wemos D1 Mini size packaging from aliexpress.

Makuna, 2024-12-15:
> with 6 outputs or more, 1st 8 are using I2S (parallel) with 5 or less, 1st is using I2S (single) and 4 are using RMT. may notify @Makuna if he has any clue what's different in NPB 2.8.0 compared to 2.7.4 in 0.14.4 regarding RMT

The only real change with RMT between those versions is the ability to compile using IDF5.  This was just some minor headers include changes to react to the IDF change but still outputs warnings about deprecated API use.
Did you turn on ESP32 core debug errors/warnings/info level to see if anything shows up in the debug output?

JorinL, 2024-12-16:
Hey there,
I have the same problem on my 3 PWM strips.
When I cut the transitiontime it is gone - with transition it flashes when turning on before dimming up and when turning off while dimming down a flash and dimming further down.

I added a small video of the current situation.
https://github.com/user-attachments/assets/f6009672-1853-451c-b2fd-f404a0fe5acc

blazoncek, 2024-12-16:
@JorinL PWM is unrelated to digital flickering. Please open a new issue.

Woutch, 2024-12-17:
Hi All,

I have the same problem in my Chrismas tree. This is a Quinled Dig uno v3 like this:
https://shop.allnetchina.cn/products/quinled-dig-uno-v3r7-digital-led-controller

It feeds 2 strings starting from the middle and string one is the upper one and 2 is the lower one.
Using the latest v14.4 I have zero problem, upgrading to v15 gives the first 20 leds a noicelike color and keep flikkering, even in solid mode, but faster when an effect is activated. 

Output 1 is using gpio 1, and also tried 16, both same flickering.
Output 2 uses gpio 3, close to 0 problems. 

Limiting the brightness seems to help, but this is brightness at 10% and problems still are like 50% more ore less.

V14.4:
[image]

V15:
[image]

With effect:
https://github.com/user-attachments/assets/ef06e29d-556d-4a89-9d80-b0d910aa616f

Solid color:
https://github.com/user-attachments/assets/e6309d58-4a9c-48f4-b7d0-670ef472287c

If anyone needs more info, ask ;-)

Hope this helps,
Wouter

Zappelphilipp, 2025-01-17:
same problem here. using 699 LED's on an AZDelivery ESP32 on GPIO 16. LED's are usually dimmed to about 10% on a solid color, after the update i get basically random color flashing everywhere. 

By dimming the Power above about 50% the random color flashing stops but there is still some flickering everywhere (at least in the right color)

sadly a downgrade seems not to work because after that the ESP seems to crash after a few minutes.

hacustom, 2025-03-12:
Curious if anyone has had any luck with this?  I am still running an older version of WLED, not the biggest issue in the world, I am not exposing these externally, but I would like to keep current with them.

miketunes, 2025-04-06:
I also have this issue, asked about it on the Quinled discord and they said to check here.

Arcitec, 2025-04-10:
I was thinking of using ESP32-WROOM-32U pins 2, 4, 12, 16.

If I understand correctly, ESP32 has a library called "RMT" which is for controlling LEDs.

Is that library only being used with certain output pins, or certain combinations?

Unfortunately it's not fully clear to me if my pin choices will be affected by this situation.

blazoncek, 2025-04-10:
ESP32 has a GPIO muxer. Any pin can be anything: PWM, I2S, I2C, RMT etc.
BTW RMT is not a library. It is a DMA HW driver.

MakerMatrix, 2025-06-02:
Same issue here. Recent model dig-Quad running two strands of 12V BTF WS2814 (128 and 169 logical LEDs). LED data pins 1, 3, and 4 (GPIO 16, 1, and 4 respectively) all flicker badly on 0.15.0.  Moving the data pins around and swapping strips in and out shows that the issue is coming from the controller itself and not the LEDs or my wiring.

0.14.4 Audioreactive works very cleanly on all outputs.

Is this issue explained yet, if not fixed in 0.15.1 RCs?

blazoncek, 2025-06-03:
@MakerMatrix yes it was explained (elsewhere). GPIO don't matter but number of LEDs do. If you use more than 300 LEDs per output, WLED will use RMT peripheral for output instead of I2S (behaviour from 0.14).

blazoncek, 2025-06-03:
No. That's for currently released 0.15.0, for 0.15.1 you'll need to check the changelog.
To overcome the issue on 0.15.0 set last output to contain more than 300 LEDs. This will force 0.14 behaviour.

hacustom, 2025-06-03:
Ah, ok. I did misunderstand, I thought you meant that changing the pixels to be <300 per GPIO was going to be the only option in version 0.15.1 and beyond. Thank you for clarifying, much appreciated.

MakerMatrix, 2025-06-03:
> [@MakerMatrix](https://github.com/MakerMatrix) yes it was explained (elsewhere). GPIO don't matter but number of LEDs do. If you use more than 300 LEDs per output, WLED will use RMT peripheral for output instead of I2S (behaviour from 0.14).

300 logical LEDs or 300 physical?  My longest strip has 169 pixels/507 leds.

I'll give -beta2 a try and report back.

MakerMatrix, 2025-06-03:
OK I see it in the changelog for 0.16-alpha (the nightly build) so let's get that mentioned in here too.

WS281x leds flickering every firmware after WLED 0.15.0-b3 https://github.com/wled/WLED/issues/4211

MakerMatrix, 2025-06-03:
@intermittech would it be possible to get a custom build of 0.15.x that disables WLED_USE_PARALLEL_I2S to put on your git?  I think there is a chance that would fix it?

https://wled.discourse.group/t/using-wled-use-parallel-i2s/12243

intermittech, 2025-06-04:
Interesting, from my understanding the main issue was fixed in v0.15.1-beta2 so I'm not 100% sure what's going on, I will give it a try sometime soon (currently in a crunch to get a video out)!

willmmiles, 2025-06-04:
@blazoncek Right, let me walk it all back.  The problem I'm seeing is with the RMT driver -- I had the test inverted in my debug code.  Parallel I2S is rock solid by comparison. :)   (I do think there is a possible weakness to interrupt latency in NPB's I2S DMA logic; but this doesn't seem to be a problem in practice.)

The glitches I am seeing are induced by network traffic, and look to me like a FIFO underrun: at some point partway through the output, the data "slips" one (or more) bit times, resulting in shifted or corrupted colors.  Tellingly, the glitch events can cause LEDs past the end of the configured count to light.

I've reproduced this on a couple of different ESP32 (WROOM/WROVER) boards from different manufacturers.

I'm digging in to the RMT driver now.

phoenix1589, 2025-06-04:
I've got similar symptoms, which I had put down to a poor quality setup (it's not a QuinLED board, I don't have level shifters etc) however the recent comments especially around the 300 LED mark have led me to do a couple more tests.

Running WLED 0.15.0 (build 2412100) on a LILYGO ESP32-Wrover, with LEDs setup as:
219 LEDs on GPIO 26
130 LEDs on GPIO 3
33 LEDs on GPIO 23

with then 2 segments of 219 LEDs and 163 LEDs (i.e. output 1 as 1 segments and output 2+3 as the other segment)

I would get frequent flashes every few seconds. 

If I change the 2nd segment down to 80 LEDs (bringing the total to 299) then the flashing stops. If I then set the 2nd segment back to 163 - the flashing does not return. If I then change the effect, or turn off back on the outputs - the flashing returns.

Doing the change again, the flashing stops again.

blazoncek, 2025-06-04:
Nice chat (revealing). TL:DR you may want to force parallel I2S which is triggered by using more than 5 outputs.
Remember, they need to be less than 300 pixels in length.

When you have >5 outputs WLED will switch to parallel I2S for firs 8 outputs and use RMT for last 8 (if configured). Unfortunately parallel I2S uses a lot of buffers so we needed to enforce a limit of 300 pixels to prevent memory drain.

As far as GPIO go, all IO are equal (ESP32 uses GPIO muxer).

MakerMatrix, 2025-06-04:
I still don't completely understand (Is there a primer I can read about these different drivers?) but feel like we are making progress because....

I added four dummy outputs of 10 each, and then removed these 40 fake pixels from segment 0.  Then I swapped the GPIOs back to the configuration where I previously had flicker and...drumroll...NO FLICKER.

hacustom, 2025-06-04:
> That might be a revalation. If I had not seen this with my own eyes and spent 20 hours on troubleshooting, I'd not believe this. But here is the behavior on 0.15.0:
> 
> First LED output (169LEDs) = GPIO3 Second LED ouput (169LEDs) = GPIO16 GPIO16 flickers
> 
> First LED ouput = GPIO16 Second LED output = GPIO3 Neither one flickers!!
> 
> The moment I hit "Save" after switching them back to the top config in the UI, flickering on GPIO16 begins.
> 
> [@hacustom](https://github.com/hacustom) do you see this?

@MakerMatrix work was busier than I had anticipated, I didn’t have time to do an upgrade yet. I know I had tested a few scenarios out when I first ran into the issue but I don’t remember the specifics anymore. I think I documented some of it in my earlier comments. I only am looking on my phone right now so I might be missing it.

hacustom, 2025-06-05:
Installed WLED_0.15.1.beta2_ESP32.bin on my main Dig-Quad.  Flashing/flickering is still present on GPIO's 1 and 16.  

Segment 1 GPIO 3 - 629 LEDs
Segment 2 GPIO 1 - 161 LEDs
Segment 3 GPIO 16 - 171 LEDs

Just downgraded back to 0.14.3 and all continues to be well there.

blazoncek, 2025-06-05:
@hacustom why are you regressing? You already [reported](https://github.com/wled/WLED/issues/4389#issuecomment-2543306676) that 6+ outputs resolves the issue.

It is now apparent that the root cause is RMT driver (and not I2S as I thought on a first glance after a revisit).

@willmmiles I suggest you don't spend a lot of time debugging RMT driver as it is far easier to (always) use parallel I2S. Especially once @Makuna manages to reduce backend buffers (which will allow larger LED count without depleting RAM).

@Makuna I've implemented complete pixel buffer management in/for 0.16 (to allow layer blending) and we no longer rely on NPB to provide `GetPixelColor()` or maintain double buffers. So it might be possible to instruct NPB to use single buffer (either during creation or `begin()` or even at compile time) if you ever want to explore that (I have not looked at new additions to NPB in quite some time so please excuse me if that is already implemented).

For anyone experiencing this issue with any version of 0.15, please revert to 0.14 and wait for 0.16 or create additional phantom outputs (1 pixel is enough) while keeping other outputs with less than 300 pixels. I don't think we'll backport RMT solution into 0.15 as it is already waiting in beta for quite some time without any proper feedback from beta testers and 0.16 will bring so many enhancements and performance improvements that it makes no sense to invest time in 0.15.x.

phoenix1589, 2025-06-05:
> > If I change the 2nd segment down to 80 LEDs (bringing the total to 299) then the flashing stops. If I then set the 2nd segment back to 163 - the flashing does not return. If I then change the effect, or turn off back on the outputs - the flashing returns.
> 
> Is the flashing only on LEDs "after" the 80th or on all LEDs on the strip?

The flashing has always occurred on all LED strips. Neither strip individually has more than 300 and this testing was changing just the segment configuration of LED count.

I have now created 3 additional LED Outputs (total of 6) with the 3 additional each have 1 LED and a random GPIO. Having the full 382 LEDs configured there is consistently no flashing.

hacustom, 2025-06-05:
If adding phantom segments is the full and final fix, that works for me and we can close this issue out. I will just not be staying on 0.14.3 and isolating these devices so that only local control is possible.

closed 2025-06-05 by hacustom

MakerMatrix, 2025-06-05:
I would tend to agree - or as I said above, at least provide some documentation and a clean way to see and control what bus gets used for which outputs.  The hacky and unclear workaround of a known bug shouldn't be the only option in production code.  I lost many hours to it.

A good start would be to report which outputs are programmed and what kind of bus they use, under (i)->Instances for each instance.  That might be a great boon to users and devs alike for troubleshooting issues.

Makuna, 2025-06-05:
At a low level, NeoPixelBus (my library) is used to drive the LEDs.  RMT implementation uses exposed APIs that on one of the last core updates got depreciated and further broke it as it gets interrupted by other ISRs (like WiFi).  NPB got updated to the new APIs, which still had issues.  The fix for those came in a very new core update.  But my understanding is that this newer core uses more memory making WLED not feasible on it at the moment (@blazoncek will need to confirm this).

MakerMatrix, 2025-06-06:
I did not notice this until just now because I was focused on flicker, but it appears that with the workaround, my 12V 60/m BTS WS2814 does not render blue, at all.  I guess I'm done with 0.15.x

blazoncek, 2025-06-06:
This has been brought up in the past. @Makuna any thoughts why would RGBW not work correctly with parallel I2S?

blazoncek, 2025-06-21:
One thing that popped during today's teleconference: excessive IRAM_ATTR used in functions for pixel drawing.
One option would be to remove IRAM_ATTR from every (non ISR) function of WLED and see what that brings.
Use custom compile with `-D WLED_SAVE_IRAM` compile flag.

willmmiles, 2025-06-25:
I take it by "DMA" you mean the limited amount of RMT output buffer memory.  I fully agree - the RMTs really are just one step up from bit-banging with those tiny little buffers that need to be serviced every 38us or so.  It's my understanding that in some newer chips (ESP32-S3 and P4), some general-purpose DMA hardware ("GDMA") was added that can be programmed to refill the RMT buffer automatically, alleviating this problem.  I believe that's one of the factors that prompted the RMT driver redesign in IDF v5.

Ultimately, both of Espressif's RMT drivers (old and new) -- and FastLED's custom driver, for that matter -- rely on the "low-priority" interrupts, which, alas, are vulnerable to latency from the FreeRTOS scheduler.  I'm hoping that a custom driver with a higher priority interrupt will be able to improve the reliability for the non-GDMA capable platforms, though at the cost of extra (assembly) code and complexity.  It'll also be interesting to see how it performs across the family; most of them have even *smaller* RMT buffers than the baseline ESP32, and of course the RISC-based ones may need a completely different solution (or not at all - could be FreeRTOS is just faster there).

DedeHai, 2025-07-03:
that is great news

netmindz, 2025-07-03:
Sounds like some good progress, but it's feeling like this is going to be a feature for 0.16 rather than a bugfix for 0.15.1

What is the recommendation to prevent/discourage WLED config that are going to deliver problematic user experience?

blazoncek, 2025-07-04:
AFAIK the patch will require NPB update.

However I briefly tested parallel I2S with 600 pixels (with NPB 2.8.3 in my fork) and it indeed seemed the ESP is not starved of RAM. I do have the code finished in my dev branch including UI improvements. Unfortunately it may need a bit of testing which I'm not able to do ATM.

Anyone interested in helping/testing forced parallel I2S is welcome to pull from my dev branch and compile. Once you have ESP running, configure 2 or more outputs with the same type of LEDs and select/check "Use parallel I2S" option (checkbox). It is still based off of 0.15 codebase with a few modifications (I would not recommend using it in production systems).

blazoncek, 2025-07-04:
> * Parallel I2S; requires all strips to have the same timing, this is not validated by WLED,

I have a fix for that ready.

willmmiles, 2025-07-09:
First test builds (based on 0.15.x branch) are up: https://github.com/willmmiles/WLED/actions/runs/16158643608
Use the "rmt" bins; I don't know why the CI system also zipped up older files along with them.  This should be considered quite experimental; please don't install them on a device that you can't easily reflash via USB.

So far I've tested ESP32 (IDF v3 and v4), and ESP32-S2.  Good chance S3 will also work.  C3 is not supported yet.  I've now received an S3 and C3 board for my bench but I don't have them set up yet.

kylewilliams321, 2025-08-01:
I see this was addressed in 15.1 release so thanks everyone!

However, note mentions led segments need to be under 600 LEDs. My longest is 766. Should I wait or do I need to split this segment in 2?  (Not sure I can but probably could break off 100-150 to new segment possibly as it's permanent lighting up on my house and the controller is in garage on one side of house so the whole longest distance to first LED comes into play..)

MakerMatrix, 2025-08-01:
Apologies for disappearing...I intend to test the new RMT driver as soon as I get time to cycle back to my LED project.  It looks like this code made the 15.1 release and I no longer need the @willmmiles test builds?

Thank you for all the work that went into this!

intermittech, 2025-08-03:
Just to chime in, I've been testing a build provided by @willmmiles that has the RMT bug fix on top of WLED v0.15.1 and that's been working excellent for me on a QuinLED Dig-Quad! So there are 2 fixes available then, using Parallel I2S (which has some limitations) and "fixed RMT" which works well for me and allows mixing of LED types over various channels and such again (I setup a test with 2x 300 ws2812b, 1x 300 sk6812 and 1x ws2812 500 LEDs) and all flickering is gone (in any GPIO order)!

So a fix is close to this issue and then everyone can upgrade to v0.15.1 without worry!

kylewilliams321, 2025-08-03:
Great news!

PR #4890 (2025-09-17): Import RmtHI driver
Pull the RMT High-priority Interrupt driver in to a vendored local library, pending inclusion in upstream NeoPixelBus.

Driver is enabled only for XTensa chips; there's some unresolved issue with nested interrupts on RISCV.

Fixes #4389 for everything except C3s.

## Summary by CodeRabbit

- New Features
  - Adds a high-priority ESP32 RMT driver for NeoPixel output and exposes many preconfigured method variants (including inverted variants) for broad LED type/speed support.

- Performance
  - More reliable transmission with reduced flicker and smoother animations at high up [... cut]