wled/WLED #4390: Internal temperature causes 'error 12: preset not found' even without any presets

orbitinstasis, 2024-12-13:
### What happened?

every time i open one of my two wled instances, I get 'error 12: preset not found'

### To Reproduce Bug

I have updated to version 15 recently and don't remember the error showing there, but I can isolate it to this mod because I emptied all my presets and the error still showed, and only stops showing when i disable the mod, and starts showing again when i re-enable the mod

### Expected Behavior

stop showing the error 

### Install Method

Self-Compiled

### What version of WLED?

0.15.0

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

lost-hope, 2024-12-13:
What preset is set in the dialog field in the setting of the internal temp preset?

orbitinstasis, 2024-12-13:
> What preset is set in the dialog field in the setting of the internal temp preset?

0

lost-hope, 2024-12-13:
Try setting that to -1

blazoncek, 2024-12-13:
0 should be a valid value and should mean "no preset" as no presets can have id 0.

orbitinstasis, 2024-12-13:
Just did a small but reasonable boundary test. 

If i set the 'preset to set' value to -1 and click save, it changes automatically back to 0. Same behaviour on the polling setting if you go anything below 10000.

I always get th error if the setting is 0 for the preset, and also occurs if the number is a positive integer higher than any preset I have saved.

However, the error never shows if the preset to set is an integer above 0 which I also have a saved preset for

lost-hope, 2024-12-13:
OK i will look into it later. Maybe tomorrow. i havent added the preset functionality to the usermod, but normally it should not try to change the preset when set to 0. (atleast that is what a quick look showed)

lost-hope, 2024-12-14:
Found the issue and made a PR.

orbitinstasis, 2024-12-14:
> Found the issue and made a PR.

Thanks - when might this move to the main branch ? I'm assuming that's the branch the devs want most users who are compiling usermods to use?

PR #4397 (2024-12-14): Fixes "preset not found" issue when using internal temp UM
fixes #4390
closed 2024-12-14 by lost-hope